espressif/esp-idf #14389: Example of ILI9341 and XPT2046 with LVGL.

allacmc, 2024-08-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm having a lot of difficulty getting the XPT2046 touchscreen driver to work in conjunction with LVGL.

I used the LVGL example with ILI9341 but this example is with the STMPE610 driver

I've already tried to port, but I'm very lost on which way to go.

The example that expressif has of ILI9341+XPT2046 works perfectly. But I need it in LVGL.

And I don't think there's an easy way to do that.

Where am I going wrong?

I even opened a POST on the LVLG forum but found little help.

https://forum.lvgl.io/t/ili9341-esp-idf-with-lvgl/17732/5

cristianfunes79, 2024-08-18:
> The example that expressif has of ILI9341+XPT2046 works perfectly. But I need it in LVGL.
Hi I'm trying to take a look but I cannot even find that example that you mention, could you please provide the project's path that you're trying to build?

allacmc, 2024-08-18:
I'm actually quite lost as to which example to use.

There is ESP-IDF and IOT-Solution.

And I don't know the difference?

Is IOT-Solution compiled into ESP-IDF?

The example I used (which has smpte610) is this one, but I need it to be in xpt2046:
https://github.com/espressif/esp-idf/tree/d7ca8b94c852052e3bc33292287ef4dd62c9eeb1/examples/peripherals/lcd/spi_lcd_touch

Other question:

https://components.espressif.com/components/lvgl/lvgl_esp32_drivers/versions/0.0.3 - This not work in LVGL 8.4 (Only LVL 7.0)

cristianfunes79, 2024-08-18:
Ok, here is the thing, that driver is still not supported. In order to add support for it, you should implement it. You can take a look into `components/eps_lcd/include/esp_lcd_panel_*` to take as an example. I can take a look for you and try to check if I can develop it but I don't have the hardware so we should work together to test it.

allacmc, 2024-08-18:
I would love to help get this hardware supported in LVGL.

How can I help?

This repository, work fine.

ILI9341+XPT2046 but was not LVGL
I need the ILI9341+XPT2046 with LVGL

https://github.com/nopnop2002/esp-idf-ili9340

This link in forum, is detail for hardware this: https://forum.lvgl.io/t/ili9341-esp-idf-with-lvgl/17732/5

allacmc, 2024-08-18:
I haven’t had any success yet.

Is there any way you can help me?

I found a library that works perfectly, Display and Touch.

But I don’t know how to apply this touch driver to LVGL.

This library is work but was is not LVGL aplyed.

[GitHub](https://github.com/nopnop2002/esp-idf-ili9340)

[GitHub - nopnop2002/esp-idf-ili9340: SPI TFT and XPT2046 touch screen...](https://github.com/nopnop2002/esp-idf-ili9340)
SPI TFT and XPT2046 touch screen controller driver for esp-idf - nopnop2002/esp-idf-ili9340

This library is work ILI9341 but was is not input driver XPT2046

[github.com](https://github.com/espressif/esp-idf/tree/d7ca8b94c852052e3bc33292287ef4dd62c9eeb1/examples/peripherals/lcd/spi_lcd_touch)
[esp-idf/examples/peripherals/lcd/spi_lcd_touch at...](https://github.com/espressif/esp-idf/tree/d7ca8b94c852052e3bc33292287ef4dd62c9eeb1/examples/peripherals/lcd/spi_lcd_touch)
Espressif IoT Development Framework. Official development framework for Espressif SoCs. - espressif/esp-idf

I need the library aplyed with LVGL and ILI9341 and XPT2046

suda-morris, 2024-08-19:
Hi @allacmc all the wheels are available in the ESP Registry. 

XPI2046: https://components.espressif.com/components/atanisoft/esp_lcd_touch_xpt2046
ILI9341: https://components.espressif.com/components/espressif/esp_lcd_ili9341

What you need to do is to add them into your project's `idf_component.yml`. 

For how to use the IDF component manager, please read: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html

allacmc, 2024-08-19:
I really appreciate Mike, suda-moris, cristianfunes.

I managed to make it work perfectly with the lib and the example that Mike / Atani posted.

However, I still have one more difficulty, his example works with LVGL 9, and I need LVGL 8.4.

Although it works in LVGL 8.4 it is not exactly perfect.

atanisoft, 2024-08-19:
https://github.com/limpens/esp32-2432S028 is LVGL 8.x based. You can pin it to a specific version of LVGL in the idf_component.yml file.

allacmc, 2024-08-20:
I managed to make it work with LVGL 8.4.0 but it's still not ideal, because the display doesn't have the colors correctly. 

I still have difficulties.

atanisoft, 2024-08-20:
Incorrect colors is likely due to how you are initializing the display, specifically the color order parameter. There are a few examples out there for the board you are using, including the LVGL 8.3.x (easily upgraded to 8.4.x) above, that can be used with little / no modifications required. In a couple weeks I'll also have the exact same display you are using and can help further after I have time to test with it.

allacmc, 2024-08-22:
Although there is no official example with these specific characteristics:

LVGL+ILI9341_XPT2046

I managed to solve the problem by making some modifications to the code.

folow the tópic for complete solution:

https://forum.lvgl.io/t/ili9341-esp-idf-with-lvgl/17732/13

closed 2024-08-22 by allacmc