espressif/esp-idf #14397: Build Error - Wifi 6 Interval Target Wake Time Example

lalves-isr, 2024-08-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Build Success

### What is the actual behavior?

Build Error

esp-idf-release-v5.3/examples/wifi/itwt/main/itwt_main.c:122:9: error: unknown type name 'wifi_itwt_setup_config_t'; did you mean 'wifi_twt_setup_config_t'?

### Steps to reproduce.

idf.py set-target esp32c6
idf.py build

QingzhaoYin, 2024-08-19:
@lalves-isr Hi，Which version (commit ID) of the IDF you are using? I didn't encounter this issue in release/v5.3 (commit ID: 0526c35e).

lalves-isr, 2024-08-19:
I used this link for the offline transfer of IDF_v5.3.
https://dl.espressif.com/dl/esp-idf/

lalves-isr, 2024-08-19:
In fact, it is strange, because I can find the missing definition in esp_wifi component. 
esp-idf-release-v5.3\components\esp_wifi\include\esp_wifi_he_types.h
Maybe some missing path definition include to find it!?

QingzhaoYin, 2025-03-31:
@lalves-isr Hi, is this issue still exist？ If not we will close this issue. The buid issue is caused by missing  esp_wifi_he_types.h or not included. Please check it.

closed 2025-03-31 by Alvin1Zhang