espressif/esp-idf #14399: Wi-Fi 1 / Wi-Fi 2 / Access Point

diplfranzhoepfinger, 2024-08-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Wi-Fi 1 / Wi-Fi 2 / Access Point

devices from Shelly can connect to 2 Wifi APs

so it is like a dual-Station + AP

how is this possible ? 

i know shelly runs Mongoose OS https://mongoose-os.com/ but is this the Reason ? 

it seems this runs still on top of ESP-IDF ?

MaxwellAlan, 2024-09-05:
Hi @diplfranzhoepfinger 

Sorry, regarding the issue you mentioned, I might not have fully grasped the point you are concerned about. If possible, could you please provide a specific example for reference?

As for the sta1+sta2+softap mode, which can be supported when h/w have 3 interface( which may include the hw FIFO for pkt tx & rx, h/w MAC register). And also there are some limitations, such as if only have one RF, which means the STA1 STA2 and softap should work in the same channel etc.

Thanks.

diplfranzhoepfinger, 2024-09-05:
my Question was how can shelly do this ? they have no special Chips, 
just ESP32 and ESP32C3, 
and they make simultaneous sta1+sta2+softap mode

igrr, 2024-09-05:
https://community.shelly.cloud/topic/1246-purpose-of-wi-fi-2-in-shelly-plug-setup-app/

> WiFi2 is a fallback for when WiFi 1 cannot be reached. As far as I know, Shelly device will only switch to WiFi2 if WiFi1 is completely unavailable. So they won't hop networks based on signal strength.

diplfranzhoepfinger, 2024-09-05:
ahhhhhhhhhhhhh. pure Fallback. 
now it makes sense.

closed 2024-09-05 by diplfranzhoepfinger