espressif/esp-idf #14403: Some wifi problem about esp32

hibfh, 2024-08-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I get 2 wifi trouble:
1st:
when my esp32 board work  a night .I found wifi disapper,and the uart log data like follows:
[08:23:10.626]←◆[0;31mE (47193068) task_wdt: Ta
[08:23:10.664]←◆sk watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[0;31mE (47193068) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (47193068) task_wdt: Tasks currently running:[0m
[0;31mE (47193068) task_wdt: CPU 0: wifi[0m
[0;31mE (47193068) task_wdt: CPU 1: IDLE1[0m
[0;31mE (47193068) task_wdt: Print CPU 0 (current core) backtrace[0m

Backtrace
[08:23:10.698]←◆ce: 0x400F7323:0x3FFC00B0 0x4008468D:0x3FFC00E0 0x4009A42C:0x3FFD5CC0 0x401777D6:0x3FFD5CE0 0x4011DBB2:0x3FFD5D00 0x4011DF07:0x3FFD5D20 0x4012F1FB:0x3FFD5D40 0x4012179E:0x3FFD5DA0 0x40117EAD:0x3FFD5DF0 0x40098195:0x3FFD5E10 0x401740E3:0x3FFD5E30 0x40099379:0x3FFD5E50

0x400f7323: task_wdt_timeout_handling at /home/q/WorkSpace/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:441

 (inlined by) task_wdt_isr at /home/q/WorkSpace/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:515

0x4008468d: _xt_lowint1 at /home/q/WorkSpace/esp/esp-idf/components/xtensa/xtensa_vectors.S:1240

0x4009a42c: hal_mac_deinit at ??:?

0x401777d6: ic_mac_deinit at ??:?

0x4011dbb2: ieee80211_phy_init at ??:?

0x4011df07: chm_set_current_channel at ??:?

0x4012f1fb: ieee80211_update_channel at ??:?

0x4012179e: cnx_node_leave at ??:?

0x40117ead: ieee80211_tx_mgt_cb at ??:?

0x40098195: ppProcTxCallback at ??:?

0x401740e3: ppProcTxDone at ??:?

0x40099379: ppTask at ??:?

what may cause it ?Need some help.
2nd:
I calibrate SSID with follow code
```
wifi_config_t wifi_config = {
        .ap = {
            .ssid = OBD_WIFI_SSID,
            .ssid_len = strlen(OBD_WIFI_SSID),
            .channel = OBD_WIFI_CHANNEL,
            .password = OBD_WIFI_PASSWORD,
            .max_connection = OBD_WIFI_STA_SIZE,
#ifdef CONFIG_ESP_WIFI_SOFTAP_SAE_SUPPORT
            .authmode = WIFI_AUTH_WPA3_PSK,
            .sae_pwe_h2e = WPA3_SAE_PWE_BOTH,
#else /* CONFIG_ESP_WIFI_SOFTAP_SAE_SUPPORT */
            .authmode = WIFI_AUTH_WPA2_PSK,
#endif
            .pmf_cfg = {
                    .required = true,
            },
        },
    };
    if (strlen(OBD_WIFI_PASSWORD) == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }
    if (strlen(obd_product_name)) {
        memcpy(wifi_config.ap.ssid,obd_product_name,strlen(obd_product_name));
        wifi_config.ap.ssid_len = strlen(obd_product_name);
    }
    wifi_config.ap.ssid_hidden = 0;
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));  
}

void obd_wifi_start_ap(){
    esp_err_t ret = esp_wifi_start();
    SoftAP_bIsStart = (ret == ESP_OK ) ? true : false;
    if(SoftAP_bIsStart){
        ESP_LOGI(WIFISAP, "wifi_init_softap finished. SSID:%s password:%s channel:%d",
             obd_product_name, OBD_WIFI_PASSWORD,OBD_WIFI_CHANNEL);
    }else{
        esp_err_to_name(ret);
    }
}
```
obd_product_name is 407S6018
It's so strange that i  can find it in my phone ,but it becomes Hidden wifi in the laptop.

The IDF branch based on 9d6583f7 commit V5.2.2

Related SDK config as follow:
#
# Wi-Fi
#
CONFIG_ESP_WIFI_ENABLED=y
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_ESP_WIFI_STATIC_TX_BUFFER=y
CONFIG_ESP_WIFI_TX_BUFFER_TYPE=0
CONFIG_ESP_WIFI_STATIC_TX_BUFFER_NUM=64
CONFIG_ESP_WIFI_CACHE_TX_BUFFER_NUM=128
# CONFIG_ESP_WIFI_STATIC_RX_MGMT_BUFFER is not set
CONFIG_ESP_WIFI_DYNAMIC_RX_MGMT_BUFFER=y
CONFIG_ESP_WIFI_DYNAMIC_RX_MGMT_BUF=1
CONFIG_ESP_WIFI_RX_MGMT_BUF_NUM_DEF=5
# CONFIG_ESP_WIFI_CSI_ENABLED is not set
CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP_WIFI_TX_BA_WIN=12
CONFIG_ESP_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP_WIFI_RX_BA_WIN=16
# CONFIG_ESP_WIFI_AMSDU_TX_ENABLED is not set
# CONFIG_ESP_WIFI_NVS_ENABLED is not set
CONFIG_ESP_WIFI_TASK_PINNED_TO_CORE_0=y
# CONFIG_ESP_WIFI_TASK_PINNED_TO_CORE_1 is not set
CONFIG_ESP_WIFI_SOFTAP_BEACON_MAX_LEN=752
CONFIG_ESP_WIFI_MGMT_SBUF_NUM=32
CONFIG_ESP_WIFI_IRAM_OPT=y
# CONFIG_ESP_WIFI_EXTRA_IRAM_OPT is not set
# CONFIG_ESP_WIFI_RX_IRAM_OPT is not set
# CONFIG_ESP_WIFI_ENABLE_WPA3_SAE is not set
CONFIG_ESP_WIFI_ENABLE_WPA3_OWE_STA=y
# CONFIG_ESP_WIFI_SLP_IRAM_OPT is not set
CONFIG_ESP_WIFI_SLP_DEFAULT_MIN_ACTIVE_TIME=50
CONFIG_ESP_WIFI_SLP_DEFAULT_MAX_ACTIVE_TIME=10
CONFIG_ESP_WIFI_SLP_DEFAULT_WAIT_BROADCAST_DATA_TIME=15
CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=y
# CONFIG_ESP_WIFI_GMAC_SUPPORT is not set
CONFIG_ESP_WIFI_SOFTAP_SUPPORT=y
# CONFIG_ESP_WIFI_SLP_BEACON_LOST_OPT is not set
CONFIG_ESP_WIFI_ESPNOW_MAX_ENCRYPT_NUM=7
# CONFIG_ESP_WIFI_NAN_ENABLE is not set
CONFIG_ESP_WIFI_MBEDTLS_CRYPTO=y
CONFIG_ESP_WIFI_MBEDTLS_TLS_CLIENT=y
# CONFIG_ESP_WIFI_WAPI_PSK is not set
# CONFIG_ESP_WIFI_11KV_SUPPORT is not set
# CONFIG_ESP_WIFI_MBO_SUPPORT is not set
# CONFIG_ESP_WIFI_DPP_SUPPORT is not set
# CONFIG_ESP_WIFI_11R_SUPPORT is not set
# CONFIG_ESP_WIFI_WPS_SOFTAP_REGISTRAR is not set

#
# WPS Configuration Options
#
# CONFIG_ESP_WIFI_WPS_STRICT is not set
# CONFIG_ESP_WIFI_WPS_PASSPHRASE is not set
# end of WPS Configuration Options

# CONFIG_ESP_WIFI_DEBUG_PRINT is not set
# CONFIG_ESP_WIFI_TESTING_OPTIONS is not set
CONFIG_ESP_WIFI_ENTERPRISE_SUPPORT=y
# CONFIG_ESP_WIFI_ENT_FREE_DYNAMIC_BUFFER is not set
# end of Wi-Fi

Xiehanxin, 2024-08-20:
hi @hibfh , in your application, do you keep the AP pn all the time? or  you try to disable or enable it sometimes?

hibfh, 2024-08-20:
@Xiehanxin What do you mean by "PN"?is it "on"? My app keep AP all the time .maybe it needs to stop for a while？I think my app keep AP on about over 3 hours and hit the trouble. btw, what may cause the strange hidden ssid in win10 ?

hibfh, 2024-08-20:
The second trouble is solved now

Xiehanxin, 2024-09-05:
hi @hibfh  did you download large files or need high througput? sometimes too much HTTP callback can make the watch dog

hibfh, 2024-09-06:
hey @Xiehanxin ，i had no http task but a huge TCP throughput task running that time, did it cause the wdt time out? Recently,I fix some bug like memory fragmentation,tcp fd not close after wifi disconnect ,and it works well so far.maybe it was solved.

closed 2025-03-07 by Alvin1Zhang