espressif/esp-idf #14404: System Crash with WiFi and Bluetooth (Client & Server) Enabled - Assertion Error in llc_disconnect.c 363

matreshka15, 2024-08-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

Enable both WiFi and Bluetooth, with Bluetooth operating in both client and server modes.
ESP32S3 should connect to other server devices smoothly.

### What is the actual behavior?

Enable both WiFi and Bluetooth, with Bluetooth operating in both client and server modes, my system sometimes crashes when ESP32S3 trys to connect to one of the two server devices with the following error:
`assert llc_disconnect.c 363, param 00000004 00000009`

This error leads to a system crash, and I have not been able to identify the root cause. 
I have tried various configurations, but the problem persists.

### Steps to reproduce.

1. Enable Bluetooth in both client and server modes.
2. Configure and connect the WiFi to a network.
3. Make sure ESP32S3 is trying to connect to a server devices
4. With some possibility, the system crashes.

### Debug Logs.

```plain
[I][BTC_TASK][1] connecting to dev: 00:80:e1:26:17:1b
W (30361) BT_GATT: gatt_connect wrong state 2
[I][BTC_TASK][1] stop scan successfully
[D][BTC_TASK][1] GATT_WRITE_EVT, conn_id 1, trans_id 74, handle 118,
[D][BTC_TASK][1] GATT_WRITE_EVT, conn_id 1, trans_id 75, handle 106,
W (30661) BT_HCI: hcif disc complete: hdl 0x4, rsn 0x3e
[D][BTC_TASK][1] GATT_WRITE_EVT, conn_id 1, trans_id 76, handle 127,
[D][BTC_TASK][1] GATT_WRITE_EVT, conn_id 1, trans_id 77, handle 115,
[I]<elbo motor>[BTC_TASK][1] motor stopped at: 107.242
[I]<elbo motor>[BTC_TASK][1] set current_limit: :3000
[D][BTC_TASK][1] GATT_WRITE_EVT, conn_id 1, trans_id 78, handle 139,
[I]<hand mot-ppl>[BTC_TASK][1] Deiniting user model
[I]<hand mot-ppl>[BTC_TASK][1] PPL set user id:1001
assert llc_disconnect.c 363, param 00000004 00000009
log record file write error!: Connection timed out
E (50931) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (50931) task_wdt:  - IDLE1 (CPU 1)
E (50931) task_wdt: Tasks currently running:
E (50931) task_wdt: CPU 0: IDLE0
E (50931) task_wdt: CPU 1: btController
E (50931) task_wdt: Print CPU 1 backtrace

Backtrace: 0x403854A2:0x3FCAF070 0x4037BD31:0x3FCAF090 0x40006FCD:0x3FCECFF0 0x4000FD0F:0x3FCED010 0x420680BB:0x3FCED030 0x4000D025:0x3FCED060 0x4002C4A5:0x3FCED080 0x4205C6CF:0x3FCED0A0 0x4037EEA3:0x3FCED0C0 0x4037F033:0x3FCED0E0 0x4038E3A1:0x3FCED110
0x403854a2: esp_crosscore_isr at C:/Users/Admin/esp/v5.1.4/esp-idf/components/esp_system/crosscore_int.c:96
0x4037bd31: _xt_lowint1 at C:/Users/Admin/esp/v5.1.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
0x40006fcd: r_assert_param in ROM
0x4000fd0f: lld_disc_ind_handler in ROM
0x420680bb: r_ke_task_schedule at ??:?
0x4000d025: r_ke_event_schedule in ROM
0x4002c4a5: r_rwip_schedule in ROM
0x4205c6cf: rw_schedule at ??:?
0x4037eea3: ble_try_turn_on_pll_track at ??:?
0x4037f033: btdm_controller_task at ??:?
0x4038e3a1: vPortTaskWrapper at C:/Users/Admin/esp/v5.1.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

In the ESP sdkconfig configuration,  # CONFIG_ESP_COEX_SW_COEXIST_ENABLE is not set, in order to decrease wifi transmitting delay.

> If enabled, WiFi & Bluetooth coexistence is controlled by software rather than hardware. Recommended for heavy traffic scenarios. Both coexistence configuration options are automatically managed, no user intervention is required. If only Bluetooth is used, it is recommended to disable this option to reduce binary file size.

matreshka15, 2024-08-20:
Detailed sdkconfig file is shown below:

> #
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.1.4 Project Configuration
#
CONFIG_SOC_MPU_MIN_REGION_SIZE=0x20000000
CONFIG_SOC_MPU_REGIONS_MAX_NUM=8
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_APPCPU_HAS_CLOCK_GATING_BUG=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_ARBITER_SUPPORTED=y
CONFIG_SOC_ADC_DIG_IIR_FILTER_SUPPORTED=y
CONFIG_SOC_ADC_MONITOR_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=24
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=4
[... 2550 lines cut ...]
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

matreshka15, 2024-08-20:
updates: the server device will actively disconnect, while ESP32S3 trying to connect. Leading to this issue.

esp-zhp, 2024-08-20:
@matreshka15 
it looks that the connection was already established before create connection,and there is a PHY update onging. However, from the existing logs, I'm still unable to find the root cause. Could you provide me with a demo to reproduce this problem?thanks.

esp-zhp, 2024-08-20:
Additionally, does the issue still persist if WiFi is not being used?

esp-zhp, 2024-08-20:
In order to further analyze the root cause of the issue, I have added HCI logging. You can apply this patch using the git am command.
[0001-add-esp32c3-s3-hci-log.zip](https://github.com/user-attachments/files/16675091/0001-add-esp32c3-s3-hci-log.zip)

closed 2024-12-17 by Alvin1Zhang