espressif/esp-idf #14405: Disable random delay before performing sntp request

EngineerOfMaster, 2024-08-20:
### Is your feature request related to a problem?

We are developping a very energy-sensitiv product (based on esp32s3 / idf 5.2.1), that is battery powered and therefore relays on little power consumption as possible. A hugh proprotion of battery life time goes by the regularly performed send attempts via wifi, otherwise the system is in deep sleep. Nice we are communicating via https we need a correct time on the device for authentification of the certificate. So we perform a sntp request each time we do a send attempt. Apparently there is a default random delay implemented (0s - 5s) before a sntp request is triggered.

### Describe the solution you'd like.

sntp implementation checks for a macro called SNTP_STARTUP_DELAY if a random delay is applied or not. Unfortunately this macro is hardcoded. This marco is checked in nstp_opts.h from lwip library:

```
#if !defined SNTP_STARTUP_DELAY || defined __DOXYGEN__
#ifdef LWIP_RAND
#define SNTP_STARTUP_DELAY          1
#else
 #define SNTP_STARTUP_DELAY          0
 #endif
#endif
```

```
sntp_init(void)
{
 ...
#if SNTP_STARTUP_DELAY
        sys_timeout((u32_t)SNTP_STARTUP_DELAY_FUNC, sntp_request, NULL);
#else
        sntp_request(NULL);
#endif
...
}

```
the feature that is requested here is to include this option (SNTP_STARTUP_DELAY) to sdkconfig, so it can be set via menuconfig.

### Describe alternatives you've considered.

To get rid of the random delay before sntp request, we add this definiton via cmake file. But in general we do not want to manipulate idf implementations from "outside" 

`add_compile_definitions(SNTP_STARTUP_DELAY=0)`

david-cermak, 2024-08-20:
Hi @EngineerOfMaster 

It's already possible to configure:
* [`CONFIG_LWIP_SNTP_STARTUP_DELAY`](https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/api-reference/kconfig.html#config-lwip-sntp-startup-delay)
* [`CONFIG_LWIP_SNTP_MAXIMUM_STARTUP_DELAY`](https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/api-reference/kconfig.html#config-lwip-sntp-maximum-startup-delay)

But only available from `v5.3`, is it okay for you to switch to it (already latest stable)?

EngineerOfMaster, 2024-08-20:
Thank you for your fast response.
I wasn`t aware of this. This is fine, at some point we will go forward with the new idf version.

closed 2024-10-30 by david-cermak