espressif/esp-idf #14408: Impossible to customize SPIRAM heap allocation

nicklasb, 2024-08-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi!

I need to allocate a ~5 MB block in SPIRAM using himem for a Tensorflow arena (~2 MB model). 
For some reason, however, I only get 4456448 bytes, even though I have a 8MB PSRAM (reported by esp_spiram_get_size).
But there are no ways in ESP-IDF to control the heap allocation (anymore).
Also, if I enable "Initialize SPI RAM during startup", 3840K is munched up by the heap allocator, which I *think* might happen silently if I manually run esp_psram_init(). 

It is unclear to me esp_psram_init() actually does, it certainly doesn't do what "Initialize SPI RAM during startup" does, and I can't find any ways to create my own heap regions to make room for larger himem allocation. 
My assumption here is that 4,4 MB is too much for there to be some internal 4 MB limit.

Spritetm, 2024-08-21:
You can implicitly lower the amount of memory used by the heap allocator by raising CONFIG_SPIRAM_BANKSWITCH_RESERVE in menuconfig.

nicklasb, 2024-08-21:
Hi, that fixed that problem, sort of. 
I am now getting a himem size: 6225920, free himem phys size: 6225920, free spiram 4191744.

However, now it is failing in the next step, esp_himem_alloc_map_range cannot allocate even nearly the amount of memory range that esp_himem_alloc was able to allocate:
------------------
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x400d0fe7
  #0  0x400d0fe7 in init_data() at src/main.cpp:47 (discriminator 1)

file: "src/main.cpp" line 47
func: void init_data()
expression: esp_himem_alloc_map_range(5*1024*1024, &range)

abort() was called at PC 0x4008900f on core 0
----------------

I have to go all the way down to ~1 MB for it to work which is strange. 
Not much is done before this, thinking about possible problems witn contigousness or something like that, but this is the first thing I do in the application, and I have seen nothing about limits for esp_himem_alloc_map_range?

Spritetm, 2024-08-21:
No, it's not. The himem API is a bank switching scheme; while it allows you to work around the limit where at max 4MiB of PSRAM is accessible, it still needs to work within that hardware limit. As such, the amount of memory you can map is limited by the amount of banks you allocate to himem using CONFIG_SPIRAM_BANKSWITCH_RESERVE. (If it were possible to map in 5MiB of RAM, we'd just do that by default and have the heap allocator handle it, no need for himem shenanigans.)

Spritetm, 2024-08-21:
As an aside, I'd suggest moving to an ESP32-S3 to solve it - that chip can have up to 32MiB of flash+ram mapped at the same time, plus it has some AI extensions that allow it to run things like TensorFlow a lot faster (given the software makes use of it).

nicklasb, 2024-08-21:
Understood. 
Yeah I know, have ordered a couple of S3s, but as they take a while I just want to verify the solution regardless of performance at this stage. Annoyingly I had one but I manage to pull the magic smoke out of it.  :-)

I do however think that this is important information, as I do not see these limitations being explictly expressed anywhere.

Thank you, closing!

closed 2024-08-21 by nicklasb