espressif/esp-idf #14409: mbedtls causing stubs from libnosys to get linked

KaeLL, 2024-08-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

`git describe`: v5.4-dev-2194-gd7ca8b94c8

`ld` output:
/idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/mbedtls/mbedtls/library/libmbedx509.a(x509_crt.c.obj):(.literal.mbedtls_x509_crt_parse_path+0x28): warning: readdir is not implemented and will always fail
/idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/mbedtls/mbedtls/library/libmbedx509.a(x509_crt.c.obj):(.literal.mbedtls_x509_crt_parse_path+0x2c): warning: closedir is not implemented and will always fail
/idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/mbedtls/mbedtls/library/libmbedx509.a(x509_crt.c.obj):(.literal.mbedtls_x509_crt_parse_path+0x10): warning: opendir is not implemented and will always fail

igrr, 2024-08-21:
Could you please attach your sdkconfig and the full build output? I suppose that VFS component is either not included in the build and/or CONFIG_VFS_SUPPORT_DIR is disabled. If that is the case, the warning makes sense to me: we do provide the stub but it will always return with an error code.

KaeLL, 2024-08-21:
> I suppose ... CONFIG_VFS_SUPPORT_DIR is disabled

You supposed correctly. Is there an argument to be made for the use of mbedtls y-select this config? Thanks, btw, and feel free to close the issue.

igrr, 2024-08-21:
> for the use of mbedtls y-select this config

There is no "enable switch" for mbedtls use, so it's not trivial. Also, mbedtls component is currently aleays linked into the app due to some dependencies from IDF itself, so if it enables VFS support, it won't be possible to disable VFS at all.

However, we can do it the other way around: make mbedtls MBEDTLS_FS_IO config option dependent on CONFIG_VFS_SUPPORT_IO && CONFIG_VFS_SUPPORT_DIR.

closed 2024-10-07 by espressif-bot