espressif/esp-idf #14412: USB HOST Interrupt out transfer error

carllin1010, 2024-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-S3（revision: v0.1）

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-USB-OTG

### Power Supply used.

USB

### What is the expected behavior?

我使用一个OTG开发板作为USB HOST，另一个OTG开发板作为USB DEV，使用HID协议双向通信，每个数据大小64B，通信频率500Hz。
I am using an OTG development board as the USB HOST and another OTG development board as the USB DEV. They are communicating bidirectionally using the HID protocol, with each data packet being 64B in size and the communication frequency being 500Hz.

### What is the actual behavior?

HOST端在发送数据几分钟之内，就会usb_host_transfer_submit就会报错ESP_ERR_NOT_FINISHED       （0x10C），urb处于inflight状态，无法再发送数据了，通过调试器发现，USB的主中断（intr_hdlr_main）已经不会再进入了

On the HOST side, within a few minutes of sending data, the function `usb_host_transfer_submit` returns the error `ESP_ERR_NOT_FINISHED (0x10C)`. The URB is in an inflight state and cannot send any more data.
Through the debugger, I found that the main USB interrupt (`intr_hdlr_main`) is no longer being triggered.

[image]

### Steps to reproduce.

# USB DEV Device
基于TUSB_HID例程，修改为增加OUT端点用于接收数据：
Based on the TUSB_HID example, I modified it to add an OUT endpoint for receiving data:
```c
/*
 * SPDX-FileCopyrightText: 2022-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

#include <stdlib.h>
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "tinyusb.h"
#include "class/hid/hid_device.h"
#include "driver/gpio.h"

#define APP_BUTTON (GPIO_NUM_0) // Use BOOT signal by default
static const char *TAG = "example";
[... 854 lines cut ...]
E (145155) CLASS: Failed to submit transfer, error 268
E (145161) CLASS: Failed to submit transfer, error 268
E (145167) CLASS: Failed to submit transfer, error 268
E (145172) CLASS: Failed to submit transfer, error 268
E (145178) CLASS: Failed to submit transfer, error 268
E (145184) CLASS: Failed to submit transfer, error 268
E (145190) CLASS: Failed to submit transfer, error 268
E (145195) CLASS: Failed to submit transfer, error 268
E (145201) CLASS: Failed to submit transfer, error 268
```

### More Information.

1. 使用FreeRTOS的单核模式不能解决
2. 修改处理器运行速度不能解决
3. 测试过90Hz、160Hz、200Hz的通信频率都会复现
4. 在发生故障后，发送数据不会进入intr_hdlr_main，当拔出DEV，intr_hdlr_main可以进入并发现设备断开。
5. 使用逻辑分析仪抓包，HOST最后一个发向OUT端点的数据帧，DEV有正常回应ACK，然后HOST就不发数据了。
6. 实际上我是要对一个HID协议设备进行通信和控制出现了该问题，然后用ESP32作为DEV（也就是上面提交的代码），也出现了该问题。
1. Using FreeRTOS in uni-core mode does not solve the problem.
2. Changing the processor running speed does not solve the problem.
3. I have tested communication frequencies of 90Hz, 160Hz, and 200Hz, and the issue still occurs, not just at these specific frequencies.
4. After the fault occurs, data transmission does not trigger `intr_hdlr_main`. Upon disconnecting the DEV, `intr_hdlr_main` can enter and detect the device disconnection.
5. Logic analyzer captures show that the HOST sends the last data frame to the OUT endpoint, DEV responds with ACK normally, but the HOST then stops sending data.
6.I actually encounter this issue when communicating with and controlling a HID protocol device. Using an ESP32 as the DEV (as demonstrated in the provided code), I also face the same issue.

tore-espressif, 2024-09-02:
Hello @carllin1010 , Thank you for the bug report, I briefly checked your code and configuration and it seems correct.
So we will debug further but it might take some more time.

In the meantime, could post a zip archive of your both device and host projects? It would help us with reproducing the problem. If you have any traces from logic analyzer, that would be useful too.

Thank you and I'm sorry for the inconvenience!

carllin1010, 2024-09-03:
@tore-espressif [code_and_analyzer_data.zip](https://github.com/user-attachments/files/16843022/code_and_analyzer_data.zip) This is the code to reproduce the problem, along with the logic analyzer data captured when the fault occurs. I hope this can help you troubleshoot the problem. You can open the logic analyzer data with this software: https://www.qdkingst.com/en/download. The last successful data transmission occurred at 6.377129s.

[image]

tore-espressif, 2024-09-03:
@carllin1010 thank you for all the information. It seems to be a problem with periodic INTR schedulet in HCD layer. Unfortunately this will take more time to properly fix. I'll try to provide a workaround for the time being

carllin1010, 2024-09-03:
I add your code like this:
```c
static void usb_send_task(void *arg)
{
    class_driver_t *driver_obj = (class_driver_t *)arg;
    vTaskDelay(5000);
    while (1) {
        driver_obj->transfer->num_bytes = 64;
        driver_obj->transfer->callback = transfer_cb;
        driver_obj->transfer->context = NULL;
        driver_obj->transfer->bEndpointAddress = 0x01;
        driver_obj->transfer->device_handle = driver_obj->dev_hdl;
        driver_obj->transfer->timeout_ms = 1000;
        esp_err_t ret = usb_host_transfer_submit(driver_obj->transfer);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "Failed to submit transfer, error %d", ret);
            driver_obj->transfer->flags = USB_TRANSFER_FLAG_ZERO_PACK;
        } 
        vTaskDelay(2);
    }
}
```
The problem still occurs and generates an assert failed:
```log
......
I (1409) CLASS: Getting Manufacturer string descriptor
TinyUSB 
I (1416) CLASS: Getting Product string descriptor
TinyUSB Device
I (1422) CLASS: Getting Serial Number string descriptor
123456
I (2311) CLASS: write_cnt = 0
......
I (6311) CLASS: write_cnt = 0
I (7311) CLASS: write_cnt = 500
......
I (19311) CLASS: write_cnt = 500
I (20311) CLASS: write_cnt = 500
I (21311) CLASS: write_cnt = 500
E (21945) CLASS: Failed to submit transfer, error 268
E (21947) CLASS: Failed to submit transfer, error 268
E (21949) CLASS: Failed to submit transfer, error 268
E (21952) CLASS: Failed to submit transfer, error 268
E (21958) CLASS: Failed to submit transfer, error 268
E (21964) CLASS: Failed to submit transfer, error 268
E (21969) CLASS: Failed to submit transfer, error 268
E (21975) CLASS: Failed to submit transfer, error 268
E (21981) CLASS: Failed to submit transfer, error 268
E (21986) CLASS: Failed to submit transfer, error 268
E (21992) CLASS: Failed to submit transfer, error 268
E (21997) CLASS: Failed to submit transfer, error 268
E (22003) CLASS: Failed to submit transfer, error 268
E (22009) CLASS: Failed to submit transfer, error 268
E (22014) CLASS: Failed to submit transfer, error 268
E (22020) CLASS: Failed to submit transfer, error 268
E (22026) CLASS: Failed to submit transfer, error 268
E (22031) CLASS: Failed to submit transfer, error 268
E (22037) CLASS: Failed to submit transfer, error 268
E (22043) CLASS: Failed to submit transfer, error 268
E (22048) CLASS: Failed to submit transfer, error 268
E (22054) CLASS: Failed to submit transfer, error 268
E (22059) CLASS: Failed to submit transfer, error 268
E (22065) CLASS: Failed to submit transfer, error 268
E (22071) CLASS: Failed to submit transfer, error 268
E (22076) CLASS: Failed to submit transfer, error 268
E (22082) CLASS: Failed to submit transfer, error 268
E (22088) CLASS: Failed to submit transfer, error 268
E (22093) CLASS: Failed to submit transfer, error 268
E (22099) CLASS: Failed to submit transfer, error 268
E (22105) CLASS: Failed to submit transfer, error 268
[... 18 lines cut ...]

Backtrace: 0x40375f6e:0x3fc9e360 0x4037c93d:0x3fc9e390 0x40389bae:0x3fc9e3c0 0x42012dc7:0x3fc9e520 0x42013125:0x3fc9e590 0x420107e4:0x3fc9e5d0 0x4201093a:0x3fc9e620 0x403788fd:0x3fc9e680 0x42010a3f:0x3fcac750 0x40048c69:0x3fcac770 0x4200979e:0x3fcac790 0x4201c776:0x3fcac7d0 0x4201bf01:0x3fcac7f0 0x4201bf52:0x3fcac810 0x4201c545:0x3fcac830 0x42020693:0x3fcac860 0x42020071:0x3fcac880 0x4201c7ed:0x3fcacba0 0x4202a139:0x3fcacbd0 0x403897c9:0x3fcacc20 0x4200a4b7:0x3fcacc80
0x40375f6e: panic_abort at /home/esp/esp/esp-idf/components/esp_system/panic.c:463
0x4037c93d: esp_system_abort at /home/esp/esp/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x40389bae: __assert_func at /home/esp/esp/esp-idf/components/newlib/assert.c:80
0x42012dc7: _buffer_parse_intr at /home/esp/esp/esp-idf/components/usb/hcd_dwc.c:2504 (discriminator 3)
0x42013125: _buffer_parse at /home/esp/esp/esp-idf/components/usb/hcd_dwc.c:2598
0x420107e4: _intr_hdlr_chan at /home/esp/esp/esp-idf/components/usb/hcd_dwc.c:909
0x4201093a: intr_hdlr_main at /home/esp/esp/esp-idf/components/usb/hcd_dwc.c:978
0x403788fd: _xt_lowint1 at /home/esp/esp/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x42010a3f: intr_hdlr_main at /home/esp/esp/esp-idf/components/usb/hcd_dwc.c:1003
0x40048c69: uart_tx_one_char in ROM
0x4200979e: _write_r_console at /home/esp/esp/esp-idf/components/newlib/syscalls.c:35
0x4201c776: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x4201bf01: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x4201bf52: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:278
0x4201c545: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x42020693: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfiprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x42020071: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1774 (discriminator 1)
0x4201c7ed: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x4202a139: esp_log_writev at /home/esp/esp/esp-idf/components/log/log.c:210
0x403897c9: esp_log_write at /home/esp/esp/esp-idf/components/log/log.c:220
0x4200a4b7: usb_send_task at /home/esp/esp/usb_host_lib/main/class_driver.c:99 (discriminator 1)
```

tore-espressif, 2024-09-03:
Sorry, I mean add the line before `usb_host_transfer_submit()`

Like
``` c
        driver_obj->transfer->num_bytes = 64;
        driver_obj->transfer->callback = transfer_cb;
        driver_obj->transfer->context = NULL;
        driver_obj->transfer->bEndpointAddress = 0x01;
        driver_obj->transfer->device_handle = driver_obj->dev_hdl;
        driver_obj->transfer->timeout_ms = 1000;
        driver_obj->transfer->flags = USB_TRANSFER_FLAG_ZERO_PACK; // Add here
        esp_err_t ret = usb_host_transfer_submit(driver_obj->transfer);
```

carllin1010, 2024-09-03:
it doesn't reboot and can continue transmission after reporting an error, The log is too long, so I placed it in this file.
[log.txt](https://github.com/user-attachments/files/16844991/log.txt)

tore-espressif, 2024-09-03:
OK, thanks for trying that out, we will investigate internally

dorianim, 2026-09-21:
Has there been any solution to this issue? I think I am having a similar problem on esp32-p4.

wuxiang1994, 2026-10-08:
> Has there been any solution to this issue? I think I am having a similar problem on esp32-p4.

I am having a similar problem on esp32-p4，too.
using P4 for UVC camera  host