espressif/esp-idf #14413: NimBLE Gatt Server & ESP-BLE-MESH provisioner coexistence

buganini, 2024-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF: v5.3

sdkconfig includes:
```
CONFIG_BLE_MESH_SUPPORT_BLE_ADV=y
CONFIG_BLE_MESH_BLE_ADV_BUF_COUNT=2
```

I used to have `Bluedroid` Gatt Server + `ESP-BLE-MESH` coexistence by using `esp_ble_mesh_start_ble_advertising`, it will send both advertisements and both gatts_profile_event_handler/esp_ble_mesh_prov_callback are invoked to handle BLE pairing/bonding and BLE-MESH provisioning.

But due to the following issue I have to move to NimBLE
https://github.com/meshtastic/firmware/issues/266

NimBLE GATT Server works and ESP-IDF-MESH works, but when both are enabled, only one of them functions correctly.

With NimBLE I have to call `ble_gap_adv_start` to register GAP event handler for pairing/bonding,
I have tried different combinations and orders of
`esp_ble_mesh_start_ble_advertising`, `ble_gap_adv_start`, `esp_ble_mesh_init/esp_ble_mesh_provisioner_prov_enable`
, but I just cannot get both handlers to work simultaneously. It seems that ESP-BLE-MESH is not integrated with NimBLE. Can someone confirm this?

Laosepy, 2024-09-11:
HI,must use nimble-mesh,This comes from the accompanying code for nimble，But the examples provided in the idf seem to have bugs，first running ,I found that it wouldn't broadcast, and there was a stack error after I corrected the sample code as explained in the comments

buganini, 2024-10-11:
I am using this patch to have my NimBLE GATT Service + ESP-BLE-MESH working together

# Changes
* Added a weak esp_ble_mesh_init_gatts to allow injection of extra services
* Added a subscribe callback in struct bt_mesh_conn_cb

# Patches
``` diff
diff --git a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
index 1cb239a3ca..2d2a01bb56 100644
--- a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
+++ b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
@@ -513,7 +513,7 @@ struct bt_mesh_conn_cb {
      *  @param conn New connection object.
      *  @param err HCI error. Zero for success, non-zero otherwise.
      */
-    void (*connected)(struct bt_mesh_conn *conn, uint8_t err);
+    void (*connected)(struct bt_mesh_conn *conn, uint8_t status);

     /** @brief A connection has been disconnected.
      *
@@ -524,6 +524,8 @@ struct bt_mesh_conn_cb {
      *  @param reason HCI reason for the disconnection.
      */
     void (*disconnected)(struct bt_mesh_conn *conn, uint8_t reason);
+
+    void (*subscribe)(struct bt_mesh_conn *conn, uint16_t attr_handle);
 };

 struct bt_mesh_prov_conn_cb {
@@ -799,6 +801,8 @@ int bt_mesh_encrypt_le(const uint8_t key[16], const uint8_t plaintext[16],
 int bt_mesh_encrypt_be(const uint8_t key[16], const uint8_t plaintext[16],
                        uint8_t enc_data[16]);

+void esp_ble_mesh_init_gatts(void);
+
 enum {
     BLE_MESH_EXCEP_LIST_SUB_CODE_ADD = 0,
     BLE_MESH_EXCEP_LIST_SUB_CODE_REMOVE,
diff --git a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
index 0418b76c54..0d97b1a907 100644
--- a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
+++ b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
@@ -652,7 +652,7 @@ static int gap_event_cb(struct ble_gap_event *event, void *arg)
             uint8_t index = BLE_MESH_GATT_GET_CONN_ID(event->connect.conn_handle);
             if (index < BLE_MESH_MAX_CONN) {
                 bt_mesh_gatts_conn[index].handle = BLE_MESH_GATT_GET_CONN_ID(event->connect.conn_handle);
-                (bt_mesh_gatts_conn_cb->connected)(&bt_mesh_gatts_conn[index], 0);
+                (bt_mesh_gatts_conn_cb->connected)(&bt_mesh_gatts_conn[index], event->connect.status);
             }
             memcpy(bt_mesh_gatts_addr, desc.peer_id_addr.val, BLE_MESH_ADDR_LEN);
             /* This is for EspBleMesh Android app. When it tries to connect with the
@@ -738,6 +738,14 @@ static int gap_event_cb(struct ble_gap_event *event, void *arg)
                                    0 /* offset */, 0)) > 0) {
             }
         }
+
+        if (bt_mesh_gatts_conn_cb != NULL && bt_mesh_gatts_conn_cb->subscribe != NULL) {
+            uint8_t index = BLE_MESH_GATT_GET_CONN_ID(event->subscribe.conn_handle);
+            if (index < BLE_MESH_MAX_CONN) {
+                bt_mesh_gatts_conn[index].handle = BLE_MESH_GATT_GET_CONN_ID(event->subscribe.conn_handle);
+                (bt_mesh_gatts_conn_cb->subscribe)(&bt_mesh_gatts_conn[index], event->subscribe.attr_handle);
+            }
+        }
         return 0;

     case BLE_GAP_EVENT_MTU:
@@ -1717,6 +1725,10 @@ void gatt_register_cb(struct ble_gatt_register_ctxt *ctxt,
     }
 }

+__attribute__((weak)) void esp_ble_mesh_init_gatts(void) {
+    // register your services here
+}
+
 void bt_mesh_gatt_init(void)
 {
     ble_att_set_preferred_mtu(BLE_ATT_MTU_DFLT);
@@ -1731,6 +1743,8 @@ void bt_mesh_gatt_init(void)
         ble_svc_gap_init();
         ble_svc_gatt_init();

+        esp_ble_mesh_init_gatts();
+
         rc = ble_gatts_count_cfg(svc_defs);
         assert(rc == 0);
```

# Usage
``` C
#include "esp_ble_mesh_ble_api.h"
#include "mesh/adapter.h" // bt_mesh_gatts_conn_cb_register

// called in esp_ble_mesh_init()
void esp_ble_mesh_init_gatts(void) {
    int rc = ble_gatts_count_cfg(gatt_svr_svcs);
    if (rc != 0) {
        return;
    }

    rc = ble_gatts_add_svcs(gatt_svr_svcs);
    if (rc != 0) {
        return;
    }
}

static void gap_connected(struct bt_mesh_conn *conn, uint8_t status) {
    uint16_t conn_handle = conn->handle;
    printf("gap_disconnected: conn_handle=%d, status=%d\n", (int)conn_handle, (int)status);
}

static void gap_disconnected(struct bt_mesh_conn *conn, uint8_t reason) {
    uint16_t conn_handle = conn->handle;
    printf("gap_disconnected: conn_handle=%d, reason=%d\n", (int)conn_handle, (int)reason);
}

static void gap_subscribe(struct bt_mesh_conn *conn, uint16_t attr_handle) {
    uint16_t conn_handle = conn->handle;
    printf("gap_subscribe: conn_handle=%d, attr_handle=%d\n", (int)conn_handle, (int)attr_handle);
}

struct bt_mesh_conn_cb conn_cb = {
    .connected = gap_connected,
    .disconnected = gap_disconnected,
    .subscribe = gap_subscribe,
};

void  ble_on_sync_proc() {
    rc = esp_ble_mesh_init(&provision, &composition);
    if (rc != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize mesh stack (err %d)", rc);
        return;
    }

    // register gap events callback
    bt_mesh_gatts_conn_cb_register(&conn_cb);

   esp_ble_mesh_node_prov_enable((esp_ble_mesh_prov_bearer_t)(ESP_BLE_MESH_PROV_ADV|ESP_BLE_MESH_PROV_GATT));
}
```

buganini, 2024-10-12:
I am using this patch to have my NimBLE GATT Services + ESP-BLE-MESH working together

# Changes
* Add a weak esp_ble_mesh_init_gatts to allow injection of extra services
* Support extended advertising
* Add a function to get the current gap event callback for event feedback (use with legacy advertising)
* Add a function to get the gap event callback for event forwarding (use with extended advertising)

# Patches
``` diff
diff --git a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
index 1cb239a3ca..2a0b1a8077 100644
--- a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
+++ b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
@@ -20,6 +20,8 @@
 extern "C" {
 #endif
 
+#define BLE_MESH_EXT_ADV_IDX 0
+
 /* BLE Mesh Max Connection Count */
 #ifdef CONFIG_BT_BLUEDROID_ENABLED
 #define BLE_MESH_MAX_CONN   CONFIG_BT_ACL_CONNECTIONS
@@ -513,7 +515,7 @@ struct bt_mesh_conn_cb {
      *  @param conn New connection object.
      *  @param err HCI error. Zero for success, non-zero otherwise.
      */
-    void (*connected)(struct bt_mesh_conn *conn, uint8_t err);
+    void (*connected)(struct bt_mesh_conn *conn, uint8_t status);
 
     /** @brief A connection has been disconnected.
      *
@@ -718,6 +720,8 @@ int bt_le_update_white_list(struct bt_mesh_white_list *wl);
 
 void bt_mesh_gatts_conn_cb_register(struct bt_mesh_conn_cb *cb);
 void bt_mesh_gatts_conn_cb_deregister(void);
+struct bt_mesh_conn_cb * bt_mesh_gatts_conn_cb_current();
+int (*bt_mesh_gap_event_cb(void))(struct ble_gap_event *event, void *arg);
 
 int bt_mesh_gatts_disconnect(struct bt_mesh_conn *conn, uint8_t reason);
 
@@ -799,6 +803,8 @@ int bt_mesh_encrypt_le(const uint8_t key[16], const uint8_t plaintext[16],
 int bt_mesh_encrypt_be(const uint8_t key[16], const uint8_t plaintext[16],
                        uint8_t enc_data[16]);
 
+void esp_ble_mesh_init_gatts(void);
+
 enum {
     BLE_MESH_EXCEP_LIST_SUB_CODE_ADD = 0,
     BLE_MESH_EXCEP_LIST_SUB_CODE_REMOVE,
diff --git a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
index 0418b76c54..9d0e9e40cb 100644
--- a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
+++ b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
@@ -17,6 +17,7 @@
 #include "host/ble_hs.h"
 #include "host/ble_uuid.h"
 #include "host/ble_att.h"
+#include "host/ble_gap.h"
 #include "host/ble_gatt.h"
 #include "services/gap/ble_svc_gap.h"
 #include "services/gatt/ble_svc_gatt.h"
@@ -652,7 +653,7 @@ static int gap_event_cb(struct ble_gap_event *event, void *arg)
             uint8_t index = BLE_MESH_GATT_GET_CONN_ID(event->connect.conn_handle);
             if (index < BLE_MESH_MAX_CONN) {
                 bt_mesh_gatts_conn[index].handle = BLE_MESH_GATT_GET_CONN_ID(event->connect.conn_handle);
-                (bt_mesh_gatts_conn_cb->connected)(&bt_mesh_gatts_conn[index], 0);
+                (bt_mesh_gatts_conn_cb->connected)(&bt_mesh_gatts_conn[index], event->connect.status);
             }
             memcpy(bt_mesh_gatts_addr, desc.peer_id_addr.val, BLE_MESH_ADDR_LEN);
[... 204 lines cut ...]
}

void  ble_on_sync_proc() {
    rc = esp_ble_mesh_init(&provision, &composition);
    if (rc != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize mesh stack (err %d)", rc);
        return;
    }

#if MYNEWT_VAL(BLE_EXT_ADV)
        int err = ble_gap_ext_adv_configure(YOUR_ADV_IDX, &params, NULL,
                                   bleprph_gap_event, NULL);
        // ble_gap_ext_adv_set_data
        // ble_gap_ext_adv_start
#else
    // obtain the current gatt proxy callback for event feedback
    proxy_conn_cb = bt_mesh_gatts_conn_cb_current();

    // overwrite gap events callback
    bt_mesh_gatts_conn_cb_register(&conn_cb);
#endif

    esp_ble_mesh_node_prov_enable((esp_ble_mesh_prov_bearer_t)(ESP_BLE_MESH_PROV_ADV|ESP_BLE_MESH_PROV_GATT));
}
```

buganini, 2024-10-14:
Here is a new patch to allow GATT Proxy + ESP-BLE-MESH working together, with less intrusive to the original code, but more flexible.

# Changes
* Add a weak esp_ble_mesh_init_gatts to allow injection of extra services
* Support extended advertising
* Add a weak wrapper `weak_esp_ble_mesh_gap_event_cb` for `esp_ble_mesh_gap_event_cb` (original `disc_cb`) to allow injection of gap event handling

# Patches
``` diff
diff --git a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
index 1cb239a3ca..3f038ece24 100644
--- a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
+++ b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
@@ -20,6 +20,8 @@
 extern "C" {
 #endif
 
+#define BLE_MESH_EXT_ADV_IDX 0
+
 /* BLE Mesh Max Connection Count */
 #ifdef CONFIG_BT_BLUEDROID_ENABLED
 #define BLE_MESH_MAX_CONN   CONFIG_BT_ACL_CONNECTIONS
@@ -799,6 +801,10 @@ int bt_mesh_encrypt_le(const uint8_t key[16], const uint8_t plaintext[16],
 int bt_mesh_encrypt_be(const uint8_t key[16], const uint8_t plaintext[16],
                        uint8_t enc_data[16]);
 
+int weak_esp_ble_mesh_gap_event_cb(struct ble_gap_event *event, void *arg);
+int esp_ble_mesh_gap_event_cb(struct ble_gap_event *event, void *arg);
+void esp_ble_mesh_init_gatts(void);
+
 enum {
     BLE_MESH_EXCEP_LIST_SUB_CODE_ADD = 0,
     BLE_MESH_EXCEP_LIST_SUB_CODE_REMOVE,
diff --git a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
index 0418b76c54..2a703ae7f1 100644
--- a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
+++ b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
@@ -17,6 +17,7 @@
 #include "host/ble_hs.h"
 #include "host/ble_uuid.h"
 #include "host/ble_att.h"
+#include "host/ble_gap.h"
 #include "host/ble_gatt.h"
 #include "services/gap/ble_svc_gap.h"
 #include "services/gatt/ble_svc_gatt.h"
@@ -398,7 +399,11 @@ static int svc_disced(uint16_t conn_handle, const struct ble_gatt_error *error,
 }
 #endif /* (CONFIG_BLE_MESH_PROVISIONER && CONFIG_BLE_MESH_PB_GATT) || CONFIG_BLE_MESH_GATT_PROXY_CLIENT */
 
-static int disc_cb(struct ble_gap_event *event, void *arg)
+__attribute__((weak)) int weak_esp_ble_mesh_gap_event_cb(struct ble_gap_event *event, void *arg) {
+    return esp_ble_mesh_gap_event_cb(event, arg);
+}
+
+int esp_ble_mesh_gap_event_cb(struct ble_gap_event *event, void *arg)
 {
     struct ble_gap_disc_desc *desc;
 #if (CONFIG_BLE_MESH_PROVISIONER && CONFIG_BLE_MESH_PB_GATT) || \
@@ -445,7 +450,7 @@ static int disc_cb(struct ble_gap_event *event, void *arg)
             }
         }
         if (!bt_mesh_atomic_test_bit(bt_mesh_dev.flags, BLE_MESH_DEV_SCANNING)) {
-            rc = ble_gap_disc(BLE_OWN_ADDR_PUBLIC, BLE_HS_FOREVER, &scan_param, disc_cb, NULL);
+            rc = ble_gap_disc(BLE_OWN_ADDR_PUBLIC, BLE_HS_FOREVER, &scan_param, weak_esp_ble_mesh_gap_event_cb, NULL);
             if (rc != 0) {
                 BT_ERR("Invalid scan status %d", rc);
                 break;
@@ -596,7 +601,7 @@ static int start_le_scan(uint8_t scan_type, uint16_t interval, uint16_t window,
     } else {
         scan_param.passive = 0;
[... 172 lines cut ...]
    rc = ble_gatts_add_svcs(gatt_svr_svcs);
    if (rc != 0) {
        return;
    }
}

void  ble_on_sync_proc() {
    rc = esp_ble_mesh_init(&provision, &composition);
    if (rc != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize mesh stack (err %d)", rc);
        return;
    }

#if MYNEWT_VAL(BLE_EXT_ADV)
    int err = ble_gap_ext_adv_configure(YOUR_ADV_IDX, &params, NULL,
                                weak_esp_ble_mesh_gap_event_cb, NULL);
    // ble_gap_ext_adv_set_data
    // ble_gap_ext_adv_start
#else
    esp_ble_mesh_start_ble_advertising(&adv_param, &adv_data);
#endif

    esp_ble_mesh_node_prov_enable((esp_ble_mesh_prov_bearer_t)(ESP_BLE_MESH_PROV_ADV|ESP_BLE_MESH_PROV_GATT));
}
```

buganini, 2024-10-29:
Here is a new patch to allow GATT Proxy + ESP-BLE-MESH working together, with less intrusive to the original code, but more flexible.

# Tested scenarios:
* Provisioned with iOS nRF Mesh app
* Send to nrf52840 with NCS v2.7.0
* Recv from nrf52840 with NCS v2.7.0

# Changes
* Add a weak esp_ble_mesh_init_gatts to allow injection of extra services
* Support extended advertising
* Add a weak wrapper `weak_esp_ble_mesh_gap_event_cb` for `esp_ble_mesh_gap_event_cb` (original `disc_cb`) to allow injection of gap event handling
## Changes to previous patch
* Handle `BLE_GAP_EVENT_EXT_DISC` in `disc_cb` for receiving mesh packets
* Properly set .connectable and .scannable in adv params for mesh advertising

# Patches
``` diff
diff --git a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
index 1cb239a3ca..3f038ece24 100644
--- a/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
+++ b/components/bt/esp_ble_mesh/core/include/mesh/adapter.h
@@ -20,6 +20,8 @@
 extern "C" {
 #endif
 
+#define BLE_MESH_EXT_ADV_IDX 0
+
 /* BLE Mesh Max Connection Count */
 #ifdef CONFIG_BT_BLUEDROID_ENABLED
 #define BLE_MESH_MAX_CONN   CONFIG_BT_ACL_CONNECTIONS
@@ -799,6 +801,10 @@ int bt_mesh_encrypt_le(const uint8_t key[16], const uint8_t plaintext[16],
 int bt_mesh_encrypt_be(const uint8_t key[16], const uint8_t plaintext[16],
                        uint8_t enc_data[16]);
 
+int weak_esp_ble_mesh_gap_event_cb(struct ble_gap_event *event, void *arg);
+int esp_ble_mesh_gap_event_cb(struct ble_gap_event *event, void *arg);
+void esp_ble_mesh_init_gatts(void);
+
 enum {
     BLE_MESH_EXCEP_LIST_SUB_CODE_ADD = 0,
     BLE_MESH_EXCEP_LIST_SUB_CODE_REMOVE,
diff --git a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
index 0418b76c54..a08838f923 100644
--- a/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
+++ b/components/bt/esp_ble_mesh/core/nimble_host/adapter.c
@@ -17,6 +17,7 @@
 #include "host/ble_hs.h"
 #include "host/ble_uuid.h"
 #include "host/ble_att.h"
+#include "host/ble_gap.h"
 #include "host/ble_gatt.h"
 #include "services/gap/ble_svc_gap.h"
 #include "services/gatt/ble_svc_gatt.h"
@@ -424,6 +425,20 @@ static int disc_cb(struct ble_gap_event *event, void *arg)
         }
         break;
     }
+#if MYNEWT_VAL(BLE_EXT_ADV)
+    case BLE_GAP_EVENT_EXT_DISC: {
+        struct net_buf_simple buf = {0};
+
+        struct ble_gap_ext_disc_desc * desc = &event->ext_disc;
+        net_buf_simple_init_with_data(&buf, (void *)desc->data, desc->length_data);
+
+        if (bt_mesh_scan_dev_found_cb) {
+            /* TODO: Support Scan Response data length for NimBLE host */
+            bt_mesh_scan_dev_found_cb((bt_mesh_addr_t *)&desc->addr, desc->rssi, desc->legacy_event_type, &buf, 0);
+        }
+        break;
+    }
[... 202 lines cut ...]
    rc = ble_gatts_add_svcs(gatt_svr_svcs);
    if (rc != 0) {
        return;
    }
}

void  ble_on_sync_proc() {
    rc = esp_ble_mesh_init(&provision, &composition);
    if (rc != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize mesh stack (err %d)", rc);
        return;
    }

#if MYNEWT_VAL(BLE_EXT_ADV)
    int err = ble_gap_ext_adv_configure(YOUR_ADV_IDX, &params, NULL,
                                weak_esp_ble_mesh_gap_event_cb, NULL);
    // ble_gap_ext_adv_set_data
    // ble_gap_ext_adv_start
#else
    esp_ble_mesh_start_ble_advertising(&adv_param, &adv_data);
#endif

    esp_ble_mesh_node_prov_enable((esp_ble_mesh_prov_bearer_t)(ESP_BLE_MESH_PROV_ADV|ESP_BLE_MESH_PROV_GATT));
}
```