espressif/esp-idf #14414: Flash Encrypted FW : Issue - "A fatal error occurred: Failed to connect to ESP32-S3: Invalid head of packet (0x80)"

venkateshgr-ayla, 2024-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

INFO: output: Writing at 0x0015dc00... (26 %)
INFO: output: Writing at 0x0015e000... (26 %)
INFO: output: Writing at 0x0015e400... (26 %)
INFO: output: 
INFO: output: A fatal error occurred: Invalid head of packet (0x80): Possible serial noise or corruption.
INFO: -------- failure logs end --------
WARNING: failures: 1, good: 0

We this issue during everytime flashing and each time the percentage is different like 8%, 70%.. etc..
And the "Writing at 0x000xxxxxx" is different everytime.

dobairoland, 2024-08-21:
This is probably a hardware issue (e.g. unstable power supply, bad USB cable) as the error message suggests because the issue happens at different moments during flashing. Enabling the `--trace` option while invoking esptool.py directly can give a little more information (especially what is the content of the receive buffer when that invalid head was identified). Also observing the state of the ESP can be useful. It is possible that it is reseted during the flashing process.

venkateshgr-ayla, 2024-08-21:
@JiaLWang @dobairoland am using the standard ESP32S3 devkit here. Should we still need to worry about the power or hardware issue? We don't see any issue to flash when we are flashing normally and no encryptions like the development and release packages.

dobairoland, 2024-08-21:
> the standard ESP32S3 devkit here

If you mean by standard the ones produced by Espressif then I'm not aware of any HW issue with those. Although you could share the revision number and the exact board you are using.

> We don't see any issue to flash when we are flashing normally

The security team will also take a look at this issue. I'm not aware of any such problem.

venkateshgr-ayla, 2024-08-22:
Can anybody share any sample script or command to create development mode and release mode encrypted package and flash successfully ?

AdityaHPatwardhan, 2024-09-04:
Hi @venkateshgr-ayla,
What do you mean by development/release mode encrypted package ?
Do you mean the encrypted flash binaries?
You can generate an encrypted binary with help of following command
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/flash-encryption.html#manually-encrypting-files

I hope this helps

AdityaHPatwardhan, 2024-09-17:
Hi @venkateshgr-ayla , I am closing the issue since it seems to be resolved.
Please feel free to re-open if required.
Thanks

closed 2024-09-17 by AdityaHPatwardhan