espressif/esp-idf #14417: What happened to "ESP_GATTC_GET_CHAR_EVT"?

gcstr, 2024-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

What happened to "ESP_GATTC_GET_CHAR_EVT".
The examples and the documentation are still pointing to this event, but the header file does not have this definition?

Which event can replace it?

esp-zhp, 2024-08-24:
[image]
In the latest code, this API and the associated events have been deprecated. Please use alternative APIs to achieve your functionality. We will update the documentation as soon as possible.

To implement the required functionality, please use the latest API esp_ble_gattc_get_char_by_uuid. Refer to the example below for guidance.
https://github.com/espressif/esp-idf/blob/dbce23f8a449eb436b0b574726fe6ce9a6df67cc/examples/bluetooth/bluedroid/ble/gattc_multi_connect/main/gattc_multi_connect.c#L212

esp-zhp, 2024-09-10:
I have updated the document 86713a6946cf91c9d46eeb73935da39884efec13

Indastri, 2024-09-11:
Fix was provided. Closing this issue. If you will encounter any further problems, reopen this or create a new issue.

Cheers!

closed 2024-09-11 by Indastri