espressif/esp-idf #14420: Dynamic mesh root node switching

TroyMitchell911, 2024-08-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Since I have enabled the dynamic allocation network,how should I know that the root node has been switched so that I can do some processing for the old root node and make the new root node take some actions. Which event can the old root node use to know that it is about to be switched? Through which event does the new root node know that it has become a new root node?

Best regard.

zhangyanjiaoesp, 2024-08-22:
@TroyMitchell911 

> Since I have enabled the dynamic allocation network

Do you mean that you are using a self-organized mesh network?  

In a self-organized network, once a root is established, a root switch generally does not occur. The switch only happens in the following three situations:

1. Root Conflict: If there is a root conflict and the network does not allow multiple roots, the conflict can be detected through `MESH_EVENT_ROOT_ASKED_YIELD` events.
2. Active Call by Root Node: When the root node actively calls `esp_mesh_waive_root`.
3. Sudden Departure of Root Node: If the root node suddenly leaves, second-layer nodes will re-elect a new root.

You can use `esp_mesh_is_root()` to check whether you are the root node.

TroyMitchell911, 2024-08-22:
> @TroyMitchell911
>
>
> Do you mean that you are using a self-organized mesh network?
>
> In a self-organized network, once a root is established, a root switch generally does not occur. The switch only [...]

Yes, my question is this. What I want to ask is, if the root node changes (what case changes are not within the scope of our discussion), then which event can know that the root node is about to change? Which event can know who the new root node is?

zhangyanjiaoesp, 2024-08-23:
@TroyMitchell911 
[image]

The `MESH_EVENT_ROOT_ADDRESS` can show the address of the root node.
And you can call `esp_mesh_is_root()` to check whether the node is root or not.

zhangyanjiaoesp, 2024-11-12:
@TroyMitchell911 can we close this issue?

closed 2024-11-12 by TroyMitchell911