espressif/esp-idf #14421: Status of nvs on (external) spi-flash

Teesmo, 2024-08-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am working on an application that would store time-series entries, that I would like to query (prompting from a web-based application) using a key, or index associated to the data I want to access. The data has to be persisted over a very long period (>10 years), and new entries would be generated every 30 minutes. My initial thoughts are, that using nvs to store this data would be ideal, but nvs is currently confined to internal flash. The latest documentation suggests that a future implementation of nvs may support writing to (external) spi-flash. What is the current status of this development?

igrr, 2024-08-22:
Addition to @KonstantinKondrashov's comment:

You can follow the approach of ext_flash_fatfs example to connect the external flash and define a partition there:

https://github.com/espressif/esp-idf/blob/dbce23f8a449eb436b0b574726fe6ce9a6df67cc/examples/storage/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c#L65-L73

Then, instead of mounting FATFS, you can initialize NVS in the partition you have defined, using [`nvs_open_from_partition`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#_CPPv423nvs_open_from_partitionPKcPKc15nvs_open_mode_tP12nvs_handle_t) function, similar to Konstantin's example above.

---

That being said, I think NVS is not going to work for this use case. If you store one key-value pair every 30 minutes, over 10 years you will generate around 175200 key-value pairs, or 1390 flash sectors, or around 5561 kB. Such a large NVS partition will require a lot of time to initialize. Additionally, according to this RAM usage estimate from the docs ([Large Amount of Data in NVS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#large-amount-of-data-in-nvs),

> For RAM usage estimation, please use the following approximate figures: each 1 MB of NVS flash partition consumes 22 KB of RAM and each 1000 keys consumes 5.5 KB of RAM.

Near the end of the 10 year period, the RAM usage of NVS may reach >100kB. Newer IDF versions have an option to allocate NVS cache in PSRAM, so this extra RAM usage may or may not be an issue, depending on whether you are using PSRAM or not. However, the initialization time of NVS for such a partition will still be quite long:

> Initialization of NVS requires approximately 0.5 seconds per 1000 keys.

175200 * 0.5 / 1000 = 87.6 seconds.

Therefor I think for your use case a either conventional filesystem or a spec might be more suitable.

Teesmo, 2024-08-23:
Thank you for the insights. Are you suggesting that I should rather use FATFS/LittleFS as it would work faster? I should be honest, I don't how file search works in filesystems (I know very little about filesystems, in fact).

closed 2025-03-17 by Teesmo