espressif/esp-idf #14423: _xt_nmi can't be overriden

JimDrewGH, 2024-08-22:
### Board

All

### Device Description

ESP32S2 Dev Module, ESP32S3 Dev Module

### Hardware Configuration

Single GPIO input on GIO06

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10 Pro

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Using the normal method for overriding the weak casting for interrupts does not work for the xt_nmi interrupt. 

When compiling the code there are no errors, but the _xt_nmi handler is not replaced by my code.  I know this because I omitted the normal 'rfi' instruction which is required to exit the interrupt handler.  This should make the ESP32 crash if the interrupt is triggered.  There is no crash.  If I change the interrupt from _xt_nmi to _xt_highint5 and omit that instruction, the ESP32 does crash as expected, and using this code works perfectly for highint4 or highint5.  So clearly the _xt_nmi weak casting is not being overridden.

### Sketch

```cpp
C++ code (test.ino):

#include "soc/periph_defs.h"

#define GPIO_INPUT_IO_0 6                               // GPIO 6 as the input
#define GPIO_INPUT_PIN_SEL (1ULL<<GPIO_INPUT_IO_0)

#define INTR_NUM 14                                     // NMI

volatile uint32_t intCounter = 0;

void setup() {
  Serial.begin(115200);
  gpio_config_t io_conf;
  io_conf.intr_type = GPIO_INTR_NEGEDGE;	            // interrupt of falling edge
  io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;	        // bit mask of the pin(s)
  io_conf.mode = GPIO_MODE_INPUT;                       // set as input mode
  io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;         // disable pull-down mode
  io_conf.pull_up_en = GPIO_PULLUP_ENABLE;	            // enable pull-up mode
  gpio_config(&io_conf);

  ESP_INTR_DISABLE(INTR_NUM);                           // disable interrupt 14
  esp_intr_alloc(ETS_GPIO_INTR_SOURCE, ESP_INTR_FLAG_NMI|ESP_INTR_FLAG_IRAM, NULL, NULL, NULL);
  ESP_INTR_ENABLE(INTR_NUM);                            // enable interrupt 14
}

void loop() {
  Serial.println(intCounter);
  delay(100);
}

Assembly code file (intcode.S):

#include <xtensa/coreasm.h>
#include <xtensa/corebits.h>
#include <xtensa/config/system.h>
#include "freertos/xtensa_context.h"
#include "esp_private/panic_reason.h"
#include "sdkconfig.h"
#include "soc/soc.h"
#include "soc/gpio_reg.h"
#include "soc/dport_reg.h"

#define TIMR_ADDR_LO 0x6001f004                         // (unsigned int*)TIMG_T0LO_REG(0); 
#define TIMR_ADDR_HI 0x6001f008 
#define TIMR_ADDR_UPD 0x6001f00c

#define NMI_INTR_STACK_SIZE 12                           // size of stack space we need
#define LX_INTR_A15_OFFSET 0                            // 1st long is A15
#define LX_INTR_A14_OFFSET 4                            // 2nd long is A14 
#define LX_INTR_A13_OFFSET 8                            // 3rd long is A13

.data
_nmi_intr_stack:
.space    NMI_INTR_STACK_SIZE

.section  .iram1,"ax"
.global   xt_nmi
.type     xt_nmi,@function
.align    4

// start of ISR
xt_nmi:                                            // entry point into interrupt handler

// Save registers - a0 saved in ISR entry

  movi  a0, _nmi_intr_stack                              // get our stack pointer
  s32i  a15, a0, LX_INTR_A15_OFFSET                     // save a15 on our stack
  s32i  a14, a0, LX_INTR_A14_OFFSET                     // save a14 on our stack
  s32i  a13, a0, LX_INTR_A13_OFFSET                     // save a13 on our stack

  movi  a14, TIMR_ADDR_UPD                              // get timer 0 update address into a14
  s32i  a0, a14, 0                                      // any write instruction to the timer update register latches the timer onto the timer registers
  movi  a14, timer0_value                               // get address of timer0_value from main program
  movi  a15, TIMR_ADDR_LO                               // get timer0 low address into a15
  l32i  a13, a15, 0                                     // get the lower 32 bits of the timer value
  s32i  a13, a14, 0                                     // store in timer0_value

/* Clear interrupt flag for input */
  movi  a14, GPIO_STATUS_W1TC_REG                       // address of interrupt flags
  movi  a15, (1 << 6)                                   // mask for GPIO 6
  s32i  a15, a14, 0                                     // store mask into interrupt flags
  memw                                                  // write through

/* Increment intr_cnt */
  movi  a14, intCounter                                 // get address of incCounter variable from main program
  l32i  a15, a14, 0                                     // get value from variable
  addi  a15, a15, 1                                     // add 1 to value
  s32i  a15, a14, 0                                     // put value back into variable
  memw                                                  // write through

/* Restore registers */
  l32i  a13, a0, LX_INTR_A13_OFFSET                     // restore a13
  l32i  a14, a0, LX_INTR_A14_OFFSET                     // restore a14
  l32i  a15, a0, LX_INTR_A15_OFFSET                     // restore a15
  
  rsr a0, EXCSAVE_5                                     // restore a0
  rfi   5                                               // exist ISR

.global ld_include_xt_nmi                      // force compiler to override weak inclusion
ld_include_xt_nmi:
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

Take the above code and create a C++ app and also an assembly file with the name having the suffix of '.S'.  I call the main app **test.ino** and the assembly file **intcode.S**, but you can use any names you like.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

KonstantinKondrashov, 2024-08-22:
Hi @me-no-dev @JimDrewGH!
There is the same issue https://github.com/espressif/esp-idf/issues/13629. Its status is a work in progress. It will be fixed soon. You can try this patch - [32767.patch](https://github.com/user-attachments/files/16714221/32767.patch). Thanks.

JimDrewGH, 2024-08-22:
@KonstantinKondrashov Unfortunately, that may not help anyone using the Arduino core (based on v4.x or v5.1.1 of the ESP-IDF).  How are patch files suppose to be used?  Maybe I can extract whatever changes are necessary from the patch file and apply them to the Arduino version?

KonstantinKondrashov, 2024-08-23:
Sorry, I have not noticed that it is for the Arduino version. It looks like you can not apply this patch and have to wait for it to be fixed in IDF and then updated the Arduino version.

JimDrewGH, 2024-08-26:
Well, at least it's a known issue.  :)

SoucheSouche, 2024-09-04:
Hi @JimDrewGH , the issue was fixed and merged internally. It will be available with the next github sync.

AxelLin, 2024-09-24:
> Hi @JimDrewGH , the issue was fixed and merged internally. It will be available with the next github sync.

I cannot find this fix in recent github sync, is this fixed?

JimDrewGH, 2024-09-24:
I missed the fact that this has been fixed.   Thanks!

JimDrewGH, 2024-09-24:
BTW, if anyone is following this, the code above only works with highint5 interrupt until you change the last two lines to use the proper EXCSAVE_7 and rti 7, which are required for NMI.  I forgot to point that out initially.

o-marshmallow, 2024-09-25:
This has been fixed in the commit: https://github.com/espressif/esp-idf/commit/928859307f92faf131f1d55ce0f8a0d6d6784b54

closed 2024-09-25 by o-marshmallow