espressif/esp-idf #14425: 基于VS CODE插件使用ESP32-S2在IDF5.3配置SDK选项，异常的出现了蓝牙的选项，并且可以勾选"When configuring SDK options for ESP32-S2 in IDF5.3 using the VS CODE plugin, an unusual Bluetooth option appeared and could be selected."

caidamao, 2024-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF5.3

### Espressif SoC revision.

ESP32-S2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

没有蓝牙选项
No Bluetooth option available

### What is the actual behavior?

出现了蓝牙选项，且可以勾选。且勾选后编译的文件由1258个增加到1523个，证明实际上蓝牙的文件真的被编译到了，但显然S2它没有蓝牙。
"The Bluetooth option appeared and can be selected. After selecting it, the number of compiled files increased from 1258 to 1523, proving that the Bluetooth files were indeed compiled in. However, it is apparent that the ESP32-S2 does not actually have Bluetooth capabilities."

### Steps to reproduce.

1. 更新到IDF5.3
2. 用VS CODE插件打开一个idf5.22的esp32-s2的工程
3. 开启SDK配置，勾选蓝牙选项
4.保存并编译
5.出现此问题

1.Update to IDF5.3
2. Open an idf5.22 esp32-s2 project with VS CODE plugin
3. Open the SDK configuration and check the Bluetooth option
4. Save and compile
5. This problem occurs

caidamao, 2024-08-23:
[image]

caidamao, 2024-08-23:
我尝试回到了5.2.2，没有复现这个问题，再次切换到5.3，问题复现

brianignacio5, 2024-08-23:
Could you share the output from here:

Click menu `View` ->  `Output` ->  Choose from dropdown the `ESP-IDF` option ?

caidamao, 2024-08-23:
> Could you share the output from here:
> 
> Click menu `View` -> `Output` -> Choose from dropdown the `ESP-IDF` option ?

[Uploading bug log.txt…]()

caidamao, 2024-08-23:
> Could you share the output from here:
> 
> Click menu `View` -> `Output` -> Choose from dropdown the `ESP-IDF` option ?

[image]

caidamao, 2024-08-23:
> Could you share the output from here:
> 
> Click menu `View` -> `Output` -> Choose from dropdown the `ESP-IDF` option ?

brianignacio5, 2024-08-23:
I believe the output you share is from the terminal output. In the bottom bar you can see that there is a tab called `OUTPUT` which has a dropdown in the right side. From there choose the `ESP-IDF`. I want to see if there is an error when you set the bluetooth enable or not.

caidamao, 2024-08-23:
> Could you share the output from here:
> 
> Click menu `View` -> `Output` -> Choose from dropdown the `ESP-IDF` option ?

[bug log.zip](https://github.com/user-attachments/files/16726908/bug.log.zip)

caidamao, 2024-08-23:
> I believe the output you share is from the terminal output. In the bottom bar you can see that there is a tab called `OUTPUT` which has a dropdown in the right side. From there choose the `ESP-IDF`. I want to see if there is an error when you set the bluetooth enable or not.

I know that it's just because of the network problems in China that I haven't managed to upload the compressed log package until now after sending the screenshot.

brianignacio5, 2024-08-23:
Thank you.

So the issue is that the bluetooth enable KConfig setting (the code behind idf.py menuconfig) is not dependent of IDF_TARGET (esp32s2) so you can enable bluetooth regardless. Not really a esp-idf vscode extension issue here.

caidamao, 2024-08-23:
I remember that these options in the figure should not be enabled by default. Is it possible that there are issues beyond just the Bluetooth option? I updated from IDF 5.2.2 to IDF 5.3, and apart from the SDK update, there were no other changes made, but the memory consumption increased by several kilobytes.

> Thank you.
> 
> So the issue is that the bluetooth enable KConfig setting (the code behind `idf.py menuconfig`) is not dependent of IDF_TARGET (esp32s2) so you can enable bluetooth regardless. Not really a esp-idf vscode extension issue here.

esp-zhp, 2024-08-26:
@caidamao 
1-对于不支持 BT 的芯片，我们将尝试隐藏 BT_ENABLED 配置项，并设置为 diable 状态。
2-你有提到 “从 IDF 5.2.2 更新到了 IDF 5.3，除了 SDK 更新之外，没有其他变化，但是内存占用增加了好几千字节。”我想确定一下：如果将 BT_ENABLED  关闭，还会增加内存吗？

esp-zhp, 2024-08-26:
@caidamao 
BT_ENABLED 我这面显示在 v5.2.2 和 v5.3 默认都是关着的。
你提到 “使用ESP32-S2在IDF5.3配置SDK选项，异常的出现了蓝牙的选项”，这不是异常，因为 BT 分为 controller 和 host，ESP32-S2 不支持 BT，准确说只是不支持 controller，而对于 host，这是纯软件做的，还是可以支持的。所以，该配置项能够打开。

caidamao, 2024-08-27:
> @caidamao BT_ENABLED 我这面显示在 v5.2.2 和 v5.3 默认都是关着的。 你提到 “使用ESP32-S2在IDF5.3配置SDK选项，异常的出现了蓝牙的选项”，这不是异常，因为 BT 分为 controller 和 host，ESP32-S2 不支持 BT，准确说只是不支持 controller，而对于 host，这是纯软件做的，还是可以支持的。所以，该配置项能够打开。

[image]
你看这个，5.22我重新拉了一个例程出来，也有蓝牙选项。我记得以前没有吧？S2有蓝牙选项，真的有点怪.

caidamao, 2024-08-27:
> @caidamao 你有提到 “从 IDF 5.2.2 更新到了 IDF 5.3，除了 SDK 更新之外，没有其他变化，但是内存占用增加了好几千字节。”我想确定一下：如果将 BT_ENABLED 关闭，还会增加内存吗？

这个是基于5.22编译的
[image]

caidamao, 2024-08-27:
> @caidamao 你有提到 “从 IDF 5.2.2 更新到了 IDF 5.3，除了 SDK 更新之外，没有其他变化，但是内存占用增加了好几千字节。”我想确定一下：如果将 BT_ENABLED 关闭，还会增加内存吗？

这是基于5.3编译的：可以看到内存使用量多了一点。
[image]

caidamao, 2024-08-27:
[5.3spi_lcd_touch.zip](https://github.com/user-attachments/files/16755162/5.3spi_lcd_touch.zip)
[5.22spi_lcd_touch.zip](https://github.com/user-attachments/files/16755163/5.22spi_lcd_touch.zip)
这两个压缩包是直接从5.2.2例程里拉取的，我手里的一个工程更明显，但因为保密原则我没办法展示
@esp-zhp

esp-zhp, 2024-08-27:
@caidamao 
你认为关闭 BT_ENABLED 后，还有 BT 的内容被遍进来吗？我是负责的 BLE 模块的，从你的反馈来看：即使 BT_ENABLED 关闭了，使用使用的内存还是增加。
这应该是其他模块的导致的。
你能够确定是那个模块导致的内存增加吗，我让对应模块的同事来支持你的问题。

caidamao, 2024-08-27:
> @caidamao 你认为关闭 BT_ENABLED 后，还有 BT 的内容被遍进来吗？我是负责的 BLE 模块的，从你的反馈来看：即使 BT_ENABLED 关闭了，使用使用的内存还是增加。 这应该是其他模块的导致的。 你能够确定是那个模块导致的内存增加吗，我让对应模块的同事来支持你的问题。

我觉得应该是跟蓝牙相关的模块无关（没有使能蓝牙的时候，蓝牙相关的代码都没有被编译进去），我目前有确认都有用到的就SPI。好像是有很多外设的旧代码选项有出现在sdk配置里，但我很难确定跟什么有关。

ginkgm, 2025-03-23:
看了一下数据，其实2KB 的 flash size 增加不算是特别大的变化，一些辅助debug的打印log都有可能造成这个增加。请问有其他更明确更严重的案例吗？

caidamao, 2025-03-31:
> [@caidamao](https://github.com/caidamao) 您好，有更多更新么？谢谢

目前没有更多的进展，我的工程目前已经往5.4以上版本的SDK迁移了。
补充信息，在5.3版本的时候，我使用相同的代码，WiFi跟蓝牙同时启用，出现了调用BLE通知堆栈溢出的问题。

caidamao, 2025-03-31:
> > 看了一下数据，其实2KB 的 flash size 增加不算是特别大的变化，一些辅助debug的打印log都有可能造成这个增加。请问有其他更明确更严重的案例吗？
> 
> 5.3.1应该这个版本，相同代码编译出来的固件，有堆栈溢出导致重启的问题。调用了ble通知。但是目前我已经迁移到了5.4的sdk了，扩大了任务堆栈，已经临时解决了该问题

注意，这个的目标芯片是ESP32而非S2，S2没有出现堆栈移除的现象。

ginkgm, 2025-03-31:
Hi @caidamao 

如果没有其他报告的话，我们驱动这边关于size的问题就关闭了。

请问 BLE + wifi 堆栈溢出的问题是否有更详细的报告？具体是什么样的应用代码，堆栈大小多少的情况下造成了溢出呢？

caidamao, 2025-03-31:
> Hi [@caidamao](https://github.com/caidamao)
> 
> 如果没有其他报告的话，我们驱动这边关于size的问题就关闭了。
> 
> 请问 BLE + wifi 堆栈溢出的问题是否有更详细的报告？具体是什么样的应用代码，堆栈大小多少的情况下造成了溢出呢？

针对size的问题可以先关闭吧。
任务堆栈是2kb，我回头有空了搞个伪代码，看看能不能复现这个现象。如果可以，我重新提个新问题。

ginkgm, 2025-03-31:
感谢！ @caidamao

closed 2025-03-31 by ginkgm