espressif/esp-idf #14426: i want only use ble scan adv name,but the libbtdm_app.a is to large ,iram used is high.how can i optimizated IRAM used

Wangmale, 2024-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

chip:esp32
idf version:4.4.4

Wangmale, 2024-08-23:
i see https://docs.espressif.com/projects/esp-idf/zh_CN/v5.3/esp32/api-guides/performance/ram-usage.html ,but it not give how to optimizate bt IRAM used

Wangmale, 2024-08-23:
Archive File DRAM .data & 0.bss .rtc_noinit IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
          libchint_net.a          4     182           0           0           0          186       23605    295530                0          0      319139
                 libbt.a       4690     397           0        1684           0         6771      129751     10449                0          0      146574
           libbtdm_app.a        780    2648           0       28798           0        32226       79295      5147                0          0      114020

Weijian-Espressif, 2024-08-27:
The esp32 Bluetooth interrupt processing function is placed in IRAM. It also needs to respond to interrupts in time when erasing and writing the flash, so it must be placed in IRAM, which cannot be saved now.

Weijian-Espressif, 2024-10-23:
@Wangmale  If you only use Classic Bluetooth or BLE, you can try this commit:5c5ab4b0c9e5295dd529f82f2c88d4b2e21a0e81

closed 2025-03-08 by Alvin1Zhang