espressif/esp-idf #14429: FreeRTOS scheduler does not preempt on Linux target

snake-4, 2024-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-2004-g8e4454b285

### Espressif SoC revision.

Linux

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

None

### Power Supply used.

USB

### What is the expected behavior?

Scheduler should preempt all tasks since the option is enabled by default in ESP-IDF.

### What is the actual behavior?

The scheduler doesn't seem to preempt tasks with busy loops on the Linux target. This results in most components locking up the system.

### Steps to reproduce.

- main/app_main.c
```c
#include <freertos/FreeRTOS.h>
#include <freertos/timers.h>
#include <freertos/task.h>
#include <stdio.h>

static void timer_cb(TimerHandle_t) {
    puts("timer_cb called");
}

static void task_cb(void *) {
    for (;;) {}
}

void app_main(void) {
    xTimerStart(xTimerCreate("Timer", pdMS_TO_TICKS(500), pdTRUE, NULL, timer_cb), 0);
    vTaskDelay(pdMS_TO_TICKS(2000));

    xTaskCreate(task_cb, "Task", 1024, NULL, tskIDLE_PRIORITY, NULL);
    vTaskDelay(pdMS_TO_TICKS(2000));
}
```

- main/CMakeLists.txt
```
idf_component_register(SRCS "app_main.c")
```

- CMakeLists.txt
```
cmake_minimum_required(VERSION 3.16)
set(COMPONENTS main)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(firmware)
```

Observe how the timer stops getting called after the task is created.

### More Information.

My previous bug report #14395 was likely caused by this issue.

louielauanu, 2024-08-28:
Hi @snake-4,

I pasted your code and tried to reproduce the issue at my end using Linux target. I slightly modified the `task_cb` to make it print every 1 ms, and here is my output:
``` I (874767564) port: Starting scheduler.
timer_cb called
timer_cb called
timer_cb called
timer_cb called
task executes
timer_cb called
timer_cb called
task executes
timer_cb called
timer_cb called
task executes
timer_cb called
timer_cb called
task executes
timer_cb called
timer_cb called
task executes
timer_cb called
timer_cb called
task executes```

I don't see timer stopped getting called when the task is scheduled. Do you mean that at your end timer output disappears? Can you share your log output?

Thanks,
Xiaoyu

snake-4, 2024-08-28:
Sorry, this report might've been in error as I also noticed that it works fine. Maybe the behavior under GDB was different due to signal configuration.

closed 2024-08-28 by snake-4