espressif/esp-idf #14430: add_dependencies early in build

malachib, 2024-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am working on a yaml generator.  Its purpose is to work around the lack of local components ability to specify their own `idf_component.yml`.

It is working pretty well, but runs too late in the process.  Although it runs before the components themselves are compiled, it runs after the component manager evaluates `idf_component.yml`.

It's very useful to run in the context of ESP-IDF build so that things like `EXTRA_COMPONENT_DIRS` is available.

Is there a way to `add_dependencies` such that my custom target is picked up by idf build early on, before managed components are evaluated?

kumekay, 2024-09-02:
Hello @malachib, to help us better understand the problem, could you please elaborate why is it important in your case to add dependencies before idf_component.yml is processed and why is it too late otherwise? Maybe you have an example?

malachib, 2024-09-02:
@kumekay Yes, gladly

Consider 3 components:

1. `main` component, primary firmware itself
2. `lib1` component, consumed directly by `main`, a git submodule within `main`
3. `lib2` component, consumed directly by `lib1`, a git submodule within `lib1`

As it stands, `idf.py` offers `lib1` and `lib2` no provision for an `idf_component.yml`.  Only `main` has that option.
I have a mechanism to scan `lib1` and `lib2` to populate `idf_component.yml` in `main`.

This is working.  However, it happens too late in the build process for `idf.py` managed component mechanism to notice.

I can brute force it with `execute_process` but that bypasses the whole nice CMake target/dependency chain and additionally might exclude me from picking up things like `EXTRA_COMPONENTS_DIR`