espressif/esp-idf #14431: Access to pulse_sig/ctrl_sig of pulse counter channel is needed for gpio-mods

gin66, 2024-08-23:
### Is your feature request related to a problem?

yes. I am the author of FastAccelStepper and use a pulse counter linked to the stepper outputs for two reasons: test and realtime pulse information. In order to make this work, the pulse counter module needs to be linked to the stepper pin outputs (step/direction) using gpio matrix/iomux modifications. The needed info for this: unit and channel is stored in private structs. So this cripples the possibilities of the esp32 product line

### Describe the solution you'd like.

getter function for signal and control info for a given pcnt_unit

### Describe alternatives you've considered.

Currently I mirror part of the private struct and derive the infos from the handles. 
Implementation for reference: [work in progress](src/FastAccelStepper_idf5_esp32_pcnt.cpp)

Kainarx, 2025-01-06:
Sorry for the late reply, we improved the gpio configuration in the driver in IDF5.4. Subsequent configuration of gpio will not overwrite the previous configuration. So you can configure gpio as the output of the stepper motor and then as the input of the pcnt. Hope this can solve your problem.

gin66, 2025-01-06:
Thanks for the late feedback.

The main issue is to find out, which pulse counter channel has been allocated by the espidf driver.  Without this info, the gpio configuration cannot be done. Unfortunately the link above does not work anymore. The respective code part is:

```
// Why the hell, does espressif think, that the unit and channel id are not
// needed ? Without unit/channel ID, the needed parameter for
// gpio_matrix_in/gpio_iomux_in cannot be derived.
//
// Here we declare the private pcnt_chan_t structure, which is not save.
struct pcnt_unit_t {
  /*pcnt_group_t*/ void *group;
  portMUX_TYPE spinlock;
  int unit_id;
  // remainder of struct not needed
};
struct pcnt_chan_t {
  pcnt_unit_t *unit;
  int channel_id;
  // remainder of struct not needed
};
```

Now either the unit and channel can be provided, or the actual needed signal and control info for the gpio matrix configuration.

Kainarx, 2025-01-06:
You no longer need to configure gpio by yourself, [this](https://github.com/espressif/esp-idf/blob/release/v5.4/components/esp_driver_mcpwm/test_apps/mcpwm/main/test_mcpwm_gen.c#L308) is an example of using the output gpio of mcpwm generator, as the input gpio of mcpwm capture. I think pcnt is similar

gin66, 2025-01-06:
Thanks for the fast reply. It is not similar, because this is not cross-module usage!

If you first configure either rmt or mcpwm to use GPIO n as output and then want to hook the pcnt to that GPIO n using the pcnt API calls, it will disconnect rmt/mcpwm ! So it is only possible AFAIk to use direct gpio matrix manipulation. And those calls need signal and control info, which is currently not accessible from the pcnt API.

BTW: Using the whole gpio/iomux matrix stuff is more a try and error approach due to lack of documentation. I have it working stable on espidf4 using this sequence:
```
void StepperQueue::connect_mcpwm_pcnt() {
  const struct mapping_s *mapping = (const struct mapping_s *)driver_data;
  mcpwm_unit_t mcpwm_unit = mapping->mcpwm_unit;
  mcpwm_gpio_init(mcpwm_unit, mapping->pwm_output_pin, _step_pin);
  // Doesn't work with gpio_matrix_in
  //  gpio_matrix_in(step_pin, mapping->input_sig_index, false);
  gpio_iomux_in(_step_pin, mapping->input_sig_index);
}
```
I have never found out, why `gpio_matrix` does not work….

On espidf5 this works _with_ `gpio_matrix`:
 ```
  uint8_t step_pin = getStepPin();
  int signal = pcnt_periph_signals.groups[0]
                   .units[unit_id]
                   .channels[channel_id]
                   .pulse_sig;
  gpio_matrix_in(step_pin, signal, 0);
  gpio_iomux_in(step_pin, signal);
  if (dir_pin != PIN_UNDEFINED) {
    pinMode(dir_pin, OUTPUT);
    int control = pcnt_periph_signals.groups[0]
                      .units[unit_id]
                      .channels[channel_id]
                      .control_sig;
    gpio_iomux_out(dir_pin, 0x100, false);
    gpio_matrix_in(dir_pin, control, 0);
    gpio_iomux_in(dir_pin, control);
  }
```
The espidf4 version is:
```
  gpio_matrix_in(step_pin, sig_idx[pcnt_unit], 0);
  gpio_iomux_in(step_pin,
                sig_idx[pcnt_unit]);  // test failure without this call
  if (dir_pin != PIN_UNDEFINED) {
    pinMode(dir_pin, OUTPUT);
    gpio_matrix_out(dir_pin, 0x100, false, false);
    gpio_matrix_in(dir_pin, ctrl_idx[pcnt_unit], 0);
    gpio_iomux_in(dir_pin, ctrl_idx[pcnt_unit]);
  }
```

gin66, 2025-01-06:
my lib is here: https://github.com/gin66/FastAccelStepper/tree/master/src

Kainarx, 2025-01-06:
[0001-test-It_is_just_a_placeholder.PATCH](https://github.com/user-attachments/files/18318538/0001-test-It_is_just_a_placeholder.PATCH) I modified a pcnt test for your reference, never mind the fail of the test,  just because I didn't modify the test value

gin66, 2025-01-06:
Thanks for the patch. Unfortunately I have no idea, how to run it. Have you checked, that mcpwm is generating pulses and the pulse counter counts those ? From my experience I doubt this works.

If I try your approach in my lib, the stepper will not run and the pulse counter will not count, because the mcpwm or rmt have been disconnected by pulse counter init.

Kainarx, 2025-01-06:
Have you updated to IDF 5.4?
[image]
[image]

gin66, 2025-01-07:
Thanks for your efforts. The screenshot is convincing and apparently 5.4 has not these past issues. I will check with espidf5.4 later. If then I still not get it working, I will reopen the comment.

closed 2025-01-07 by gin66