espressif/esp-idf #14433: Internal error in mdns_lookup_selfhosted_service function

nopnop2002, 2024-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

Chip is ESP32D0WDQ6 (revision 1)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 development board

### Power Supply used.

USB

### What is the expected behavior?

mdns_lookup_selfhosted_service function does not report errors

### What is the actual behavior?

mdns_lookup_selfhosted_service function reports error

### Steps to reproduce.

1. Initialize mdns
```
static void initialise_mdns(void)
{
    //get MAC address
    uint8_t mac[6];
    esp_read_mac(mac, ESP_MAC_WIFI_STA);
    char hostname[64];
    sprintf(hostname, "%s-%02X%02X%02X", "esp32", mac[3], mac[4], mac[5]);
    ESP_LOGI(TAG, "mdns hostname set to: [%s]", hostname);

    //initialize mDNS
    ESP_ERROR_CHECK( mdns_init() );
    //set mDNS hostname (required if you want to advertise services)
    ESP_ERROR_CHECK( mdns_hostname_set(hostname) );
    //set default mDNS instance name
    ESP_ERROR_CHECK( mdns_instance_name_set("ESP32 with mDNS") );

    //structure with TXT records
    mdns_txt_item_t serviceTxtData[3] = {
        {"board", "esp32"},
        {"u", "user"},
        {"p", "password"}
    };

    //initialize service
    //ESP_ERROR_CHECK( mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData, 3) );
    ESP_ERROR_CHECK( mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, NULL, 0) );
}
```

2.Call mdns_lookup_selfhosted_service.

```
    mdns_result_t *results = NULL;
    esp_err_t err = mdns_lookup_selfhosted_service(NULL, "_http", "_tcp", 20, &results);
    if (err) {
        ESP_LOGE(TAG, "Lookup Failed: %s", esp_err_to_name(err));
        return;
    }
    if (!results) {
        ESP_LOGW(TAG, "No results found!");
        return;
    }
    mdns_query_results_free(results);
```

### Debug Logs.

```plain
E (64727) mdns: Cannot allocate memory (line: 6186, free heap: 222388 bytes)
```
```

david-cermak, 2024-08-30:
Hi @nopnop2002 

This issue has been fixed in [v1.4](https://github.com/espressif/esp-protocols/releases/tag/mdns-v1.4.0) in https://github.com/espressif/esp-protocols/commit/d4da9cb0

nopnop2002, 2024-08-30:
@david-cermak 

Thank you for your comment.

After updating the component registry version, it started working correctly.

closed 2024-08-30 by nopnop2002