espressif/esp-idf #14434: CONFIG_FREERTOS_HZ and struct "<unnamed>" has no field "left_align" issue

Cimby1, 2024-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Build project like in the other folder.

### What is the actual behavior?

Throw 2 error:

1. identifier "CONFIG_FREERTOS_HZ" is undefined
2. struct "<unnamed>" has no field "left_align"

### Steps to reproduce.

1. Step create any project
2. Step build it
3. Step rename the main.c file to anything and change it format to for example .txt
4. Step create a new main.c, copy and paste a working code here
Now it will throw errors like mine

### Build or installation Logs.

```plain
*  Executing task in folder DAC: c:\ESP_IDF_TOOLS\tools\ninja\1.11.1\ninja.exe  

[1/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\esp_projects\DAC\build\bootloader\esp-idf\esptool_py && c:\ESP_IDF_TOOLS\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/ESP_IDF/v5.3/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/esp_projects/DAC/build/bootloader/bootloader.bin"
Bootloader binary size 0x6880 bytes. 0x780 bytes (7%) free.
[2/4] Linking CXX executable DAC.elf
FAILED: DAC.elf 
cmd.exe /C "cd . && C:\ESP_IDF_TOOLS\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=C:/esp_projects/DAC/build/DAC.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\DAC.elf.rsp -o DAC.elf  && cd ."
C:/ESP_IDF_TOOLS/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(app_startup.c.obj):(.literal.main_task+0x24): undefined reference to `app_main'
C:/ESP_IDF_TOOLS/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(app_startup.c.obj): in function `main_task':
C:/ESP_IDF/v5.3/esp-idf/components/freertos/app_startup.c:199:(.text.main_task+0x99): undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "c:\ESP_IDF_TOOLS\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1.
```

### More Information.

I didn't change my void app_main(void) function name in any code that I tried to copy paste like this. So I guess the suggested problem in the build logs are not valid. 

If I creat a new project usigng the ESP IDF wizard, and paste my code there it works flawlessly.

louielauanu, 2024-08-26:
Hi @Cimby1,

I'm trying to understand and reproduce this issue at my end. I picked up a random project "/esp-idf/examples/system/esp_event/user_event_loops/" and rename main.c to main.txt and create a new main.c, but it still can be compiled.

To clarify the issue and help us check it further, can you please describe what did you modify in your project and show your file hierarchy and a valid complete log?

Thanks,
Xiaoyu

Cimby1, 2024-08-26:
Hi, thanks for helping me out!

In the menuconfig I changed the memory size to 4MB, and to auto detect mode.

For my components I only added ESP-IDF.

It is a weird bug, that is not happening on my side all the time.

I try to build the project on another PC, to see if the issue moves or not.

What kind of log can I attach ? I though the build error contains enough information.

louielauanu, 2024-08-27:
> Hi, thanks for helping me out!
>
> In the menuconfig I changed the memory size to 4MB, and to auto detect mode.
>
> For my components I only added ESP-IDF.
>
> It is a weird bug, that is not happening on my [...]

Hi @Cimby1,

Did you reproduce it in another computer? I followed the steps you pointed that enable flash auto detect and change flash size to 4MB, and also create a renamed txt file, but still didn't hid the same issue.

For the issue signatures  `identifier "CONFIG_FREERTOS_HZ" is undefined`
 and `struct "" has no field "left_align`, I don't find it in the log, and I can't see how flash configurations may affect FreeRTOS. It will be helpful to further debugging if a complete context can be provided.

Regards,
Xiaoyu

Cimby1, 2024-08-29:
Hi, sorry for the late reply I was a little busy with other projects. 
I try to bring back this issue at the weekend.

Cimby1, 2024-09-04:
So after days of trying I solved this problem by rewriting the CMakeLists.txt in the main folder. 

The SRCS section was empty, but I don't know why.

`idf_component_register(SRCS
                    INCLUDE_DIRS ".")
`

I add "main.c" to it and it worked again. I did not modify this file between renamings, as I mentioned earlier.

`idf_component_register(SRCS "main.c"
                    INCLUDE_DIRS ".")
`

But the the real problem is that VSCode gives a different error code, and it is very misleading.

igrr, 2024-09-04:
Looks like this might be an issue with VS Code extension. Please see https://github.com/espressif/vscode-esp-idf-extension/issues/1170#issuecomment-2017873247 and leave a comment there if you have a way to reproduce the issue.

brianignacio5, 2024-10-24:
There is a setting called `idf.enableUpdateSrcsToCMakeListsFile` which is `true` by default which updates the CMakeLists.txt automatically based when you add or delete source files in your project.

You can disable it by setting it to `false` in your `.vscode/settings.json`.

Could you describe steps to reproduce what did you do with sources ? We haven't received any issue with this feature.

Cimby1, 2024-10-24:
This auto add feature, do sometime work for me and sometime not.
 
I've been using ESP idf in VScode since, and I came to the conclusion that I was too dependent on error codes. 
This free hz etc message can pop up at every minute for no reason. When you are a newbie at esp programing it is so misleading.

So in the end the error cause was me for not properly including headers in source files and not adding source files to Cmakelist.
And not creating components and their folders as they should be created. 

It was a user error, but the error message was not helpful at all.

closed 2024-12-23 by Alvin1Zhang