espressif/esp-idf #14437: ULP source fails to compile when including both GPIO and I2C headers

sidwarkd, 2024-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect the main.c file for a ULP project to compile when including ulp_riscv_i2c.h and ulp_riscv_gpio.h.

### What is the actual behavior?

Many redefinition errors around various typedefs.

### Steps to reproduce.

I've included a repo using VSCode and devcontainers for easy reproduction of this issue. Pull the repo and checkout the [**redefinition_bug**](https://github.com/sidwarkd/rtc_i2c_issue/tree/redefinition_bug) branch.

1. Launch the project in VSCode
2. Reopen project in container development
3. Run idf.py build from the command line in VSCode

### Build or installation Logs.

```plain
ections -MD -MT CMakeFiles/ulp_main.dir/workspaces/rtc_i2c_bugs/main/ulp/main.c.obj -MF CMakeFiles/ulp_main.dir/workspaces/rtc_i2c_bugs/main/ulp/main.c.obj.d -o CMakeFiles/ulp_main.dir/workspaces/rtc_i2c_bugs/main/ulp/main.c.obj -c /workspaces/rtc_i2c_bugs/main/ulp/main.c
In file included from /workspaces/rtc_i2c_bugs/main/ulp/main.c:7:
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:18:5: error: redeclaration of enumerator 'GPIO_NUM_0'
   18 |     GPIO_NUM_0 = 0,     /*!< GPIO0, input and output */
      |     ^~~~~~~~~~
In file included from /home/dev/esp/idf/components/ulp/ulp_riscv/include/ulp_riscv_i2c.h:15,
                 from /workspaces/rtc_i2c_bugs/main/ulp/main.c:6:
/home/dev/esp/idf/components/hal/include/hal/gpio_types.h:169:5: note: previous definition of 'GPIO_NUM_0' with type 'enum <anonymous>'
  169 |     GPIO_NUM_0 = 0,     /*!< GPIO0, input and output */
      |     ^~~~~~~~~~
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:19:5: error: redeclaration of enumerator 'GPIO_NUM_1'
   19 |     GPIO_NUM_1 = 1,     /*!< GPIO1, input and output */
      |     ^~~~~~~~~~
/home/dev/esp/idf/components/hal/include/hal/gpio_types.h:170:5: note: previous definition of 'GPIO_NUM_1' with type 'enum <anonymous>'
  170 |     GPIO_NUM_1 = 1,     /*!< GPIO1, input and output */
      |     ^~~~~~~~~~
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:20:5: error: redeclaration of enumerator 'GPIO_NUM_2'
   20 |     GPIO_NUM_2 = 2,     /*!< GPIO2, input and output */
      |     ^~~~~~~~~~
/home/dev/esp/idf/components/hal/include/hal/gpio_types.h:171:5: note: previous definition of 'GPIO_NUM_2' with type 'enum <anonymous>'
  171 |     GPIO_NUM_2 = 2,     /*!< GPIO2, input and output */
      |     ^~~~~~~~~~
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:21:5: error: redeclaration of enumerator 'GPIO_NUM_3'
   21 |     GPIO_NUM_3 = 3,     /*!< GPIO3, input and output */
      |     ^~~~~~~~~~
/home/dev/esp/idf/components/hal/include/hal/gpio_types.h:172:5: note: previous definition of 'GPIO_NUM_3' with type 'enum <anonymous>'
  172 |     GPIO_NUM_3 = 3,     /*!< GPIO3, input and output */
      |     ^~~~~~~~~~
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:22:5: error: redeclaration of enumerator 'GPIO_NUM_4'
[... 94 lines cut ...]
      |     ^~~~~~~~~~~
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:38:5: error: redeclaration of enumerator 'GPIO_NUM_20'
   38 |     GPIO_NUM_20 = 20,
      |     ^~~~~~~~~~~
/home/dev/esp/idf/components/hal/include/hal/gpio_types.h:189:5: note: previous definition of 'GPIO_NUM_20' with type 'enum <anonymous>'
  189 |     GPIO_NUM_20 = 20,   /*!< GPIO20, input and output */
      |     ^~~~~~~~~~~
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:39:5: error: redeclaration of enumerator 'GPIO_NUM_21'
   39 |     GPIO_NUM_21 = 21,   /*!< GPIO21, input and output */
      |     ^~~~~~~~~~~
/home/dev/esp/idf/components/hal/include/hal/gpio_types.h:190:5: note: previous definition of 'GPIO_NUM_21' with type 'enum <anonymous>'
  190 |     GPIO_NUM_21 = 21,   /*!< GPIO21, input and output */
      |     ^~~~~~~~~~~
/home/dev/esp/idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:40:3: error: conflicting types for 'gpio_num_t'; have 'enum <anonymous>'
   40 | } gpio_num_t;
      |   ^~~~~~~~~~
/home/dev/esp/idf/components/hal/include/hal/gpio_types.h:216:3: note: previous declaration of 'gpio_num_t' with type 'gpio_num_t'
  216 | } gpio_num_t;
```

### More Information.

To get the project to successfully build, simply remove the i2c references in the main.c ulp file which includes commenting out the include of the header file as well as the _I2C Operations_ section of **main.c**

I also tried this in the v5.3 branch of IDF with the same results.

louielauanu, 2024-08-26:
Hi @sidwarkd,

Thanks for reporting this bug and provide the helpful reproduction repo. Our fix will be deleting the redundant gpio_num_t enum in ulp_core/include/ulp_riscv_gpio.h and unify it with the definition in "hal/gpio_types.h". This fix will come soon on IDF v5.3, but may not be backported to IDF v5.1.1.

Thanks,
Xiaoyu

AxelLin, 2024-08-26:
> This fix will come soon on IDF v5.3, but may not be backported to IDF v5.1.1.

@louielauanu 
I'm surprised you don't backport compile fix for v5.1 branch? (Isn't v5.1 brnach still in maintenance period?)

louielauanu, 2024-08-26:
> > This fix will come soon on IDF v5.3, but may not be backported to IDF v5.1.1.
> 
> @louielauanu I'm surprised you don't backport compile fix for v5.1 branch? (Isn't v5.1 brnach still in maintenance period?)

This fix will be ported to v5.1, but may not on v5.1.1, because it's code freeze. To obtain it, user have to fetch the next v5.1 release version, i.e. v5.1.5.

sidwarkd, 2024-08-26:
@louielauanu Thanks for the info. I understand about it being in a bugfix release of 5.1.x. That makes sense. :+1:

closed 2025-01-09 by Alvin1Zhang