espressif/esp-idf #14438: Mulit-source ULP code causes main app to hang

sidwarkd, 2024-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Having ULP source code in two separate C source files should work without issue

### What is the actual behavior?

When GPIO codes is moved into a secondary source file to get around #14437 the main application hangs as soon as a function is called that is in the second source file.

### Steps to reproduce.

1. Clone the [repo found here](https://github.com/sidwarkd/rtc_i2c_issue/tree/main_app_hang_bug)
2. Check out the **main_app_hang_bug** branch
3. Open the project in VSCode
4. Reopen the project in a devcontainer
5. Build and flash the program onto an S3 dev board
6. Notice the infinite while loop in main.cpp stops logging when the ULP code runs
7. Remove the call to `do_gpio` in the ULP main.c file
8. Rebuild and reflash the project
9. Notice the main app runs fine now

The only thing the `do_gpio` function does is call a few GPIO operations. You can see the same operations in main.c of the ULP source. When the exact same code runs from main.c it works fine without issue. To see this you will have to comment out the I2C code and headers as the bug described in #14437 prevents you from having both GPIO and I2C in the same file.

### Debug Logs.

```plain
I (300) app_start: Starting scheduler on CPU0
I (305) app_start: Starting scheduler on CPU1
I (305) main_task: Started on CPU0
I (315) main_task: Calling app_main()
I (315) main: In the main loop 0
....hangs forever here.....
```

### More Information.

I could not reproduce this issue on the v5.3 branch of the IDF so something appears to have been fixed but perhaps not back-ported to the v5.1.1 branch. I'm hoping it's a known fix that can be applied to 5.1.1 since it's still in the maintenance window.

louielauanu, 2024-08-27:
Hi @sidwarkd,

For your original issue, the attached patch could fix this issue without the need to move the gpio operations into a go_gpio function in separate source file. For this issue, I suspect it is because gpio 3 is somehow multiplexed with the console UART on v5.1. I did some experiments that if using some other gpio ports, the main program's print won't get stuck. I'm working on further debegging this and it may take some time.

Thanks,
Xiaoyu
[fix.patch.txt](https://github.com/user-attachments/files/16756321/fix.patch.txt)

louielauanu, 2024-08-30:
Hi @sidwarkd,

I further checked this case. The root cause is that `SET_PERI_REG_MASK` and similar macros are defined in both `components/soc/esp32s3/include/soc/soc.h` and 
`components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_register_ops.h` header files. In the `gpio.c`, `"ulp_riscv_gpio.h"` is included before `"ulp_riscv_utils.h"`, which caused wrong `SET_PERI_REG_MASK` and similar macros are used. We have modified it in the later versions as shown in the attached patch that includes `"ulp_riscv_register_ops.h"` in `components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h` to overwrite such macro definition. 
[uart_hang.patch.txt](https://github.com/user-attachments/files/16815477/uart_hang.patch.txt)

louielauanu, 2024-09-02:
> @louielauanu Thank you for the research on this. Will this fix get added to the 5.1.x branch?

Yes, it will be ported to v5.1 soon.

sudeep-mohanty, 2024-10-14:
Fixed with 3088bebc23ac7a8393805142726197577cd23f3d

closed 2024-10-14 by sudeep-mohanty