espressif/esp-idf #14448: ESP32P4 I2S-TDM模式下，BCLK频率最大不超过6.66MHz，MCLK频率最大不超过20MHz；

awoyzc, 2024-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

SP32P4 I2S-TDM模式下，设置48K采样率，16通道，16位数据，BCLK时钟应该为48000*16*16=12,288,000Hz，实际上不论如何设置，BCLK时钟最大不超过6.66MHz

### What is the actual behavior?

1、ESP32P4 I2S-TDM模式下，BCLK频率最大不超过6.66MHz，MCLK频率最大不超过20MHz；给MCLK以40MHz的外部时钟，BCLK频率也只会是6.66MHz，在这这个时钟下TDM无法支持48K，16比特，16通道采样，采样率大约只有26K；
2、降低位数，改为8位会导致BCLK也降低为3.33MHz,也就是采样率同样大约只有26K；
3、降低通道数改为8通道（连续的），位数为16位，BCLK反而可以达到6.66.采样率大约为48K
4、I2S_CLK_SRC_APLL不支持，会出现：esp_clk_tree: esp_clk_tree_src_get_freq_hz(82): freq shouldn't be 0
5、i2s_chan_config_t中dma_desc_num 默认值为6，dma_frame_num 默认值为240，不修改用默认值会提示dma_frame不能超过127，会自动修改为127

### Steps to reproduce.

static i2s_chan_handle_t i2s_tdm_init(void)
{
    i2s_chan_handle_t rx_chan = NULL;
    ESP_LOGI(TAG, "Create I2S receive channel");
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(0, I2S_ROLE_MASTER); // I2S_ROLE_SLAVE);
    chan_cfg.dma_desc_num = 8;
    chan_cfg.dma_frame_num = 120;
    ESP_ERROR_CHECK_WITH_LINE(i2s_new_channel(&chan_cfg, NULL, &rx_chan));

    ESP_LOGI(TAG, "i2s_new_channel OK");
    ESP_LOGI(TAG, "Configure I2S receive channel to TDM mode");

    i2s_tdm_config_t tdm_cfg = {
        .slot_cfg = I2S_TDM_PHILIPS_SLOT_DEFAULT_CONFIG(16, I2S_SLOT_MODE_STEREO, 0xFF), 
        .clk_cfg = {
            .clk_src = I2S_CLK_SRC_DEFAULT,
            //.ext_clk_freq_hz = 36864000,                 // 36864000,
            .sample_rate_hz = 48000,  // 采样率根据需求调整
            .mclk_multiple = I2S_MCLK_MULTIPLE_384, // 如果使用MCLK, 否则可以忽略
            //.bclk_div = 8,
        },
        //.clk_cfg = I2S_TDM_CLK_DEFAULT_CONFIG(48000),
        .gpio_cfg = {
            .mclk = I2S_MCK_IO,
            .bclk = I2S_BCK_IO,
            .ws = I2S_WS_IO,
            .dout = I2S_DOUT_IO,
            .din = I2S_DIN_IO,
        },
    };
    ESP_ERROR_CHECK_WITH_LINE(i2s_channel_init_tdm_mode(rx_chan, &tdm_cfg));
    ESP_LOGI(TAG, "Configure I2S receive channel to TDM mode OK!");
    return rx_chan;
}

i2s_channel_enable(rx_chan);
static int16_t i2s_readraw_buff[1024];        
size_t bytes_read = 0;
i2s_channel_read(rx_chan, i2s_readraw_buff, sizeof(i2s_readraw_buff), &bytes_read,pdMS_TO_TICKS(1000));
...

caidamao, 2024-08-27:
羡慕有P4的

L-KAYA, 2024-08-27:
APLL正在支持中~

P4的I2S时钟源在不支持APLL的情况下最高只能到40M（即XTAL频率），MCLK至少2分频，所以MCLK只能到20M。

需要尽快测试的话可以先尝试一下这个patch:
[0001-feat-clock-support-apll-clock-on-p4.patch](https://github.com/user-attachments/files/16758368/0001-feat-clock-support-apll-clock-on-p4.patch)

awoyzc, 2024-08-28:
打了补丁了，成功实现16位，16通道，48K采样，感谢
但I2S_CHANNEL_DEFAULT_CONFIG中 dma_frame_num默认为240，会出错，
提示：W (361) i2s_common: dma frame num is out of dma buffer size, limited to 127
无法内存对齐，只能改为120，正常采样

L-KAYA, 2024-08-28:
> 但I2S_CHANNEL_DEFAULT_CONFIG中 dma_frame_num默认为240，会出错， 提示：W (361) i2s_common: dma frame num is out of dma buffer size, limited to 127 无法内存对齐，只能改为120，正常采样

这个问题的修复已经提交了，审核中~

closed 2024-08-30 by espressif-bot

DatanoiseTV, 2024-09-06:
After patching, I still see:

```
  E (360) i2s_common: The required APLL frequency exceed its maximum value
                                                                                                                 E (370) i2s_tdm: i2s_tdm_calculate_clock(64): sample rate is too large for the current clock source
                                        E (380) i2s_tdm: i2s_tdm_set_clock(76): clock calculate failed
                                                                                                      E (390) i2s_tdm: i2s_channel_init_tdm_mode(231): initialize channel failed while setting clock
```
```
    i2s_tdm_config_t rx_tdm_cfg = {
.clk_cfg = {
        .clk_src = I2S_CLK_SRC_APLL,
        //.ext_clk_freq_hz = 36864000,                 // 36864000,
        .sample_rate_hz = 48000,  // 采样率根据需求调整
        .mclk_multiple = I2S_MCLK_MULTIPLE_512, // 如果使用MCLK, 否则可以忽略
        //.bclk_div = 8,
    },
        .slot_cfg = I2S_TDM_MSB_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_STEREO,
                                                    I2S_TDM_SLOT0 | I2S_TDM_SLOT1 | I2S_TDM_SLOT2 | I2S_TDM_SLOT3 |
                                                    I2S_TDM_SLOT4 | I2S_TDM_SLOT5 | I2S_TDM_SLOT6 | I2S_TDM_SLOT7 | I2S_TDM_SLOT8 | I2S_TDM_SLOT9 | I2S_TDM_SLOT10 | I2S_TDM_SLOT11 | I2S_TDM_SLOT12 | I2S_TDM_SLOT13 | I2S_TDM_SLOT14 | I2S_TDM_SLOT15),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,    // some codecs may require mclk signal, this example doesn't need it
            .bclk = EXAMPLE_TDM_BCLK_IO2,
            .ws   = EXAMPLE_TDM_WS_IO2,
            .dout = EXAMPLE_TDM_DOUT_IO2,
            .din  = EXAMPLE_TDM_DIN_IO2,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv   = false,
            },
        },
    };
```