espressif/esp-idf #14454: Increase of LWIP_MAX_SOCKETS to 32 instead of 16

jzwyssig, 2024-08-27:
### Is your feature request related to a problem?

The `LWIP_MAX_SOCKETS` setting in `esp-idf/components/lwip` is currently limited to 16, which can be restrictive since ESP-IDF can easily handle more open sockets. This limit is quickly reached when running services like an HTTP server. Increasing the limit to 32 would solve this problem without causing any issues, while still keeping the default value at 10. Right now, I have to manually patch ESP-IDF to work around this limit, so it would be great if this change could be made directly in the codebase.

### Describe the solution you'd like.

Patch applied to `esp-idf`:
```
diff --git a/components/lwip/Kconfig b/components/lwip/Kconfig
index a47bbe9ecd..3a219200cf 100644
--- a/components/lwip/Kconfig
+++ b/components/lwip/Kconfig
@@ -135,8 +135,8 @@ menu "LWIP"
 
     config LWIP_MAX_SOCKETS
         int "Max number of open sockets"
-        range 1 16
-        default 10
+        range 1 32
+        default 32
         help
             Sockets take up a certain amount of memory, and allowing fewer
             sockets to be open at the same time conserves memory. Specify
```

atanisoft, 2024-08-28:
This has been done on [master](https://github.com/espressif/esp-idf/commit/48295d31ca90d693356b632750630efd2f1b34eb) but not yet backported to release branches.

It would be good to get this backported to at least v5.3.x

AxelLin, 2024-11-07:
@david-cermak Any progress for backporting to stable branches?

hitecSmartHome, 2024-12-12:
Would appreciate the backport to 5.3

euripedesrocha, 2024-12-12:
@hitecSmartHome @AxelLin the backport process is done for 5.3 and 5.2 branches, the change is available in 5.3.2.

closed 2024-12-12 by euripedesrocha