espressif/esp-idf #14460: BT Nimble LE pairing takes too long - watchdog timer timeout

eriksl, 2024-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Irrelevant here

### Power Supply used.

USB

### What is the expected behavior?

I expect the pairing to complete, as it has done previously, using both Linux and Android host device.

### What is the actual behavior?

Watchdog timer timeout during. The timeout does occur deep inside the IDF code, somewhere in mbedtls, it's not in my code.

### Steps to reproduce.

1. Setup BT GATT on ESP32-S3
2. Try to pair (with no pre-existing bonds)
3. Wait long, over 5 seconds.

### More Information.

When I increase the watchdog timer timeout to 10 seconds, the issue is gone. Well, it's worked around. I don't think
a) pairing should take this long
b) even if it takes so long, the watchdog timer is never fed

rahult-github, 2024-08-29:
Hello @eriksl ,

Can you possibly give more information on how to reproduce the issue .

> Setup BT GATT on ESP32-S3
Try to pair (with no pre-existing bonds)
Wait long, over 5 seconds.

1. By setup BT GATT , which application are you running on S3 , is it bleprph ? 
2. Try to pair --> Who is initiating the pairing and by which method ? ( e.g. Bond option on third party phone like nrfconnect ? Am assuming no change is done in default IOCAP on S3 side  ?  Please share your sdkconfig for this . 
4.  Wait long, over 5 seconds  -->  What is the remote device being used ? Is there no event received on S3 side.  Can you please enable Nimble Debug logs and share console log too ?   The way to enable debug logs : 

> Component config → Log output → Default log verbosity --> Set to Debug
> Component config → Bluetooth → NimBLE Options → NimBLE Host log verbosity --> Set to Debug

eriksl, 2024-08-30:
I am running my own application, https://github.com/eriksl/esp32 but not all required information to set up Nimble is available in the documentation and I had to guess some things from the examples like blerph. I will make sure the repository on github is up to date at the end of the day, right now it's a bit outdated due to holidays and a major overhaul + IDF upgrade.

The ESP32 is the peripheral, the Linux host is the central. The central initiates the pairing. It uses whatever method the peripheral offers. To rule out an issue with Bluez I also tried pairing using Android and that gives exactly the same result. Please note, you mention nRF, but the issue is way before the point where an application can connect. It's really during _pairing_.

IOCAP of course hasn't changed. Actually none of the BT setup code has changed.

The remote is waiting for the S3, not the other way around! The S3 seems to be busy with "something" and finally gives up when the watchdog timer intervenes.

I don't think the debug log will yield anything useful, but the backtrace from the watchdog timer will be very informative I think. The intervention always takes place at the same point in the code, so it looks like the processor stalls there for some reason. Look like some function to handle hardware offload from within MBEDTLS.

I'll attach the relevant code on the ESP32-S3 code, the stacktrace and the config.

eriksl, 2024-08-30:
Relevant sources.

eriksl, 2024-08-30:
sdkconfig
[sdkconfig.zip](https://github.com/user-attachments/files/16812949/sdkconfig.zip)

rahult-github, 2024-09-05:
Hi @eriksl ,   

the backtrace shows mbedtls functions which , i guess, are causing infinite loop and hence the issue.  

Can you please share the way to compile the code against IDF SDK you have shared ( either the bluetooth-src.zip or even the github repo) . I tried few ways like making folder inside IDF or exporting paths, but it fails. 

I took the init content from bt.c and replicated it in SDK sample bleprph example.  I tested against phone and bluez and both worked fine.  So, if you can share your compilation setup against IDF SDK to get the application compiled in its entirety , it would help us to follow the same steps and debug more by reproducing issue locally. 

[idf.txt](https://github.com/user-attachments/files/16892239/idf.txt)

[bluez.txt](https://github.com/user-attachments/files/16892310/bluez.txt)

Changed example file for reference:  [main.txt](https://github.com/user-attachments/files/16892648/main.txt)

eriksl, 2024-09-05:
My github repository is now completely up to date and should build cleanly now.

https://github.com/eriksl/esp32

Steps to build and test:

- clone repository
- you will need a Linux-like build environment because I am using symbolic links. This is because I intend to support multiple ESP32 models and not all model's builds will have all source files or will have the all same source files. By using symlinks I can use one source in more models, if possible.
- chdir to s3/develop
- run idf.py there

That should be sufficient. If there are any difficulties, please let me know and I will fix it.

Build as is now, it should work. If you change the watchdog timer back to the default of 6 seconds, the watchdog timer will go off.

Thank you for looking into this!

Please note there is no endlessly looping code (either in my app or in mbedtls) because the pairing will succeed if given enough time (set watchdog timer to 10 seconds).

eriksl, 2024-09-06:
Update, please clone this repository again or pull it (although cloning is probably safest). I have managed to make it ESP32-sub-type and board-type independent without the symlinks kludge. It now looks like a completely normal project, with just two extra menuconfig entries. The board selection currently does not have any influence on the BT code, so you can choose any of the two you like.

I don't expect any building issues now.

rahult-github, 2024-09-06:
Compilation is ok now . Thanks. 

but the chip fails to boot. 

> --- esp-idf-monitor 1.4.0 on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1988
load:0x403c8700,len:0x4
load:0x403c8704,len:0xda0
load:0x403cb700,len:0x3518
entry 0x403c8918

 The menuconfig has many changes compared to default example sdkconfig. I will try to figure out which option is causing this and check more

eriksl, 2024-09-06:
Interesting. It will probably have to do with me using a Wemos Lolin S3-mini that has 4 MB flash and 2 MB SPIRAM and I have them "hard" configured in the sdkconfig, I am afraid it can't be done otherwise without issues. If you can get a similar setup somewhere, I guess it could work. Maybe you need to the chip type to autodetect.

If this is a problem I may be able to make a version that runs without SPIRAM.

Sorry for the inconvencience.

rahult-github, 2024-09-09:
> This may have to do with me having all MBEDTLS memory allocated from SPIRAM 

Ok. This can be possible. Do you observe issue if SPIRAM support is disabled and internal memory is used ?

rahult-github, 2024-09-24:
Closing this issue currently. Feel free to reopen in case of any further updates.

closed 2024-09-24 by rahult-github