espressif/esp-idf #14461: Regression: Nimble LE doesn't receive packets > ~256 bytes even though the MTU is set to 560 bytes on both sides

eriksl, 2024-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Irrelevant here

### Power Supply used.

USB

### What is the expected behavior?

I expect to be able to send (from the Linux host) packets of 560 bytes and to be able to receive them on the ESP32-S3.

### What is the actual behavior?

Packets > 273 bytes are dropped.

### Steps to reproduce.

1. Negotiate MTU of 560 on Linux HCI host (this is the maximum Bluez supports for BLE)
2. Configure IDF for an MTU of 560 in the menuconfig
3. Send large packets (> 273) bytes from Linux to ESP32
4. They are sent OK, but never arrive at the Nimble API

This used to work in v5.2.2.

rahult-github, 2024-08-29:
Hi @eriksl ,

> Negotiate MTU of 560 on Linux HCI host (this is the maximum Bluez supports for BLE)
> Configure IDF for an MTU of 560 in the menuconfig

May i know which MTU is being reffered to here and how is it being set on IDF ?  For nimble, the one MTU option available is ATT_PREFFERED_MTU which is configurable from menuconfig. However, per spec the largest value is 517 . So please share more information on this. 

Do you have any btmon log and device side nimble debug logs that you can share for this ?  

For receiving the packets at device side, which API or event are you referring to ?  May be you can share some application code , to give more information of what your application is doing. 

In case you have success/ failure logs, please share them too.

eriksl, 2024-08-30:
The MTU is indeed set using the menuconfig. I couldn't find any reference to some Nimble API function to set it. It has been this way in v5.2.2 and there it worked.

You say "as per spec", what spec? I couldn't find any hard/clear limitation on this. I just found out that Bluez (on the Linux side) implements a hardcoded limit of around 560 bytes, independent from what is configured in the HCI, which accepts almost anything.

I can see the "MTU update event" is called, I'll add a print to see to what value the MTU is changed now. I suspect it's really 560 like before.

I have strace logs from the application, where you can see the packet is sent successfully, in one go. I have tried btmon in the past, but the usefulness appears to be very limited. I can try again.

The application is actually very simple. It defines one GATT attribute that can be written. As confirmation, an _indication_ is sent. That's all. I need larger MTU values because this is used to upload an OTA image (as an alternative to wlan). I am now using packets of ~192 bytes, which does work, but makes the whole process even more slow than it was. Uploading an image now takes almost 10 minutes. In the past I could do it in ~2-3 minutes. Which still is very slow, if you take BT speed in account vs. the amount of data.

See attachment for source.

eriksl, 2024-08-30:
sdkconfig
[sdkconfig.zip](https://github.com/user-attachments/files/16813115/sdkconfig.zip)

eriksl, 2024-08-30:
So this is what happens, I think.

Central role, sends 554 bytes, succesfully, in one go (strace)
```
sendto(3, "\x12\x10\x00\x...", 554, 0, NULL, 0) = 554
```

Central (Linux) and peripheral connect and negotiate MTU (btmon, client side):
```
< HCI Command: Write Authenticated Payload Timeout (0x03|0x007c) plen 4                                                                                                               #19 [hci0] 6.103643
        Handle: 2048 Address: DC:54:75:E1:A4:C2 (Espressif Inc.)
        Timeout: 30000 msec (0x0bb8)
e32if[2718641]: < ACL Data TX: Handle 2048 flags 0x00 dlen 7                                                                                                                          #20 [hci0] 6.103756
      ATT: Exchange MTU Request (0x02) len 2
        Client RX MTU: 560
> HCI Event: Command Complete (0x0e) plen 6                                                                                                                                           #21 [hci0] 6.104513
      Write Authenticated Payload Timeout (0x03|0x007c) ncmd 1
        Status: Success (0x00)
        Handle: 2048 Address: DC:54:75:E1:A4:C2 (Espressif Inc.)
```
Server side:
```
> ACL Data RX: Handle 2048 flags 0x02 dlen 7                                                                                                                                          #24 [hci0] 6.194938
      ATT: Exchange MTU Response (0x03) len 2
        Server RX MTU: 560
```

But then it gets interesting. Keep the value **251** in mind.
```
> HCI Event: LE Meta Event (0x3e) plen 11                                                                                                                                             #23 [hci0] 6.193516
      LE Data Length Change (0x07)
        Handle: 2048 Address: DC:54:75:E1:A4:C2 (Espressif Inc.)
        Max TX octets: 251
        Max TX time: 2120
        Max RX octets: 27
        Max RX time: 328
```

Now have a look at the point where the 554 bytes are actually sent:
```
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                               #44 [hci0] 10.020465
        Num handles: 1
        Handle: 2048 Address: DC:54:75:E1:A4:C2 (Espressif Inc.)
        Count: 1
        #40: len 251 (45 Kb/s)
        Latency: 44 msec (41-45 msec ~44 msec)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                               #45 [hci0] 10.021464
        Num handles: 1
        Handle: 2048 Address: DC:54:75:E1:A4:C2 (Espressif Inc.)
        Count: 1
        #41: len 251 (44 Kb/s)
        Latency: 45 msec (41-45 msec ~44 msec)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                               #46 [hci0] 10.022457
        Num handles: 1
        Handle: 2048 Address: DC:54:75:E1:A4:C2 (Espressif Inc.)
        Count: 1
        #42: len 56 (9 Kb/s)
        Latency: 46 msec (41-46 msec ~45 msec)
```
The data is actually split into chunks of **251** bytes!

Now when we look at the a receiving application:
```
$ bt receive 273 bytes
```
It only receives the first chunk (and has a few appended to it, someway, I don't known..., 251 -> 273). The "length" referred to here, is the "data length" from the om_buf as received from the "access_cb" function in the "gatt definition".

When I decrease the data sent to e.g. 128 bytes, it all works like clockwork.

rahult-github, 2024-08-31:
> It only receives the first chunk (and has a few appended to it, someway, I don't known..., 251 -> 273). The "length" referred to here, is the "data length" from the om_buf as received from the "access_cb" function in the "gatt definition".

The data is in linked form for the om buff.  You can refer to [this](https://github.com/espressif/esp-idf/blob/master/components/bt/common/btc/profile/esp/blufi/nimble_host/esp_blufi.c#L127) and see how to traverse the om buf structure in a loop to get all data.

eriksl, 2024-08-31:
Yes, I f*cked this up, indeed, but not exactly like this.

I know an mbuf can be chained. What went wrong:
- changed my code a little bit to allocate just as many bytes as required and pass it upstream, instead of using a generic, large buffer, that got copied and re-used
- to find out the length of the buffer to be allocated, I used the om_len struct member, instead of using the function os_mbuf_len(), which exists exactly for this purpose.

What made me assume wrong was the max length of the data passed exactly corresponding with the same protocol exchange value and having read about this in the git log.

So please excuse me, the issue can be closed. It now works as before.

closed 2024-09-02 by rahult-github