espressif/esp-idf #14463: HTTPS OTA Bug on ESP32 S3

akashpraan, 2024-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4

### Espressif SoC revision.

ESP32 S3 Wroom 1U (8MB Flash)

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I have had multiple testing of OTA and my application file is 1.4 MB and I have had success as well and this is the expected behaviour

### What is the actual behavior?

Certain times, there is an error from assert failed: http_utils_append_string http_utils.c:72 (old_str), I want to understand how to solve it

And many times it is successful as well. 

`static void event_handler(void* arg, esp_event_base_t event_base,int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) 
    {
        esp_wifi_connect();
    } 
    else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) 
    {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) 
        {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI("OTA", "retry to connect to the AP");
        } 
        else 
        {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI("OTA","connect to the AP fail");
    } 
    else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) 
    {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI("OTA", "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
        s_retry_num = 0;
        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
[... 309 lines cut ...]
I (113995) esp_https_ota: Written image length 851942
I (118549) esp_https_ota: Written image length 866614
I (133143) esp_https_ota: Written image length 881286
I (133158) esp_https_ota: Written image length 884709
I (136636) esp_https_ota: Written image length 899381
I (138024) esp_https_ota: Written image length 914053
I (138039) esp_https_ota: Written image length 917476
I (141632) esp_https_ota: Written image length 932148
I (142116) esp_https_ota: Written image length 946820
I (142131) esp_https_ota: Written image length 950243
I (145834) esp_https_ota: Written image length 964915
I (146043) esp_https_ota: Written image length 979587
I (146058) esp_https_ota: Written image length 983010
I (149830) esp_https_ota: Written image length 997682
I (150370) esp_https_ota: Written image length 1012354
I (150384) esp_https_ota: Written image length 1015777
I (154641) esp_https_ota: Written image length 1030449
I (155171) esp_https_ota: Written image length 1045121
I (155185) esp_https_ota: Written image length 1048544
I (158660) esp_https_ota: Written image length 1063216

assert failed: http_utils_append_string http_utils.c:72 (old_str)

and then the ESP Restarts
```

nileshkale123, 2024-08-30:
Thank you for raising this issue. To help further, could you please provide the sdkconfig file and the verbose logs? You can enable verbose logging by navigating to idf.py menuconfig -> Component config -> Log -> Log Level -> Default log verbosity. This information will help us diagnose the problem more effectively.

akashpraan, 2024-08-30:
[Couldnt_allocate_memory_to_upgrade_data_buffer.txt](https://github.com/user-attachments/files/16815835/Couldnt_allocate_memory_to_upgrade_data_buffer.txt)
[OTA_Basic_COde.txt](https://github.com/user-attachments/files/16815836/OTA_Basic_COde.txt)

Hello, when I increase the log verbosity to debug, I get an error stating, Couldnt_allocate_memory_to_upgrade_data_buffer
[SDKConfig.txt](https://github.com/user-attachments/files/16815845/SDKConfig.txt)

Also, the bug doesn't happen always, you could say 3 to 4 out of 10 times. 

I would like to know the answer to this, any way I can make the buffer memory error go away so that I can provide the logs to you?

I have attached my basic OTA code, other than reading the network parameters and certificate+url, everything else is there in the code as well as the error logs for Couldnt_allocate_memory_to_upgrade_data_buffer.

akashpraan, 2024-08-31:
@nileshkale123 anything? please help with the issue

mahavirj, 2024-09-04:
@akashpraan 

- Is it possible to share URL where the OTA update file is hosted? We can try this URL in the existing IDF examples to check if there is any issue with the OTA update process.
- You are enabling partial http download feature with large buffer size, any specific reason for this? What happens if you disable this feature?

> assert failed: http_utils_append_string http_utils.c:72 (old_str)

This error points to insufficient heap memory in the system. You may follow this [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/performance/ram-usage.html) to optimize the memory usage.

akashpraan, 2024-09-18:
@mahavirj 

Is it possible to share URL where the OTA update file is hosted? We can try this URL in the existing IDF examples to check if there is any issue with the OTA update process.

https://drive.google.com/uc?export=download&id=15BQ6Oi0Eq18DZXDbuFqmWDLdWUginE86
[CRT.txt](https://github.com/user-attachments/files/17046579/CRT.txt)

`esp_http_client_config_t clientConfig = 
 {
     .url = srv_url,
     .timeout_ms = 60000,
     .cert_pem = cert_data,  // Use the certificate read from NVS (or fallback)
     .keep_alive_enable = true,
     .buffer_size = 4096,  //1048*14=14672, 16384->16KB, 32768->32KB, 65536->64Kb, I am not able to use 32K or 64K. 
     .buffer_size_tx = 4096,        //TX Buffer, Main Buffer
     .event_handler = client_event_handler,
 };

  esp_https_ota_config_t ota_config = 
  {
      .http_config = &clientConfig,
      .http_client_init_cb = _http_client_init_cb,
      .partial_http_download = true,
      .max_http_request_size = 4096,         //SHould match to TX Buffer size above
  };`

I apologize for the delay and I have also attached my final SDK Config

[SDKConfig.txt](https://github.com/user-attachments/files/17046616/SDKConfig.txt)

Now, I fail for OTA if I increase my https buffer size to 8KB, stating the same heap memory issue, I tried freeing some Heap Memory but wasn't able to achieve much.

akashpraan, 2024-09-23:
@mahavirj @nileshkale123 
I have three retries for OTA in my application

Now i have noticed it a couple of times that on the first retry it will say that Mismatch Chip ID and on the second try OTA will start to download files. 

Below are the logs: 
I (2650) OTA: Waiting for semaphore
I (2654) OTA: Received semaphore
I (2658) OTA: OTA timer started successfully.
I (2664) OTA-FAIL-READ-NVS: Successfully read OTA fail counter from NVS: 0
W (2671) MAIN: OTA retry counter in OTA APP: 0
I (4506) wifi:<ba-add>idx:1 (ifx:0, 00:31:92:30:47:a6), tid:4, ssn:1, winSize:64
I (42244) esp_https_ota: Starting OTA...
I (42245) esp_https_ota: Writing to partition subtype 16 at offset 0x280000
I (42250) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=1024
I (42251) HEAP: Free heap during OTA: 36932
E (42256) esp_https_ota: Mismatch chip id, expected 9, found 27757
E (42265) RUN-OTA: OTA failed.
I (42267) OTA-FAIL-READ-NVS: Successfully read OTA fail counter from NVS: 0
I (42276) OTA-FAIL-COUNT-WRITE-NVS: Successfully committed OTA fail counter to NVS: 1
I (43284) wifi:state: run -> init (0)
I (43285) wifi:pm stop, total sleep time: 31485277 us / 42222223 us

I (43285) wifi:<ba-del>idx:0, tid:0
I (43286) wifi:<ba-del>idx:1, tid:4
I (43289) wifi:new:<2,0>, old:<2,1>, ap:<255,255>, sta:<2,1>, prof:1
I (43298) OTA: retry to connect to the AP
I (43300) OTA: connect to the AP fail
I (43326) wifi:flush txq
I (43327) wifi:stop sw txq
I (43327) wifi:lmac stop hw txq
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037bebe
--- 0x4037bebe: esp_cpu_wait_for_intr at C:/ESP/esp-idf/v5.1.4/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1758
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e0c
entry 0x403c9908
I (31) boot: ESP-IDF v5.1.4-dirty 2nd stage bootloader
I (31) boot: compile time Sep 23 2024 12:37:30
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.1
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (77) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (84) boot:  2 factory          factory app      00 00 00010000 00270000
I (92) boot:  3 ota_0            OTA app          00 10 00280000 00270000
I (99) boot:  4 ota_1            OTA app          00 11 004f0000 00270000
I (107) boot:  5 storage          Unknown data     01 82 00760000 008a0000
I (114) boot: End of partition table
I (118) boot: Defaulting to factory image
I (123) esp_image: segment 0: paddr=00010020 vaddr=3c100020 size=4b87ch (309372) map
I (187) esp_image: segment 1: paddr=0005b8a4 vaddr=3fc95f00 size=04774h ( 18292) load
I (191) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=f9898h (1022104) map
I (376) esp_image: segment 3: paddr=001598c0 vaddr=3fc9a674 size=00340h (   832) load
I (377) esp_image: segment 4: paddr=00159c08 vaddr=40374000 size=11e9ch ( 73372) load
I (406) boot: Loaded app from partition at offset 0x10000
I (406) boot: Disabling RNG early entropy source...
I (418) cpu_start: Multicore app
[... 112 lines cut ...]
I (2578) OTA: Received semaphore
I (2582) OTA: OTA timer started successfully.
I (2588) OTA-FAIL-READ-NVS: Successfully read OTA fail counter from NVS: 1
W (2595) MAIN: OTA retry counter in OTA APP: 1
I (7708) wifi:<ba-add>idx:1 (ifx:0, 00:31:92:30:47:76), tid:4, ssn:0, winSize:64
I (11886) esp_https_ota: Starting OTA...
I (11887) esp_https_ota: Writing to partition subtype 16 at offset 0x280000
I (11897) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=1024
I (11897) HEAP: Free heap during OTA: 35236
I (11928) esp_https_ota: Written image length 1024
I (11929) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=3823
I (11929) HEAP: Free heap during OTA: 35236
I (11936) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=1297
I (11939) HEAP: Free heap during OTA: 38708
I (11985) esp_https_ota: Written image length 6144
I (17016) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=4858
I (17017) HEAP: Free heap during OTA: 35000
I (17020) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=1286
I (17022) HEAP: Free heap during OTA: 38472
I (17069) esp_https_ota: Written image length 12288
I (20695) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=4852
I (20695) HEAP: Free heap during OTA: 36528
I (20700) CLIENT_EVENT: HTTP_EVENT_ON_DATA, len=1292
I (20700) HEAP: Free heap during OTA: 38264
I (20779) esp_https_ota: Written image length 18432

akashpraan, 2024-09-23:
@nileshkale123 @mahavirj PLEASE HELP as the performance of the OTA is unstable, it sometime takes 20 mins as well and later fails due to 

> * You are enabling partial http download feature with large buffer size, any specific reason for this? What happens if you disable this feature?
My understanding is I don't have enough memory/space to download the complete file in one go so I did this. My application is 1.5MB

akashpraan, 2024-09-24:
> I tested the configuration you mentioned by embedding in [simple ota example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/simple_ota_example) and the OTA was completed successfully in less than 1 minute using the link you provided, with the partial HTTP download feature disabled.

Ok for Now I will also use this and get my OTA working seamlessly. 

Thank you for the help

closed 2024-09-24 by akashpraan

nileshkale123, 2024-11-29:
Hello @akashpraan ,

The issue raised primarily stems from insufficient memory. To address this, we have created an internal MR to handle such scenarios effectively.

Thank you for bringing this to our attention.