espressif/esp-idf #14465: ESP32S3 WIFI join AP failure

beanjs, 2024-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-2596-g6e5414b6c4

### Espressif SoC revision.

ESP32S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Join AP should work.

### What is the actual behavior?

Failed to join the AP. The log indicates that the verification timed out.

### Steps to reproduce.

using wifi/getting_started example
[Uploading sdkconfig.zip…](sdkconfig)

### Debug Logs.

```plain
I (23) boot: ESP-IDF v5.4-dev-2596-g6e5414b6c4 2nd stage bootloader
I (23) boot: compile time Aug 30 2024 13:16:25
I (24) boot: Multicore bootloader
I (28) boot: chip revision: v0.2
I (31) boot: efuse block revision: v1.3
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=240ech (147692) map
I (128) esp_image: segment 1: paddr=00034114 vaddr=3fc99a00 size=04b18h ( 19224) load
I (132) esp_image: segment 2: paddr=00038c34 vaddr=40374000 size=073e4h ( 29668) load
I (140) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=79b50h (498512) map
I (230) esp_image: segment 4: paddr=000b9b78 vaddr=4037b3e4 size=0e5e4h ( 58852) load
[... 753 lines cut ...]
D (9481) wifi:scan end: arg=0x3fc9f7fa, status=0, ss_state=0x3
D (9481) wifi:perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
V (9481) wifi:change: chan<3,0>, dur<0,120>
V (9491) wifi:ht20 freq=2422, chan=3
V (9491) wifi:enter start op, arg=0x3fc9f7fa
V (9491) wifi:scan operation: state=0x3, chan<3,0>, arg=0x3fc9f7fa, status=0
V (9501) wifi:scan specific ssid=WY216
V (9501) wifi:start max timer
D (9631) wifi:scan end: arg=0x3fc9f7fa, status=0, ss_state=0x3
D (9631) wifi:perform scan: ss_state=0x9, chan<4,0>, dur<0,120>
V (9631) wifi:change: chan<4,0>, dur<0,120>
V (9631) wifi:ht20 freq=2427, chan=4
V (9631) wifi:enter start op, arg=0x3fc9f7fa
V (9641) wifi:scan operation: state=0x3, chan<4,0>, arg=0x3fc9f7fa, status=0
V (9651) wifi:scan specific ssid=WY216
V (9651) wifi:start max timer
D (9771) wifi:scan end: arg=0x3fc9f7fa, status=0, ss_state=0x3
D (9771) wifi:perform scan: ss_state=0x9, chan<5,0>, dur<0,120>
V (9771) wifi:change: chan<5,0>, dur<0,120>
V (9781) wifi:ht20 freq=2432, chan=5
V (9781) wifi:enter start op, arg=0x3fc9f7fa
V (9781) wifi:scan operation: state=0x3, chan<5,0>, arg=0x3fc9f7fa, status=0
V (9791) wifi:scan specific ssid=WY216
V (9791) wifi:start max timer
```

beanjs, 2024-08-30:
[sdkconfig.zip](https://github.com/user-attachments/files/16812149/sdkconfig.zip)

beanjs, 2024-09-02:
smartconfig can receive configuration information，but join AP always failure

```log
I (43155) smartconfig: TYPE: AIRKISS
I (43155) smartconfig: T|PHONE MAC: ea:34:b4:16:69:7f
I (43155) smartconfig: T|AP MAC:5c:27:d4:98:c9:8e
I (43156) smartconfig: Found channel on 11-0. Start to get ssid and password...
I (45910) smartconfig: T|pswd: wy12345678
I (45910) smartconfig: T|ssid: WY216
I (45910) wifi:ic_disable_sniffer
W (00:01:06.733) WIFI: ssid:WY216 password:wy12345678
W (45916) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I (47505) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
I (47505) wifi:state: init -> auth (0xb0)
I (48506) wifi:state: auth -> init (0x200)
I (48509) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
I (50973) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
I (50974) wifi:state: init -> auth (0xb0)
I (51976) wifi:state: auth -> init (0x200)
I (51978) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
I (00:01:13.427) WL: total: 9428(KB)  free: 9428(KB) pst: 100
```

zhangyanjiaoesp, 2024-09-04:
@beanjs It seems the ESP32S3 send auth packet, but doesn't receive the auth packet from AP. Is there any special about your AP？Is your AP a router or a hotspot？Can you  provide a packet capture using Wireshark?

beanjs, 2024-09-05:
请问怎么使用Wireshark进行抓包？

zhangyanjiaoesp, 2024-09-05:
@beanjs  you can refer this link https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/wireshark-user-guide.html#wireshark

beanjs, 2024-09-05:
你发的这个教程连接，感觉里面没什么干货啊。就是简单的软件使用，就我现在的问题也没有具体的抓包方法和查找问题的方法。另外这个问题是出现在谋几个模块上，有的模块又是没有问题的。所以我感觉这个不是路由器的问题吧。

zhangyanjiaoesp, 2024-09-05:
> 就是简单的软件使用，就我现在的问题也没有具体的抓包方法和查找问题的方法。

在过滤条件里面设置 AP 的 mac 地址和 STA 的 mac 地址，设置正确的 channel，抓包就可以了。如下图，显示了 STA 连接 AP 的过程：
[image]
抓包保存后，当去掉过滤条件之后会显示 mac ACK 的抓包，表示收到了对方的包。
[image]

> 这个问题是出现在谋几个模块上，有的模块又是没有问题的

这个情况你之前并没有说明，如果是这样的话，可能是模块的问题，不过还是可以抓包确认一下。

beanjs, 2024-09-05:
Authentication数据包就没有发出，这个是不是说明芯片坏了？但是我刚才测试了芯片的AP模式，这个模式又是工作正常的，就是STA模式在连接AP的时候，没有发送Authentication数据包。不过smartconfig又能接收到手机发出的AP配置信息。

beanjs, 2024-09-05:
[2024-09-05_11.12.10.257.pcap.zip](https://github.com/user-attachments/files/16883547/2024-09-05_11.12.10.257.pcap.zip)

beanjs, 2024-09-05:
@zhangyanjiaoesp 您好，我大概理清楚原因了。会不会是SDK里面对Authentication的发送进行了某种判断，然后因为没有打不到某个条件导致没有发出Authentication数据包，从而导致了`wifi:start 1s AUTH timer`超时从而触发了`WIFI_EVENT_STA_DISCONNECTED`事件

zhangyanjiaoesp, 2024-09-05:
`wifi:start 1s AUTH timer` 这个timer 是在 auth 发送了之后才启动的。station example 会默认重连5次，5 次都连接不成功吗？

beanjs, 2024-09-05:
都不成功呢，每次都是因为`wifi:start 1s AUTH timer`超时引起`WIFI_EVENT_STA_DISCONNECTED `事件。

zhangyanjiaoesp, 2024-09-05:
> [2024-09-05_11.12.10.257.pcap.zip](https://github.com/user-attachments/files/16883547/2024-09-05_11.12.10.257.pcap.zip)
会不会是这个抓包里面没有抓到auth 包？你的 STA 的 mac 地址是多少？

beanjs, 2024-09-05:
这个抓包是能抓到Authentication数据包的，发出来的这个是有问题那个模块的抓包文件。另外正常的模块是可以抓到Authentication包的，验证过程和你发给我那个图片的差不多。我就没有发了，我现在另外给你抓个正常模块的包。

[2024-09-05_15.08.39.900.pcap.zip](https://github.com/user-attachments/files/16885379/2024-09-05_15.08.39.900.pcap.zip)

zhangyanjiaoesp, 2024-09-05:
你用 release/v5.3 的分支测试一下？

beanjs, 2024-09-05:
也还是一样的，我拉取的是最新的代码，也就是合并5.3.1的。

```
I (24443) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
D (24451) wifi:connect_op: status=0, auth=5, cipher=3 
D (24455) wifi:auth mode is not none
D (24459) wifi:connect_bss: auth=1, reconnect=0
I (24463) wifi:state: init -> auth (0xb0)
D (24469) wifi:start 1s AUTH timer
D (24470) wifi:clear scan ap list
D (25470) wifi:auth timeout
I (25470) wifi:state: auth -> init (0x200)
D (25473) wifi:stop beacon and connect timers
D (25473) wifi:connect status 1 -> 4
D (25473) wifi:reason: auth expire(2)
```

zhangyanjiaoesp, 2024-09-05:
> 这个抓包是能抓到Authentication数据包的，发出来的这个是有问题那个模块的抓包文件。另外正常的模块是可以抓到Authentication包的，验证过程和你发给我那个图片的差不多。我就没有发了，我现在另外给你抓个正常模块的包。
> 
> [2024-09-05_15.08.39.900.pcap.zip](https://github.com/user-attachments/files/16885379/2024-09-05_15.08.39.900.pcap.zip)

这个抓包是有问题的还是没问题的？ STA 的MAC 地址是多少？我过滤抓包看一下

zhangyanjiaoesp, 2024-09-05:
> 也还是一样的，我拉取的是最新的代码，也就是合并5.3.1的。
> 
> ```
> I (24443) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
> D (24451) wifi:connect_op: status=0, auth=5, cipher=3 
> D (24455) wifi:auth mode is not none
> D (24459) wifi:connect_bss: auth=1, reconnect=0
> I (24463) wifi:state: init -> auth (0xb0)
> D (24469) wifi:start 1s AUTH timer
> D (24470) wifi:clear scan ap list
> D (25470) wifi:auth timeout
> I (25470) wifi:state: auth -> init (0x200)
> D (25473) wifi:stop beacon and connect timers
> D (25473) wifi:connect status 1 -> 4
> D (25473) wifi:reason: auth expire(2)
> ```
> 
> ```
> 0x420b0b00: cnx_start_handoff_cb at ??:?
> ```

5.3分支和 master 分支还是有区别的，如果 5.3 也连不上，可能和分支没什么关系。

beanjs, 2024-09-05:
这个是正常模块的抓包数据，你查找Authentication就可以了。

[image]

zhangyanjiaoesp, 2024-09-05:
> > 这个抓包是能抓到Authentication数据包的，发出来的这个是有问题那个模块的抓包文件。另外正常的模块是可以抓到Authentication包的，验证过程和你发给我那个图片的差不多。我就没有发了，我现在另外给你抓个正常模块的包。
> > [2024-09-05_15.08.39.900.pcap.zip](https://github.com/user-attachments/files/16885379/2024-09-05_15.08.39.900.pcap.zip)
> 
> 这个抓包是有问题的还是没问题的？ STA 的MAC 地址是多少？我过滤抓包看一下

@beanjs 失败的设备的 MAC 地址是多少？要查看一下其他包，看是不是有其他包影响了。

beanjs, 2024-09-05:
@zhangyanjiaoesp 失败设备的MAC是ec:da:3b:5b:71:20。现在我们是单独弄了个无线路由器来测试的，这个路由器只无线设备只连接了手机和ESP32S3模块，手机是用来使用airkiss进行配网的。

zhangyanjiaoesp, 2024-09-05:
[image]

设置过滤条件之后一个包都没有，看起来STA 时候的TX模块坏了，建议你换其他 ESP32S3 模块吧。

beanjs, 2024-09-05:
STA和AP的时候使用的不是相同的TX和RX模块么？现在这个模块的AP模式是正常的。另外就是如果TX模块不正常的话smartconfig应该是不能正确接收配网信息的吧？

zhangyanjiaoesp, 2024-09-05:
smartconfig 是通过sniffer 模式接收配网信息的。

zhangyanjiaoesp, 2024-09-05:
> STA和AP的时候使用的不是相同的TX和RX模块么？现在这个模块的AP模式是正常的

AP 模式的时候能正常接收外面 STA 的连接？那你把设备设置成 STA+softAP 模式，再连接路由器试试？

zhangyanjiaoesp, 2024-09-09:
@beanjs 有什么进展吗？

beanjs, 2024-09-09:
@zhangyanjiaoesp 还是一样的问题，我现在是越来越蒙了。蓝牙是正常的，softAP是正常的，就STA不正常，我也是醉了。

zhangyanjiaoesp, 2024-09-09:
> 那你把设备设置成 STA+softAP 模式，再连接路由器试试？

@beanjs 这样试过吗？

beanjs, 2024-09-10:
> > 那你把设备设置成 STA+softAP 模式，再连接路由器试试？
> 
> @beanjs 这样试过吗？

@zhangyanjiaoesp 试过来的，这个模式下面STA还是有问题，但是softAP没有问题，我现在真的懵逼了。

zhangyanjiaoesp, 2024-09-10:
作为softAP被其他设备成功连接的 log 能上传一下吗？sdkconfig 能上传一下吗？

beanjs, 2024-09-11:
@zhangyanjiaoesp 可以，我先把手上的事情弄一下，这几天都在研究这个事情，项目进度都慢下来了。弄完就抓日志给你。

zhangyanjiaoesp, 2024-09-11:
好的，建议切换信道测试一下，连接其他信道的AP 测试一下。

beanjs, 2024-09-23:
@zhangyanjiaoesp 你看看会不会是这个引起的，我把射频功率调整到+21db以后就容易出现这个问题。恢复到+9db以后，发生这个问题的几率就变小了。另外在+21db的程序运行以后在切换成+9db的时候，问题依旧。

zhangyanjiaoesp, 2024-09-27:
@beanjs 你的意思是调用 `esp_wifi_set_max_tx_power(power)` 设置 power 值为 21 或者 9 ?

AxelLin, 2024-11-07:
@beanjs 
How is the status of this issue now?
Can you answer https://github.com/espressif/esp-idf/issues/14465#issuecomment-2378643555 ?

beanjs, 2024-11-07:
@AxelLin Already switched to esp32c3 chip

zhangyanjiaoesp, 2024-12-18:
@beanjs Can we close this issue?

closed 2024-12-19 by beanjs