espressif/esp-idf #14466: ERROR: .espressif\python_env\idf5.2_py3.10_env\Scripts\python.exe doesn't exist!

kdschlosser, 2024-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

For `export.bat` to run without error

### What is the actual behavior?

I am running `install.bat all` and that completes without error. It creates a venv of `.espressif\python_env\idf5.2_py3.8_env` and when I run `export.bat` I get the following error.

```shell
Checking if Python packages are up to date...
ERROR: .espressif\python_env\idf5.2_py3.10_env\Scripts\python.exe doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
```

it's looking for a completely different version of python. I am not sure why this is happening.

### Steps to reproduce.

clone the IDF repo.
checkout version 5.2
run install.bat
run export.bat

Alvin1Zhang, 2024-08-30:
Thanks for reporting.

kdschlosser, 2024-09-02:
> Could you check please the environment variables set in you Windows? Do you have IDF_PYTHON_ENV_PATH or other ESP-IDF variables defined?

no variables set for esp idf

    run install.bat
    run export.bat

> Did you run these in sequence in the same terminal?
Yes I ran the batch files in that order.

> Is the terminal a clean one before running install.bat or you are running this in an already active ESP-IDF environment (started with an ESP-IDF shortcut or export.bat already run in it)?

This is a clean shell and both scripts are being run from the same shell. No active IDF environments.

mfialaf, 2024-09-02:
Hi @kdschlosser,

it's strange that there are 2 pythons detected, I can think of a few things to check:
- What happens when you delete the `.espressif\python_env` and re-run the install script?
- What happens when you delete the `.espressif\python_env` and run the advised command `python.exe idf_tools.py install-python-env`?
- What gives you command `where python.exe` in your CMD? Does this change after running the install script?

It seems that the system is confused about what is the default Python version.
To resolve the issue, my suggestion is to either reinstall the Python versions or install a newer version and set it as the default during installation. 
Few notes to installing Python on Windows:
- Do you recall how you installed the Python versions? I strongly recommend using the official Python installer: https://www.python.org/downloads/windows/
- Python versions installed from Windows Microsoft Store cause issues sometimes. Especially when installed with other Python versions.

kdschlosser, 2024-09-02:
I have the Microsoft Store disabled. Though when I do the `where python.exe` I get this as the return...

```
C:\Python38\python.exe
C:\Users\********\AppData\Local\Microsoft\WindowsApps\python.exe 
```

I tried to get the version from the last entry but I get an error saying the version doesn't exist.

I have Python installed, 3.8, 3.10 and 3.11 and all 3 were installed using the Python installation utility and not from the Microsoft store.

> What happens when you delete the .espressif\python_env and re-run the install script?

It recreates the same environment

> What happens when you delete the .espressif\python_env and run the advised command python.exe idf_tools.py install-python-env?

it creates the same environment

If I specify the full path to my python 3.10 installation and run idf_tools it creates the environment it is looking for when the failure occurs.

I think I figured out what is going on.. The install script is creating the environment based off what is returned from `where`. and I find that to be rather foolish. Firstly there should be a parameter that can be passed to the install.bat script so the python path can be user set.... The issue is I have an alias set up using `DOSKEY` that points python to my 3.10 installation. The export.bat script doesn't use `where` when running whatever python script it is running to set up the environment. It is blindly making a call to the shell to run `python` not knowing what version is actually going to get used. If `where` is going to be the way the install script picks which Python to use then `where` should also get run to collect the python executable that is going to be used when running the python script from export.bat.

Personally I feel `where` should not be used at all for any portion of this. What should be done is a simple registry lookup for Python installations and the one that should get used is the one that is the highest version that the ESP-IDF is able to run with. I am using this logic to make that decision. People tend to install things in the order in which they are released. As a new version comes out they install it. so you would have lower version of Python installed first. When a new path is added to the PATH environment variable how is it typically done? Well it can either be prepended or it can be appended. Appending is a whole lot easier to do in just about every programming language that is written and that is usually what gets done. The results of running `where` is then going to list the oldest versions first yes? Yup... What versions of python are the most likely to get uninstalled from a users PC?? The oldest ones.. So why on earth would the install script use `where` to determine which python to use?

this regkey gives the different installed python versions along with their installation path. it even tells you what features are installed with that version so you can check to see if the needed parts are installed for python to run. Things like pip which are fairly important to have and if the installation doesn't have it installed then why even bother using it?
```
/HKEY_LOCAL_MACHINE/SOFTWARE/Python/PythonCore
```

Here is a batch file script that queries the registry and collects the highest version of Python that is installed onto the system. I am sure it could be written better as I am not a window batch file expert or anything like that. There should also be a check in there to see if the executable even exists prior to setting it as the one that has been found. There also needs to be a check to see if anything has been found. But it gives the basic idea and this is what should be in both the `install.bat` and also the `export.bat` files for selecting the version of python to use. Even where will report erroneous paths to the python executable so a check for the existence of the executable should ALWAYS be done.

```batch
@echo off
setlocal
setlocal enabledelayedexpansion

set "_version_major=0"
set "_version_minor=0"
set _found_python_path=""

for /f "tokens=1" %%a in ('REG QUERY HKEY_LOCAL_MACHINE\SOFTWARE\Python\PythonCore /s /f InstallPath ^| findstr "HKEY_LOCAL_MACHINE"') do (
  set _python_reg=%%a

  for /f "tokens=3" %%b in ('REG QUERY !_python_reg! /v ExecutablePath') do (
    set _python_path=%%b
  )

  for /f "tokens=5 delims=\" %%v in ("%%a") Do (
    set _version=%%v

    for /f "tokens=1,2 delims=." %%p in ("%%v") Do (
      set "_major=%%p"
      set "_minor=%%q"

      IF !_major! gtr !_version_major! (
        set "_version_major=%%p"
        set "_version_minor=%%q"
        set _found_python_path=!_python_path!
      ) ELSE (
        IF !_major!==!_version_major! IF !_minor! gtr !_version_minor! (
          set "_version_major=%%p"
          set "_version_minor=%%q"
          set _found_python_path=!_python_path!
        )
      )
    )
  )
)
echo !_found_python_path!
endlocal
```

The above is a better way to collect the executable location for python. It will not get screwed up by something like an alias being set.

kdschlosser, 2024-09-02:
another note is any time a subprocess is started from inside of a python script to run another session of python the fill path to the running python executable of python should be given. this is easily gotten from `sys.executable` passing `python` only as the executable name can and will cause problems if there are any aliases in place. Never rely on thing being how you would think they would be. Always write code that removed the possibility of variations and providing FULL paths to an executable and not relying on things being in the `PATH` is the ONLY way to be 100% sure.

fhrbata, 2024-09-03:
Hello @kdschlosser ,

I'm not using Windows, so I can't confirm this, but I think batch files run in a separate context, and doskey macros don't carry over between different contexts or sessions.

https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/doskey

> To run a macro, type the macro name at the command prompt, starting at the first position. If the macro was defined with $* or any of the batch parameters $1 through $9, use a space to separate the parameters. You cannot run a doskey macro from a batch program.

In my opinion, what's happening is that while `install.bat` runs in a different context, the doskey macro isn't visible, so `python.exe` points to your 3.8 version. This means the virtual environment is installed with this Python version.
 However, `export.bat` runs in the current cmd.exe context, where your doskey macro is available and points to Python 3.10.

This is just my guess what might be happening and I of course may be wrong. Also this doesn't solve the problem you are seeing, I'm just trying to understand how this situation can happen.

Thank you

mfialaf, 2024-09-03:
Hi @kdschlosser 

I am glad you find a way to make it work.

I understand your view on finding and using the latest version of Python. It's important to consider that there are numerous use cases for Python, and developers may require different versions for specific purposes. Thus, they may implement various ways to invoke particular versions. Understanding the diverse needs of users,  we use the most common method to detect Python.

To fix your problem, I can recommend two solutions:
  - Set up our `IDF_PYTHON_ENV_PATH` environmental variable after you create the required python version directory `.espressif\python_env\idf5.2_py3.10_env`. This ensures that the ESP-IDF finds the Python environment you want to use every run.
- Use our [ESP-IDF Tools Installer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer). You can choose to download or use existing ESP-IDF version, and all the required tools with python virtual environment are set up automatically.

fhrbata, 2024-09-03:
> batch files run in the same context as the command shell they were launched from and the command they run also run in that same cmd shell unless they explicitly start a new one.
> 
> right in the export.bat file there are doskey commands to map python source files to python

My apologies for the noise. I didn't realize that `export.bat` is actually executed on Windows rather than sourced as it is on Linux. Thank you

dobairoland, 2024-09-09:
@kdschlosser Did the suggestion of @mfialaf about setting that environment variable or using the Windows installer helped you to resolve the issue?

closed 2024-09-16 by dobairoland