espressif/esp-idf #14467: i2c master read ESP_ERR_TIMEOUT when ticks_to_wait is set to 0

dizcza, 2024-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4-700-ge026fd1f81

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO T8

### Power Supply used.

USB

### What is the expected behavior?

`i2c_master_read_from_device` returns ESP_OK

### What is the actual behavior?

`i2c_master_read_from_device` returns ESP_ERR_TIMEOUT

### Steps to reproduce.

```
uint8_t data[2] = { 0 };
TickType_t ticks_to_wait = 0;
esp_err_t err = i2c_master_read_from_device(0, 0x28, data, sizeof(data), ticks_to_wait);
```

If I set `ticks_to_wait = pdMS_TO_TICKS(10)`, `i2c_master_read_from_device` returns ESP_OK but the sampling frequency drops from 830 to 750 Hz, which is not acceptable.

### More Information.

I have an I2C pressure sensor which works on release/v5.1 Mar 24th `TickType_t ticks_to_wait = 0` @ 830 Hz and it does not work  on the latest release/v5.1 branch if `ticks_to_wait` is set to 0. The sensor can output its internal code error in case of timeout but the result of `i2c_master_read_from_device` was still ESP_OK. Now it's always `ESP_ERR_TIMEOUT` unless I add non-zero waiting time to `i2c_master_read_from_device`.

dizcza, 2024-08-30:
Posting the content of `sdkconfig.default` if it helps:
```
CONFIG_BOOTLOADER_LOG_LEVEL_WARN=y
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_SPI_SLAVE_ISR_IN_IRAM=n
CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC=n
CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=n
CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID=n
CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT=n
CONFIG_GPIO_CTRL_FUNC_IN_IRAM=y
CONFIG_RMT_SUPPRESS_DEPRECATE_WARN=y
CONFIG_MCPWM_ISR_IRAM_SAFE=y
CONFIG_HTTPD_MAX_REQ_HDR_LEN=768
CONFIG_ESP_HTTPS_OTA_DECRYPT_CB=y
CONFIG_ESP32_REV_MIN_3=y
CONFIG_PM_ENABLE=y
CONFIG_PM_SLP_IRAM_OPT=y
CONFIG_PM_RTOS_IDLE_OPT=y
CONFIG_PM_SLP_DISABLE_GPIO=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH=y
CONFIG_RINGBUF_PLACE_ISR_FUNCTIONS_INTO_FLASH=y
CONFIG_ESP_MAIN_TASK_STACK_SIZE=16384
CONFIG_ESP_INT_WDT_TIMEOUT_MS=300
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=30
CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL_5=y
CONFIG_ESP_WIFI_ENABLE_WPA3_OWE_STA=n
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
CONFIG_ESP_COREDUMP_CHECKSUM_SHA256=y
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=12
CONFIG_ESP_COREDUMP_STACK_SIZE=1500
CONFIG_FATFS_LFN_STACK=y
CONFIG_FATFS_MAX_LFN=128
CONFIG_FATFS_ALLOC_PREFER_EXTRAM=n
CONFIG_FREERTOS_USE_TICKLESS_IDLE=y
CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH=y
CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM=y
CONFIG_LWIP_IPV6=n
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=5744
CONFIG_LWIP_TCP_WND_DEFAULT=5744
CONFIG_LWIP_SNTP_MAX_SERVERS=2
CONFIG_LWIP_SNTP_UPDATE_DELAY=600000
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE=y
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH="components_esp32/ota_loader/ca_cert.pem"
CONFIG_PTHREAD_DEFAULT_CORE_0=y
CONFIG_SPI_FLASH_SUPPORT_ISSI_CHIP=n
CONFIG_SPI_FLASH_SUPPORT_MXIC_CHIP=n
CONFIG_SPI_FLASH_SUPPORT_GD_CHIP=n
CONFIG_SPI_FLASH_SUPPORT_WINBOND_CHIP=n
CONFIG_SPI_FLASH_ENABLE_ENCRYPTED_READ_WRITE=n
CONFIG_WS_TRANSPORT=n
CONFIG_ULP_COPROC_ENABLED=y
CONFIG_ULP_COPROC_RESERVE_MEM=600
CONFIG_VFS_SUPPORT_SELECT=n
CONFIG_VFS_SUPPORT_TERMIOS=n
CONFIG_HULP_MACRO_OPTIMISATIONS=y
CONFIG_SENSORLIB_ESP_IDF_OLD_API=y
CONFIG_MIPI_DISPLAY_EXCLUSIVE_BUS=n
CONFIG_MIPI_DISPLAY_PIN_MISO=2
CONFIG_MIPI_DISPLAY_PIN_MOSI=15
CONFIG_MIPI_DISPLAY_PIN_CLK=14
CONFIG_MIPI_DISPLAY_PIN_CS=27
CONFIG_MIPI_DISPLAY_PIN_DC=26
CONFIG_MIPI_DISPLAY_PIN_RST=-1
CONFIG_MIPI_DISPLAY_PIN_BL=-1
```

mythbuster5, 2024-08-30:
I guess is this commit https://github.com/espressif/esp-idf/commit/4c896cbb3a516b0dd91666323593b6d534b86a80

dizcza, 2024-08-30:
> I guess is this commit [4c896cb](https://github.com/espressif/esp-idf/commit/4c896cbb3a516b0dd91666323593b6d534b86a80)

You're right. After I reverted this commit, the issue disappeared and I'm getting the expected 830 Hz sample freq.

AxelLin, 2024-09-13:
> I guess is this commit [4c896cb](https://github.com/espressif/esp-idf/commit/4c896cbb3a516b0dd91666323593b6d534b86a80)

@mythbuster5 Would you fix this?

AxelLin, 2024-10-13:
@mythbuster5 Any update to fix this?

AxelLin, 2024-11-04:
> I guess is this commit [4c896cb](https://github.com/espressif/esp-idf/commit/4c896cbb3a516b0dd91666323593b6d534b86a80)

@mythbuster5 When will you fix this regression?

AxelLin, 2025-01-12:
> > I guess is this commit [4c896cb](https://github.com/espressif/esp-idf/commit/4c896cbb3a516b0dd91666323593b6d534b86a80)
> 
> @mythbuster5 Would you fix this?

@mythbuster5 Any update? It's strange that you don't fix regression.

AxelLin, 2025-02-12:
@mythbuster5 This has been broken since 4c896cb, do you have ETA for the fix?

AxelLin, 2025-03-13:
@mythbuster5 
Do you still maintain the i2c bug fixes? 

@igrr 
This is not the first time I experienced that no response from espressif for several month.
What's your suggestion to handle such situation?

mythbuster5, 2025-03-14:
It's hard to say this as a regression because i2c is no way to reture without timeout if the ticks_to_wait has been set to 0 right? Change original behavior not always means regression.

@o-marshmallow  Please also suggest as this is your commit.

o-marshmallow, 2025-03-14:
The commit pointed previously [4c896cb](https://github.com/espressif/esp-idf/commit/4c896cbb3a516b0dd91666323593b6d534b86a80) fixed an issue where calling `i2c_master_cmd_begin` would result in a timeout of at least 1s, no less. This is **not** a regression. In fact, by reverting this commit, passing `0` as `ticks_to_wait` to the function would behave the same way as if `I2C_CMD_ALIVE_INTERVAL_TICK` was given, which is an invalid behavior.

With the latest commit on that version, by passing 0 to `i2c_master_cmd_begin`, you are telling the driver that the transaction shall finish almost instantly, you're not giving time to the driver to receive any data from the interrupt service routine. In fact, as described in the code, `wait_time` would always result in 0 being passed to `xQueueReceive`. The latter will return immediately no matter if the queue is full or empty.

As to why the frequency is slower with this commit compared to the former one, I can't explain it, both bheaviors should be the same, passing a non-0 value to `xQueueReceive` can result in a context switch which may hold up and delay your I2C task.

Please note that this driver is now the legacy one, it is advised to update to newer versions of IDF where a more sturdy driver is available.

AxelLin, 2025-03-17:
@mythbuster5 @o-marshmallow 

IMHO, https://github.com/espressif/esp-idf/issues/14467#issuecomment-2320789746 already proves this is a regression, or at least this is a breaking change.
Which means you need to either fix it or document the change so user can know the application code needs update after updating esp-idf.
If you guys think this is not a regression, yon should reply earlier rather than zero response for half year.