espressif/esp-idf #14470: esp_http_client mbedtls_ssl_handshake returned -0x3000

dizcza, 2024-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4-700-ge026fd1f81

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO T8

### Power Supply used.

USB

### What is the expected behavior?

`esp_http_client_perform` results in ESP_OK with no mbedtls errors

### What is the actual behavior?

```
E (12:30:14.572) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (12:30:14.580) HTTP_CLIENT: Connection failed
```

### Steps to reproduce.

```python
app = Flask(__name__)
auth = HTTPBasicAuth()

@app.route("/tasks")
@auth.login_required
def tasks():
    tasks_text = """
    250 all download '2024-02-15 06:10' '2024-02-15 06:15'
    298 all download '2024-08-29 11:10' '2024-08-29 11:44'
    """
    return Response(tasks_text, status=200, mimetype="text/plain")
```

```
gunicorn --certfile /home/vps/cert/ca_cert.pem --keyfile /home/vps/cert/ca_key.pem --worker-class gevent -b 0.0.0.0:8070 app_web:app
```

```c
void ota_http_client_config_init(esp_http_client_config_t *config) {
    memset(config, 0, sizeof(esp_http_client_config_t));
    config->username = ENV_OTA_SERVER_USER;
    config->password = ENV_OTA_SERVER_PASS;
[... 942 lines cut ...]
In both cases, I'm using the same `sdkconfig.defaults` (see https://github.com/espressif/esp-idf/issues/14467#issuecomment-2320515015) and the same HTTPS endpoint.

This code works fine with no errors:
```
I (00:00:04.098) esp_netif_handlers: example_netif_sta ip: 192.168.3.113, mask: 255.255.255.0, gw: 192.168.3.1
I (00:00:04.100) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.3.113
I (00:00:04.109) example_common: Connected to example_netif_sta
I (00:00:04.116) example_common: - IPv4 address: 192.168.3.113,
I (00:00:04.124) HTTP_CLIENT: Connected to AP, begin http example
I (00:00:04.129) main_task: Returned from app_main()
I (00:00:04.389) esp-x509-crt-bundle: Certificate validated
I (00:00:06.176) HTTP_CLIENT: HTTPS Status = 200, content_length = 109
I (00:00:06.177) HTTP_CLIENT: 
250 all download '2024-02-15 06:10' '2024-02-15 06:15'
298 all download '2024-08-29 11:10' '2024-08-29 11:44'
I (00:00:06.196) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (00:00:07.484) esp-x509-crt-bundle: Certificate validated
I (00:00:09.126) HTTP_CLIENT: HTTPS Status = 200, content_length = 109
I (00:00:09.127) HTTP_CLIENT: 
250 all download '2024-02-15 06:10' '2024-02-15 06:15'
298 all download '2024-08-29 11:10' '2024-08-29 11:44'
I (00:00:09.137) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (00:00:10.442) esp-x509-crt-bundle: Certificate validated
I (00:00:12.006) HTTP_CLIENT: HTTPS Status = 200, content_length = 109
```

dizcza, 2024-08-30:
Fixed by increasing the keep-alive timeout on the server side.

In particular, I added `--keep-alive 10` to my gunicorn command.

Not sure if I need to enable the keep-alive in the ESP HTTP client config -- added it to be safe.

dizcza, 2024-08-30:
Increasing keep alive helps but hasn't fully resolved the problem. Sometimes I still get

```
E (15:56:57.241) esp-x509-crt-bundle: PK verify failed with error FFFFBD70
E (15:56:57.245) esp-x509-crt-bundle: Failed to verify certificate
E (15:56:57.255) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (15:56:57.272) HTTP_CLIENT: Connection failed
W (15:56:57.275) app_http: app_http_client_get_tasks status = 0, content_length = 0
E (15:56:59.479) esp-x509-crt-bundle: PK verify failed with error FFFFBD70
E (15:56:59.483) esp-x509-crt-bundle: Failed to verify certificate
E (15:56:59.509) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (15:56:59.513) HTTP_CLIENT: Connection failed
W (15:56:59.515) app_http: app_http_client_get_tasks status = 0, content_length = 0
```

nileshkale123, 2024-09-25:
Hello @dizcza ,

Sorry for the delay in reply.

Could you please try providing a certificate through "cacert_buf" method for this endpoint (in place of certificate bundle)? Please refer to code at [this](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_request/main/https_request_example_main.c#L193).

More information on various approaches of server verification are [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#tls-server-verification)

dizcza, 2024-09-27:
@nileshkale123 I'm not using the `https_request` API. I'm using `esp_http_client` API for both GET and POST requests.

I guess this corresponds to setting `cert_pem`:

```c
void ota_http_client_config_init(esp_http_client_config_t *config) {
    extern const char ca_cert_pem_start[] asm("_binary_ca_cert_pem_start");
    extern const char ca_cert_pem_end[] asm("_binary_ca_cert_pem_end");

    memset(config, 0, sizeof(esp_http_client_config_t));
    config->username = ENV_OTA_SERVER_USER;
    config->password = ENV_OTA_SERVER_PASS;
    config->auth_type = HTTP_AUTH_TYPE_BASIC;
    config->timeout_ms = OTA_LOADER_TIMEOUT_MS;
    config->skip_cert_common_name_check = true;
    config->cert_pem = ca_cert_pem_start;
    config->cert_len = ca_cert_pem_end - ca_cert_pem_start;
    config->transport_type = HTTP_TRANSPORT_OVER_SSL;
}
```

As you might have guessed, I'm using the same client config for both HTTPS OTA updates and HTTPS GET/POST client requests.

And I still get the same errors

```
W (14:33:10.300) app_http: Last esp error code: ESP_ERR_MBEDTLS_SSL_SETUP_FAILED
W (14:33:10.303) app_http: Last mbedtls failure: 0x7f00
E (14:33:20.250) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (14:33:20.255) esp-tls: create_ssl_handle failed
E (14:33:20.259) esp-tls: Failed to open new connection
E (14:33:20.262) transport_base: Failed to open a new connection
E (14:33:20.272) HTTP_CLIENT: Connection failed, sock < 0
```

And on the server side

```python
Traceback (most recent call last):
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/workers/base_async.py", line 65, in handle
    util.reraise(*sys.exc_info())
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/util.py", line 626, in reraise
    raise value
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/workers/base_async.py", line 48, in handle
    req = next(parser)
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/http/parser.py", line 42, in __next__
    self.mesg = self.mesg_class(self.cfg, self.unreader, self.source_addr, self.req_count)
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/http/message.py", line 180, in __init__
    super().__init__(cfg, unreader, peer_addr)
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/http/message.py", line 54, in __init__
    unused = self.parse(self.unreader)
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/http/message.py", line 192, in parse
    self.get_data(unreader, buf, stop=True)
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/http/message.py", line 183, in get_data
    data = unreader.read()
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/http/unreader.py", line 37, in read
    d = self.chunk()
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gunicorn/http/unreader.py", line 64, in chunk
    return self.sock.recv(self.mxchunk)
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gevent/_ssl3.py", line 555, in recv
    return self.read(buflen)
  File "/home/vps/miniconda3/lib/python3.7/site-packages/gevent/_ssl3.py", line 386, in read
    return self._sslobj.read(nbytes or 1024)
OSError: [Errno 0] Error
```

nileshkale123, 2024-10-07:
Hello @dizcza 

The error code 0x7F00 relates to a heap memory issue, indicating that there might not be enough memory available during the SSL setup phase. Could you please check the free heap memory before initiating the connection, especially just before the esp_http_client configuration is set up?

If the available memory is low, you might consider optimizing memory usage or increasing the heap size, if possible.

Additionally, could you please share a small executable example using an IDF project so that we can reproduce the issue and provide further assistance?

dizcza, 2024-10-10:
Hmm. It seems I cannot reliably reproduce the issue: it happened only once or twice daily during the last two days.

Could it be due to changing `esp_crt_bundle_attach` to `cert_pem`... hard to tell.

I need to add another ESP32 device to talk to the same server and see what happens.

I'll reopen the issue if I get more description.

closed 2024-10-10 by dizcza

dizcza, 2024-11-01:
Ok, after I

* on the ESP-IDF side, changed `esp_crt_bundle_attach` to `cert_pem` and
* on the server side, replaced gunicorn with nginx,

the problem disappeared.

Either of the steps alone didn't solve the problem. Only when I did both things, the issue was gone. Stop using gunicorn!