espressif/esp-idf #14472: mbedtls_ssl_ciphersuite_uses_psk() not found when compiling with v5.3

pptz, 2024-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Successful build of my project (successfully built with v5.2.1)

### What is the actual behavior?

Build fails with v5.3

### Steps to reproduce.

It would take some work to write a minimal example on which the issue reproduces, but I'm using libcoap with mbedtls and have PSK enabled (`CONFIG_MBEDTLS_PSK_MODES=y` etc).

### Build or installation Logs.

```plain
/Users/dev/dev/nami-esp/tmp/nami-esp/components/coap/libcoap/src/coap_mbedtls.c: In function 'set_ciphersuites':
/Users/dev/dev/nami-esp/tmp/nami-esp/components/coap/libcoap/src/coap_mbedtls.c:1069:18: error: implicit declaration of function 'mbedtls_ssl_ciphersuite_uses_psk'; did you mean 'mbedtls_ssl_ciphersuite_get_id'? [-Werror=implicit-function-declaration]
 1069 |         else if (mbedtls_ssl_ciphersuite_uses_psk(cur)) {
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  mbedtls_ssl_ciphersuite_get_id
```
```

### More Information.

The function in question is declared in `esp-idf/components/mbedtls/mbedtls/library/ssl_ciphersuites_internal.h`, line 25. I verified with #pragma prints that the corresponding #defines are set, and the above file is included in `esp-idf/components/mbedtls//mbedtls/library/ssl_misc.h`, and from there elsewhere. I ran out of ideas about how this could be a misconfiguration problem on my end, hence I report this.

sfzhi, 2024-09-01:
You should update the CoAP component to version 4.3.4~3.

mahavirj, 2024-09-02:
@pptz This issue was fixed through https://github.com/obgm/libcoap/pull/1381. As suggested in above comment, please use latest version of `coap` from the [component manager](https://components.espressif.com/components/espressif/coap/).

closed 2024-09-02 by mahavirj