espressif/esp-idf #14474: Bootloader update/upgrade

eriksl, 2024-08-31:
### Is your feature request related to a problem?

This is not really a feature request (although it might be, depending on your answer). I could not find any documentation on this.

When IDF is upgraded from v5 to v6 at some time in the future, will the (v5) boot loader still be able to load v6 images?

This is something I will need, because my ESP32s will be at unreachable places, where I cannot simply put an USB plug in.

Or alternatively, will there be an API that allows the boot loader to be upgraded OTA? I know I can come up with something like that myself, just overwrite the flash sectors. But it feels a bit odd to use the OTA system for the image and then have to kludge something myself for the boot loader.

KaeLL, 2024-09-02:
#14195

KonstantinKondrashov, 2024-09-03:
Hi @eriksl!
> will the (v5) boot loader still be able to load v6 images?

Yes, the v5.x bootloader will be able to load a v6.x application.

> Or alternatively, will there be an API that allows the boot loader to be upgraded OTA?

We are working on this feature. It will be possible to do it, but with some limitations. Bootloader OTA update can not be done safely, if power is dropped in the middle of the writing to the bootloader partition then the device will become unbootable.

> I know I can come up with something like that myself, just overwrite the flash sectors. But it feels a bit odd to use the OTA system for the image and then have to kludge something myself for the boot loader.

Yes, you are right the current OTA component requires not too many changes to support it for bootloader.

eriksl, 2024-09-03:
That is great news, thanks!

KonstantinKondrashov, 2024-09-04:
I close this issue. When the bootloader OTA update feature is released then you will see it in #14195. Thanks.

closed 2024-09-04 by KonstantinKondrashov