espressif/esp-idf #14475: issue:The system crashed while downloading the program

AaronPMM, 2024-08-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My development environment is as follows:
VirtualBox: 7.0.18r162988 (Qt5.15.2)
Ubuntu: 20.04  (8GB of memory allocated)
esp-adf:e27cd5262986dbf93785d44fb71dfa854b8e9763 (commit ID, master)
esp-idf:347800bcda913fcce862fa753a9132c84d189dd4(commit ID, master)

idf.py -p /dev/ttyACM0 flash command is used to download the program to the development board. During this process, the system crashes, causing the program to fail to be downloaded. The following problems exist:
1. Is there a problem with my configured environment? If there is a problem with the environment configuration, what is the correct configuration?
2. Have you ever encountered this situation?
3. How to solve the crash problem?

radimkarnis, 2024-09-01:
Hello @AaronPMM,
please provide the log of the failure when running the `idf.py -p /dev/ttyACM0 flash` command.
Without it, it is impossible to tell what happened and debug your issue. Thank you.

dobairoland, 2024-09-02:
Probably this is not directly related to the issue but ESP-ADF master is not yet supporting ESP-IDF master. See https://github.com/espressif/esp-adf?tab=readme-ov-file#idf-version for more information.

closed 2024-09-16 by dobairoland