espressif/esp-idf #14476: usage of i2c diver causes StoreProhibited exception

Andrew-Exeter, 2024-08-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2-639-g43098fc4de, v5.3-383-g0bbd728196, v5.4-dev-2596-g6e5414b6c4

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

NodeMCU-ESP32 ESP32 DEVKITV1

### Power Supply used.

USB

### What is the expected behavior?

`i2c_master_transmit_receive`  function should return error code on i2c transaction failure.

### What is the actual behavior?

`StoreProhibited` exception is thrown by accessing to null ptr in function `i2c_ll_read_rxfifo`.

### Steps to reproduce.

1. Create new project from "template-app" template.
2. Copy and paste provided  `main/main.c` and `main/CMakeLists.txt` files.
3. Update i2c device address and register address. (I was reading status register of QMC5883 magnetometer.)
4. Simulate i2c device malfunction or bus interference.

- **main/main.c**
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2c_master.h"

void app_main(void)
{
  i2c_master_bus_config_t i2c_mst_config = {
    .i2c_port = I2C_NUM_1,
    .sda_io_num = GPIO_NUM_19,
    .scl_io_num = GPIO_NUM_18,
    .clk_source = I2C_CLK_SRC_DEFAULT,
    .glitch_ignore_cnt = 7,
    .intr_priority = 0,
    .trans_queue_depth = 0,
    .flags = {
      .enable_internal_pullup = true,
    }
  };
[... 178 lines cut ...]
0x400db127: s_i2c_send_commands at C:/Users/Andrew_Exeter/esp/v5_3/components/esp_driver_i2c/i2c_master.c:461
0x400db2c0: s_i2c_transaction_start at C:/Users/Andrew_Exeter/esp/v5_3/components/esp_driver_i2c/i2c_master.c:584
0x400db395: s_i2c_synchronous_transaction at C:/Users/Andrew_Exeter/esp/v5_3/components/esp_driver_i2c/i2c_master.c:872
0x400dbe01: i2c_master_transmit_receive at C:/Users/Andrew_Exeter/esp/v5_3/components/esp_driver_i2c/i2c_master.c:1097
0x400d7a2a: app_main at C:/Users/Andrew_Exeter/espressif_projects/i2c_test/main/main.c:48
0x400eb358: main_task at C:/Users/Andrew_Exeter/esp/v5_3/components/freertos/app_startup.c:208
0x400863e5: vPortTaskWrapper at C:/Users/Andrew_Exeter/esp/v5_3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### More Information.

My project crashed time by time and always with same exception/backtrace.
To exclude posibilities that problem is related to other project specific code or configuration I created new project as described in "Steps to reproduce" section, but with same behavior.
Clean installations of esp-idf and tools didn't help. (I tried versions v5.2, v5.3, master)
I tried `/examples/peripherals/i2c/i2c_tools` example too. During i2c device register reading ("i2cget" command) occurs exactly same exception.

This problem araised mostly (almost exclusively) with QMC5883 sensor in comparison to other i2c devices.

It looks that `i2c_isr_receive_handler` function tries to write some data to user read/write buffer (`data` member of `i2c_operation_t`) which is not assigned to i2c operation of type `I2C_LL_CMD_STOP`, but I could be wrong.

I am attaching backtrace, states of lacal variables and states of same other i2c driver related objects acquired using GDB debugger:

[gdb-backtrace.txt](https://github.com/user-attachments/files/16825172/gdb-backtrace.txt)
[gdb-i2c_master_bus_t-related.txt](https://github.com/user-attachments/files/16825173/gdb-i2c_master_bus_t-related.txt)
[gdb-locals.txt](https://github.com/user-attachments/files/16825174/gdb-locals.txt)

mythbuster5, 2024-09-02:
No it shouldn't be this case `It looks that i2c_isr_receive_handler function tries to write some data to user read/write buffer (data member of i2c_operation_t) which is not assigned to i2c operation of type I2C_LL_CMD_STOP, but I could be wrong.` Umm, could you please try that in the latest master, setting `.scl_wait_us = 0,` with larger value like 20*1000, and see if there is anything wrong.

Secondly, what about slowing down scl frequency

As for QMC5883 is the exclusive sensor that causes this error, interesting and I'm going to got one for further debugging.

mythbuster5, 2024-09-02:
By the way, are you sure `0x0D` is correct in your case? I think `I2C hardware NACK detected` thought the address is wrong. Even this is not the root cause of this bug

Andrew-Exeter, 2024-09-02:
Thank you for your reply.

I tried following combinations of configuration changes and always was able to reproduce the same exception/backtrace:
- `.scl_speed_hz = 50000`
- `.scl_wait_us = 20000`
- `.scl_speed_hz = 50000`, `.scl_wait_us = 20000`
- `.scl_speed_hz = 10000`, `.scl_wait_us = 20000`
- `.scl_speed_hz = 10000`, `.scl_wait_us = 40000`
- `.scl_speed_hz = 10000`, `.scl_wait_us = 40000`, `.i2c_port = I2C_NUM_0`
- `.scl_speed_hz = 10000`, `.scl_wait_us = 40000`, `.intr_priority = 3`
- `.scl_speed_hz = 10000`, `.scl_wait_us = 40000`, `.glitch_ignore_cnt = 14`
- `.scl_speed_hz = 10000`, `.scl_wait_us = 40000`, `.glitch_ignore_cnt = 28`
- `.scl_speed_hz = 10000`, `.scl_wait_us = 40000`, `.glitch_ignore_cnt = 56`

Tests has been done with esp-idf repository updated to v5.4-dev-2620-g6673376297 version.

Addresses of QMC5883 and status register should be valid, they are taken over from library, which was able to read the sensor properly, and values matches with register map in datasheet. To be sure, I tested it using `/examples/peripherals/i2c/i2c_tools`:
```
i2c-tools> i2cconfig --port=1 --freq=400000 --sda=19 --scl=18
I (165256) gpio: GPIO[19]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (165256) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- 0d -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
i2c-tools> i2cget -c 0x0D -r 0x06
0x00 
i2c-tools> i2cset -c 0x0D -r 0x09 0x01
I (510556) cmd_i2ctools: Write OK
i2c-tools> i2cget -c 0x0D -r 0x06
0x05 
```
_(After sensor is switched to continuous mode by `MODE` field in "Control Register 1" (`0x09`), `DRDY` and `DOR` flags in "Status Register" (`0x06`) indicates operation.)_

Some `I2C hardware NACK detected` are in the debug log because circuit is on a breadboard, not ideal connections ...

mythbuster5, 2024-09-03:
> I tried following combinations of configuration changes and always was able to reproduce the same exception/backtrace:

Strange, anyway, my sensor is on the way, I will try to reproduce that locally and see what is exactly happening..

mythbuster5, 2024-09-04:
I got a QMC5883 but regretfully didn't reproduce the same issue. But anyway, I go through the gdb log again, It seems that it really want to receive data in STOP command as you described. I don't know why this happened excactly.. still debugging.

Because you can reproduce easily, may you please try add a line https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2c/i2c_master.c#L430 here `i2c_master->contains_read = false;` and see if this will happen once more?  Thanks @Andrew-Exeter

mythbuster5, 2024-09-04:
But..

[image]

From the gdb log, the interrupt has been triggered in a very strange place, it shouldn't be theoretical. So, do you use multi core to run i2c or enable some special configs?

mythbuster5, 2024-09-05:
no more need to check. I tried several scenario like nack or timeout, no produce anyway... i will continue going through. But a suggestion for you might to decline the error report like nack or timeout in order that try not to disturb the state machine..

mythbuster5, 2025-01-21:
Well, I tried QMC5883 with esp32 in your code and can't reproduce issue. So... If no further update, I will close this issue.

Andrew-Exeter, 2025-01-21:
I plan to look at bus and try to find some anomaly occuring before crash. Would it be possible to reopen this issue, if I find more specific method to reproduce crash?

Thank you for your help.
I appreciate the time and effort you put into this issue.

closed 2025-01-22 by mythbuster5