espressif/esp-idf #14477: portYIELD_CORE must be defined if in portmacro.h if configNUMBER_OF_CORES > 1

peterkosek, 2024-08-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF Eclipse Plugin Version: 3.0.0.202406051940,  v5.3-dirty

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have successfully compiled this without problems in the past.  I placed a queue in the code to read CAN data, and compile failed.  I removed the queue and all related code, and it still fails after a full clean.  I restarted my OS, updated from 5.2 to 5.3, and after a full clean, it still fails.  

The error given does not seem to make sense.  

### What is the actual behavior?

In file included from C:/Users/pkose/eclipse_wksp3/Garden_card_master_v3/main/main.c:15:
C:/Espressif/frameworks/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:147:10: error: #error "Missing definition: portYIELD_CORE() must be defined if in portmacro.h if configNUMBER_OF_CORES > 1"
  147 |         #error "Missing definition: portYIELD_CORE() must be defined if in portmacro.h if configNUMBER_OF_CORES > 1"
      |          ^~~~~

on main.c:15  I have a #include for "freertos/FreeRTOS.h"  as part of many includes

#include <stdio.h>
#include <stdbool.h>
#include <unistd.h>
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <string.h>
#include <inttypes.h>
#include <math.h>

#include "freertos/projdefs.h"
#include "portmacro.h"
#include "sdkconfig.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"
#include "freertos/timers.h"
#include "freertos/semphr.h"
#include "esp_chip_info.h"
#include "nvs_flash.h"

### Steps to reproduce.

With this setup in Eclipse and at the command line compile environment, I get an error at compile, as shown above.  
[sdkconfig.json](https://github.com/user-attachments/files/16825508/sdkconfig.json)
[cart_master.zip](https://github.com/user-attachments/files/16825516/cart_master.zip)
[main.zip](https://github.com/user-attachments/files/16825517/main.zip)

### Build or installation Logs.

```plain
[1000/1006] ccache C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.3-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/pkose/eclipse_wksp3/Garden_card_master_v3/build/config -IC:/Espressif/frameworks/esp-idf-v5.3/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v5.3/components/freertos/config/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/freertos/config/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.3/components/freertos/config/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.3/components/freertos/esp_additions/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/include/soc/esp32s3 -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/dma/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/ldo/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/port/esp32s3/. -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/port/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/heap/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/log/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/soc/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/soc/esp32s3 -IC:/Espressif/frameworks/esp-idf-v5.3/components/soc/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/hal/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/hal/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_rom/include/esp32s3 -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_rom/esp32s3 -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_system/port/include/private -IC:/Espressif/frameworks/esp-idf-v5.3/components/xtensa/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/xtensa/deprecated_include -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/port/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/port/freertos/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/port/esp32xx/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/port/esp32xx/include/arch -IC:/Espressif/frameworks/esp-idf-v5.3/components/lwip/port/esp32xx/include/sys -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_gpio/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_pm/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/mbedtls/port/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/mbedtls/mbedtls/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/mbedtls/mbedtls/library -IC:/Espressif/frameworks/esp-idf-v5.3/components/mbedtls/esp_crt_bundle/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/mbedtls/mbedtls/3rdparty/everest/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/mbedtls/mbedtls/3rdparty/p256-m -IC:/Espressif/frameworks/esp-idf-v5.3/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_app_format/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_bootloader_format/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/app_update/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/bootloader_support/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/bootloader_support/bootloader_flash/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_partition/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/efuse/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/efuse/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_mm/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/spi_flash/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/pthread/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_timer/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_gptimer/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_ringbuf/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_uart/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/vfs/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/app_trace/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_event/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/nvs_flash/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_pcnt/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_spi/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_mcpwm/include -IC:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_ana_cmpr/include -IC:/Espressif [... cut ...]
  147 |         #error "Missing definition: portYIELD_CORE() must be defined if in portmacro.h if configNUMBER_OF_CORES > 1"
      |          ^~~~~
```

### More Information.

I have looked at the portmacro.h file, and in the version in C:\Espressif\frameworks\esp-idf-v5.3\components\freertos\FreeRTOS-Kernel-SMP\portable\xtensa\include\freertos portYIELD_CORE() is defined.  

I am building for a ESP32S3, and so I believe that this is the relevant portmacro.h file.  The sdkconfig confirms 'Building ESP-IDF components for target esp32s3'.

Thank you

peterkosek, 2024-09-01:
I resolved this by removing the 'portmacro.h' header that is depreciated in esp-idf V 5+.

closed 2024-09-01 by peterkosek

mahmood-saadat, 2024-09-23:
Hi. I'm facing the same issue in IDF5.2.2. Can you explain in more detail how you solved this?

peterkosek, 2024-09-24:
> WARNING - UNVERIFIED EXTERNAL EMAIL: This email came from a sender outside of PeaceHealth. Do not open any attachments, click on any links, or provide information unless you recognize the email address it came from and were expecting this message.

I removed this from by main.c

#include ‘portmacro.h’

It had been required in older IDF version for some of the functionality in my code.  It is no longer needed, and seemed to cause the error that I encountered.

Peter

Sensitivity: General Business Use. This document contains proprietary information and is intended for business use only.

This message is intended solely for the use of the individual and entity to whom it is addressed, and may contain information that is privileged, confidential, and exempt from disclosure under applicable state and federal laws. If you are not the addressee, or are not authorized to receive for the intended addressee, you are hereby notified that you may not use, copy, distribute, or disclose to anyone this message or the information contained herein. If you have received this message in error, immediately advise the sender by reply email and destroy this message.

peterkosek, 2025-02-05:
The solution is on the website.

Remove #include “pormacro.h”

Sensitivity: General Business Use. This document contains proprietary information and is intended for business use only.

@peterkosek<https://urldefense.com/v3/__https://github.com/peterkosek__;!!HYnAmvs!XRjMS-aF2DyJwNxto6kpgjtGgAhKVGJ9v1qYPYuBF3zZvFVPstMqpT2AMYi4TUVKTmIEI2wFXztlG7zqE9zvclU$> Did you ever figure this out? If yes, what was the solution?

YGDL, 2025-06-06:
[image]
我使用ESP32-S3也遇到了，我选择注释掉这两个条件编译的东西，能正常使用。