espressif/esp-idf #14478: CLion & esp-idf error: Monitor requires standard input to be attached to TTY

solyvan, 2024-09-01:
Hi,

 I would like to work with esp32-wrover-ie on CLion.

- [ ] I have followed the steps from : https://www.jetbrains.com/help/clion/esp-idf.html#troubleshoot-win 

- [ ] I have done the app (Build) and flash (Build) steps successfully. 

- [ ] And when I Build monitor (Ctrl+Fn9) I have this error: 

`error: Monitor requires standard input to be attached to TTY`

If you any suggestion to solve this, I will be very gratefull.

fhrbata, 2024-09-01:
Hello @solyvan ,

I believe this problem was already discussed here https://github.com/espressif/esp-idf/issues/14233, with possible solution https://youtrack.jetbrains.com/issue/CPP-34647/ESP-IDF-Monitor-error-Monitor-requires-standard-input-to-be-attached-to-TTY

HTH

solyvan, 2024-09-01:
Hi @fhrbata ,

It works, thank you!

closed 2024-09-01 by solyvan