espressif/esp-idf #14479: more details for promiscuous mode packet for esp32c6

babybakarbakar, 2024-09-01:
### Describe the solution you'd like.

we try to debug why we can't hear some routers with esp32C6 - and our prime suspect is the MCS to high.
but esp32C6 does not have MCS field in the `wifi_pkt_rx_ctrl_t` struct (like esp32S3).

1. can you add this field to the struct?
2. and every meta-data of the packets that can be added will be very nice:)

zhangyanjiaoesp, 2024-09-05:
@babybakarbakar 
 The current struct can get the  MCS info.
For ESP32C6, the `wifi_pkt_rx_ctrl_t` struct corresponds to the `esp_wifi_rxctrl_t` struct, you can get the MCS info from `he_siga1` filed. https://github.com/espressif/esp-idf/blob/6673376297b921d438790d195014b860eaf8bb17/components/esp_wifi/include/esp_wifi_he_types.h#L256

And then the `he_siga1` filed need to convert to different struct according to the `cur_bb_format`.
https://github.com/espressif/esp-idf/blob/6673376297b921d438790d195014b860eaf8bb17/components/esp_wifi/include/esp_wifi_he_types.h#L281
```
typedef enum {
    RX_BB_FORMAT_11B      = 0,           /**< the reception frame is a 11b MPDU */
    RX_BB_FORMAT_11G      = 1,           /**< the reception frame is a 11g MPDU */
    RX_BB_FORMAT_11A = RX_BB_FORMAT_11G, /**< the reception frame is a 11a MPDU */
    RX_BB_FORMAT_HT       = 2,           /**< the reception frame is a HT MPDU */
    RX_BB_FORMAT_VHT      = 3,           /**< the reception frame is a VHT MPDU */
    RX_BB_FORMAT_HE_SU    = 4,           /**< the reception frame is a HE SU MPDU */
    RX_BB_FORMAT_HE_MU    = 5,           /**< the reception frame is a HE MU MPDU */
    RX_BB_FORMAT_HE_ERSU  = 6,           /**< the reception frame is a HE ER SU MPDU */
    RX_BB_FORMAT_HE_TB    = 7,           /**< the reception frame is a HE TB MPDU */
} wifi_rx_bb_format_t;
```
For the specific struct, please refer to the [`esp_wifi_he_types_private.h`](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_private/esp_wifi_he_types_private.h) .

zhangyanjiaoesp, 2024-11-12:
@babybakarbakar can we close this issue?