espressif/esp-idf #14483: esp32c3 can not control gpio 11

TroyMitchell911, 2024-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Chip: esp32c3 FN4

I set gpio11 to switch the level once every second, but in the end I only got a voltage of about 3v3 to 3v0.

I think the code is ok because I changed 11 to 21 (21 is connected to an led) and everything works fine.

BTW: On the chip, gpio11 is the VDD_SPI identifier.

Code:

```c
gpio_config_t cfg;
cfg.mode = GPIO_MODE_OUTPUT;
cfg.pull_up_en = GPIO_PULLUP_ENABLE;
cfg.pull_down_en = GPIO_PULLDOWN_ENABLE;
cfg.pin_bit_mask = 1ull << GPIO_NUM_11;
gpio_config(&cfg);

while (1) {
  s_led_state = !s_led_state;
  gpio_set_level(GPIO_NUM_11, s_led_state);
  vTaskDelay(CONFIG_BLINK_PERIOD / portTICK_PERIOD_MS);
}
```

nopnop2002, 2024-09-03:
GPIO11 of ESP32S3 is assigned to output power supply for flash.

Unfortunately, the following commands are not supported by ESP32C3.

```
espefuse.py set_flash_voltage OFF
```

See here.
https://docs.espressif.com/projects/esptool/en/latest/esp32c3/espefuse/set-flash-voltage-cmd.html

TroyMitchell911, 2024-09-04:
> GPIO11 of ESP32C3 is assigned to output power supply for flash.
> 
> Unfortunately, the following commands are not supported by ESP32C3.
> 
> ```
> espefuse.py set_flash_voltage OFF
> ```
> 
> See here. https://docs.espressif.com/projects/esptool/en/latest/esp32c3/espefuse/set-flash-voltage-cmd.html

Humm..I have tried this command befor I see ur response:

```bash
espefuse.py -p /dev/ttyACM0 burn_efuse VDD_SPI_AS_GPIO 1
```

And now, I can not flash and run on this chip.

What do u recommend?

nopnop2002, 2024-09-04:
>What do u recommend?

Unfortunately I can't think of any

closed 2024-09-06 by TroyMitchell911